lumapu/ahoy #912: [Bug] Appearent Power always 19 var event at PF=1

DB1BMN, 2023-04-29:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus, round dot

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.12

### Github Hash

f67a68e

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Default configuration.
0.6.0 by WebInstaller
manualle updated to 0.6.12 by *.bin upload

### Error description

From Version 0.5.66+ (earlier versions not tested) observed that in "Live View" the Appearent Power is displayed always approx 19 var (18.6 ... 19.3 observed) even when Power Factor is exactly 1 and several hundered watts are produced while clear bright sun shine.

When measured at the same time with a "TASMOTA WLAN-Plug" there is no appearent power and all power injected is real (Watt == VA).

Inverter: Hoymiles HM-1200; S/N 11618104xxxx

Can anybody confirm?

[image]

[image]

GHolli, 2023-07-12:
This will be caused by rounding:
P = 692 W, Q = 19.3 var => S = 692.2691 VA, power factor = 0.9996
So I don't see a problem at the calculation here.

But I can't say anything about the accuracy of your inverter nor your metering plug (BTW, Tasmote requires calibration for better accuracy).

lumapu, 2023-07-23:
This value is read from inverter directly - so Ahoy can't do anything here.

closed 2023-07-23 by lumapu