lumapu/ahoy #913: [Improvement] Make Date-Format ISO-conform

DB1BMN, 2023-04-29:
Hi,

on several pages (Start Page, Live, System, Sunrise / Sunset) the date format is in "German Style" i.e. DD.MM.YYYY.

I propose to change it to [ISO8601-Style](https://de.wikipedia.org/wiki/ISO_8601) i.e. YYYY-MM-DD or at least to make it configurable in the Web-UI. 
Display view seems to be already correct :-)
Taking a quick look at 'api/index' it seems internally the Unix-Timestamp is used, right?

Best Regards, Mark

[image]

Argafal, 2023-05-01:
I made a quick experiment to see what this would look like. I used the toISOString() method. It outputs the date and time following  ISO8601, as per your suggestion. See also https://www.w3schools.com/jsref/jsref_toisostring.asp.

The result looks like this:
`Polling inverter(s), will pause at sunset 2023-05-01T20:12:02.000Z`

I honestly don't know if Zulu time is a better choice for someone living in CEST, but this is now strictly ISO8601. Maybe those who find this issue important could experiment a little bit and come up with a specific suggestion how to display the date/timestamp. I copy a diff of my patch below as starting point for you. Thanks :)

```
diff --git a/src/web/html/index.html b/src/web/html/index.html
index 72537e5..cdcc236 100644
--- a/src/web/html/index.html
+++ b/src/web/html/index.html
@@ -102,15 +102,15 @@
                 if(obj["disNightComm"]) {
                     if(((obj["ts_sunrise"] - obj["ts_offset"]) < obj["ts_now"])
                         && ((obj["ts_sunset"] + obj["ts_offset"]) > obj["ts_now"])) {
-                        commInfo = "Polling inverter(s), will pause at sunset " + (new Date((obj["ts_sunset"] + obj["ts_offset"]) * 1000).toLocaleString('de-DE'));
+                        commInfo = "Polling inverter(s), will pause at sunset " + (new Date((obj["ts_sunset"] + obj["ts_offset"]) * 1000).toISOString());
                     }
                     else {
                         commInfo = "Night time, inverter polling disabled, ";
                         if(obj["ts_now"] > (obj["ts_sunrise"] - obj["ts_offset"])) {
-                            commInfo += "paused at " + (new Date((obj["ts_sunset"] + obj["ts_offset"]) * 1000).toLocaleString('de-DE'));
+                            commInfo += "paused at " + (new Date((obj["ts_sunset"] + obj["ts_offset"]) * 1000).toISOString());
                         }
                         else {
-                            commInfo += "will start polling at " + (new Date((obj["ts_sunrise"] - obj["ts_offset"]) * 1000).toLocaleString('de-DE'));
+                            commInfo += "will start polling at " + (new Date((obj["ts_sunrise"] - obj["ts_offset"]) * 1000).toISOString());
                         }
                     }
                 }
```

DB1BMN, 2023-05-01:
Hi thanks for quick testing!

I think there might me some options or flags to adjust to time-zones and DST if necessary. I think NTP delivers also UTC only.
At least in PHP you have plenty of formatting options: https://www.php.net/manual/en/datetime.format.php

My only wish was to print date in YYYY-mm-dd format so just swap days and year.
And shorten sunrise/sunset times to minutes, see also: https://github.com/lumapu/ahoy/issues/914#issue-1689616457

Argafal, 2023-05-01:
Nice, good for looking into it. Maybe @DB1BMN you can come up with a specific suggestion how to change the code and then we can ask @lumapu to integrate it. The places you need to touch you can see in my example diff.

DB1BMN, 2023-05-01:
LOL, what a pleasure!

But I'm sorry, my programming skills are somewhere between '8051 Assemlby and QBASIC.
So the code you postet is only Czech Villages to me.

But I can do some testing, habe several boards spare here and not afraid to fry them ;-)

dtuuser, 2023-07-24:
ESP-Time: 2023-07-24, 07:15:04
Geht bei mit nach den Update 2 Stunden nach

closed 2023-08-05 by lumapu