lumapu/ahoy #915: [Bug]

ralfsfb, 2023-04-29:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus, round dot

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.12

### Github Hash

weiß nicht, habe die *.zip runter geladen

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

[image]

Der verwendete Elko hat eine Kapazität von 2200 µF

### Error description

Seit der Version 0.6.x ist das  Antennenmodul nicht mehr erreichbar und es erfolgt die Meldung, dass die Verkabelung überprüft werden soll. Wird anschließend bei identischer Verkabelung und Konfiguration wieder eine ältere Version (z. B. 0.5.107) geladen, funktioniert alles wieder, so dass es nicht an der Verkabelung liegen kann.

Unabhängig vom Problem mit der 0.6er Version konnte heute trotz mehrerer Neustarts keine Verbindung zu den Invertern aufgebaut werden. In der Systeminfo habe ich dann gesehen, dass eine Datenrate von 2 Mbps anstelle der üblichen 250 kbps aktiv war. Mit welchem Parameter kann man die erforderliche Datenrate von 250 kbps erzwingen?

Besten Dank für eure Hinweise und Tipps im Voraus.

Argafal, 2023-05-01:
Kannst du noch einen Screenshot von deinen Einstellungen in Version 0.6.12 machen, bitte? Ich denke, dein Screenshot oben ist von der 0.5.107, verstehe ich das richtig?

HorstyS, 2023-05-01:
Ich habe glaube ich das selbe Problem. 
Ich habe eine DTU mit der 0.5.101 die super funktioniert. Meine zweite mit der 0.6.12 (zuvor auch 0.5.101, dann 0.6.0) und dem selben Inverter funktioniert seit dem Update auf die 0.6.x nur noch sehr sporadisch oder gar nicht mehr.

ralfsfb, 2023-05-01:
> Kannst du noch einen Screenshot von deinen Einstellungen in Version 0.6.12 machen, bitte? Ich denke, dein Screenshot oben ist von der 0.5.107, verstehe ich das richtig?

Die Screenshots sind mit der 0.5.107 erstellt worden, das ist richtig. Habe natürlich die Einstellungen mit der 0.6.x verglichen - und diese waren identisch. Hatte ja auch gehofft, dass es an einer anderen Pinbelegung liegt, doch das scheint es nicht zu sein.

Argafal, 2023-05-01:
Zeig doch mal, bitte.

HorstyS, 2023-05-02:
Ich habe heute mal folgendes gemacht: 
Ich habe über den Webinstaller (ahoy Webseite) mal die Installation neu angestoßen. Dabei wurde automatisch die 0.6.9 installiert. Danach habe ich das WLAN neu verbunden, die PINs wieder zugewiesen wie es zuvor war und den Inverter neu eingetragen.
Seitdem läuft Ahoy nun seit  1 1/2 Stunden stabil und die Datenrate bleibt konstant bei 250 kbps.

Bei der Installation wird wohl auch zuvor alles im Speicher gelöscht, so wie ich das in einer Meldung gesehen habe. Liegt es vielleicht daran?

Argafal, 2023-05-02:
@HorstyS Wenn ich das richtig verstehe, ging es bei einem Update für dich nicht, aber bei einer Neuinstallation schon. 

Vielleicht liegt's daran: Bei einem _Upgrade_ von 0.5.x auf 0.6.x müssen (auf ESP32) drei zusätzliche Pins manuell gesetzt werden: MISO, MOSI, und SCLK. Siehe auch die Release-Notes/Doku. Bei einer Neuinstallation werden diese Pins automatisch initialisiert. 

@ralfsfb Ich bin gespannt auf deinen Setup Screenshot von 0.6.x.  Wie sehen deine MOSI, MISO und SCLK Pins aus?

HorstyS, 2023-05-02:
Korrekt, so sieht es wohl aus....
Ich kann mich schon erinnern, dass ich bei dem Upgrade auf die 0.6.x PINs noch setzen musste, aber danach schien aller erst mal soweit zu funktionieren. Das Display hat funktioniert, es wurden ja auch Daten nach dem Neustart empfangen. Aber das war glaube ich immer nur einmalig, danach war Schluss und es wurden immer nur 2 kbps angezeigt.

lumapu, 2023-05-03:
meines wissens sind die 250kbps nicht konfigurierbar. @beegee3, @rejoe2 habt ihr eine Idee hierzu?

Wie erwartet ist es hart codiert:
https://github.com/lumapu/ahoy/blob/7be4ae4536437852d3c95fba07ee8e083ef75aa1/src/hm/hmRadio.h#L124

Kann es sein, dass ihr NRF24 Module habt, die keine `+`-Variante sind? Ohne Plus können die mW. nur 2Mbps, genau hier liegt der Unterschied. Eigenartig ist allerdings, dass es schon mal ging, das würde aus meiner Sicht eher auf einen SPI-Kommunikationsfehler hinweisen.

HorstyS, 2023-05-04:
Also bei mir ist es definitiv die +-Version. Steht so auf der Platine (habe alles im Markershop bestellt, daher gehe ich davon aus, dass es original ist) und es wird ja auch 250 kbps angezeigt am Anfang. Dann bricht es irgendwann am 2 kbps ein und das war's dann. 
Aktuell läuft es bei mir noch nach dem Löschen und Neuinstallieren der 0.6.9 seit ein paar Tage....mit 250 kbps.

Was mir noch auffällt: Ich habe ein eInk Display. Wenn keine Daten mehr kommen wird das "milchig", sprich der Kontrast des Displays reduziert sich wohl. Keine Ahnung ob das auch was zu bedeuten hat?!

Argafal, 2023-05-04:
Ich denke, dass nach einem Update die SPI Pins nicht richtig konfiguriert waren, und der Fehler daher kam. Dafür spricht dass nach einer Neuinstallation wieder alles geht. Für @HorstyS scheint es ja jetzt zu laufen. Kann @ralfsfb auch noch berichten?

beegee3, 2023-05-04:
> meines wissens sind die 250kbps nicht konfigurierbar

bestätige: die 250kbps sind fix im Code verankert.
Bei meinen diversen Radio Tests hatte ich allerdings festgestellt, dass der Inverter auch bei 2MBps antwortet (nach ca. 2-3 Minuten ohne Reaktion). Allerdings ist der channel switch timeslot nicht darauf angepasst, so dass eher selten eine Antwort empfangen wird.

@ralfsfb die Version ist selbst kompiliert. Wenn noch Probleme bestehen, probier' doch bitte mal die offiziellen Versionen aus.

ralfsfb, 2023-05-04:
Bin leider aktuell nicht daheim für direkte Kommunikation mit den Invertern, sonst hätte ich bereits früher Rückmeldungen geben können. Habe jedoch den ESP32 dabei und habe versucht, von einem anderen PC aus die 0.6.9 als auch die 0.6.12 zu übersetzen, um den ESP32 zu bespielen. Leider kommt es immer zu Fehlermeldungen, aus denen ich noch nicht schlau werde.
Folgendes erscheint im Terminalfenster von VS Code:

 *  Task wird ausgeführt: C:\Users\ralf\.platformio\penv\Scripts\platformio.exe run 

Processing esp32-wroom32-release (platform: espressif32@>=6.1.0; board: lolin_d32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Firmware Revision: g0000000
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin_d32.html
PLATFORM: Espressif 32 (6.2.0) > WEMOS LOLIN D32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20008.0 (2.0.8)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP Async WebServer @ 1.2.3+sha.1dde9cf
|   |-- AsyncTCP @ 1.1.1
|   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- ArduinoJson @ 6.21.2
|-- RF24 @ 1.4.5
|   |-- SPI @ 2.0.0
|-- Time @ 1.6.1
|-- espMqttClient @ 1.4.2+sha.06c150c
|   |-- AsyncTCP @ 1.1.1
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- ArduinoJson @ 6.21.2
|-- Timezone @ 1.2.4+sha.7cf1425
|   |-- Time @ 1.6.1
|-- U8g2 @ 2.34.17
|   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- GxEPD2 @ 1.5.2
|   |-- Adafruit GFX Library @ 1.11.5
|   |   |-- Adafruit BusIO @ 1.14.1
|   |   |   |-- Wire @ 2.0.0
|   |   |   |-- SPI @ 2.0.0
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Adafruit BusIO @ 1.14.1
|   |   |-- Wire @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
[... 407 lines cut ...]
     case 503: return F("Service Unavailable");
                      ^
C:/Users/ralf/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:38:42: error: cannot convert 'const char*' to 'const __FlashStringHelper*' in return
 #define F(string_literal) (string_literal)
                                          ^
.pio/libdeps/esp32-wroom32-release/ESP Async WebServer/src/WebResponses.cpp:83:22: note: in expansion of macro 'F'
     case 504: return F("Gateway Time-out");
                      ^
C:/Users/ralf/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:38:42: error: cannot convert 'const char*' to 'const __FlashStringHelper*' in return
 #define F(string_literal) (string_literal)
                                          ^
.pio/libdeps/esp32-wroom32-release/ESP Async WebServer/src/WebResponses.cpp:84:22: note: in expansion of macro 'F'
     case 505: return F("HTTP Version not supported");
                      ^
C:/Users/ralf/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:38:42: error: cannot convert 'const char*' to 'const __FlashStringHelper*' in return
 #define F(string_literal) (string_literal)
                                          ^
.pio/libdeps/esp32-wroom32-release/ESP Async WebServer/src/WebResponses.cpp:85:22: note: in expansion of macro 'F'
     default:  return F("");
                      ^
Compiling .pio\build\esp32-wroom32-release\lib341\Wire\Wire.cpp.o
*** [.pio\build\esp32-wroom32-release\lib8a5\ESP Async WebServer\WebRequest.cpp.o] Error 1
*** [.pio\build\esp32-wroom32-release\lib8a5\ESP Async WebServer\WebResponses.cpp.o] Error 1

Als nächstes werde ich versuchen - wie auch oben empfohlen, die 0.6.9 per Webinstaller zu installieren und damit die Kommunikationseinstellungen wieder herzustellen.

closed 2023-07-23 by lumapu