lumapu/ahoy #917: HM 600

Mike223311, 2023-04-30:
Hallo,
👍👍👍
erst mal meinen Dank an das Team der Ahoy- als auch openDTU.
👍👍👍

Ich betreibe ein Balkonkraftwerk (zwei Platten a‘ 385 Watt) mit einem Hoymiles HM 600 mit Seriennummer 114xxxxxxxxx. (Baujahr 2022)
Habe mir die Ahoy- als auch die openDTU gem. Anleitung zusammen gestellt.
Die Ahoy mit ESP8266 und NRF24L01+
Die open mit ESP32 und NRF24L01+
Beide mit der aktuellen Software geflasht.
Beide funktionieren bei mir einwandfrei, keine Probleme.

Problem:
Beide DTU‘s habe ich nun bei einem Freund ausprobiert, der auch einen Hoymiles HM600, allerdings ein älteres Model mit Seriennummer 104xxxxxxxxx betreibt.
Die AhoyDTU erkennt diesen WR und ich kann die Leistungsdaten auslesen, allerdings ist kein Powermanagement möglich.
Die openDTU erkennt den WR nicht, kein Auslesen, kein Powermanagement möglich.

Wäre super wenn die Profis beider Team‘s eine Lösung finden würden.

Grüsse

Joh49, 2023-04-30:
mit Seriennummer 104xxx ist das ein MI 600 und kein HM 600. 
Es wurde mehrfach erwähnt daß bei diesem Typ das Prowermanagement nicht funktioniert auch nicht mit der Original DTU-Pro.

Mike223311, 2023-04-30:
Sorry, bin neu hier ….. konnte bisher nicht alles lesen :-(Danke für die Info!Grüsse Am 30.04.23 um 17:44 schrieb Joh49
                
            
            
                
                    Von: "Joh49" ***@***.***>Datum: 30. April 2023An: "lumapu/ahoy" ***@***.***>Cc: "Mike223311" ***@***.***>,"Author" ***@***.***>Betreff: Re: [lumapu/ahoy] HM 600 (Issue #917)
                
                
mit Seriennummer 104xxx ist das ein MI 600 und kein HM 600.
Es wurde mehrfach erwähnt daß bei diesem Typ das Prowermanagement nicht funktioniert auch nicht mit der Original DTU-Pro.

closed 2023-05-02 by lumapu