lumapu/ahoy #918: Leistung verbessern

Glider-52, 2023-04-30:
Hallo,
Meine beiden Module werden offenbar bei 320,4 W vom Wechselrichter ab-geregelt. Kann man das ändern?.
Und wenn ja, bitte wie?
Ansonsten super Projekt!
Viele Grüße, Lothar

## Hardware
  * [ x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

BastyESP, 2023-04-30:
Einfach gegen einen HM-800 austauschen...schon erhöht sich die Leistung auf bis 428,6w.

mroenne2022, 2023-04-30:
Warscheinlich ein HM-600,  der riegelt jeden Eingang bei 320W ab , ist also normal und bei dem WR nicht zu ändern 🤷🏻

QuickStopGroceries, 2023-05-05:
> Einfach gegen einen HM-800 austauschen...schon erhöht sich die Leistung auf bis 428,6w.

Exakt bei diesem Wert hat mein HM-800 auch beide Eingänge abgeregelt. Ich war etwas überrascht, weil ich 2 Panels mit 400 Watt Peak angeschlossen habe. Offensichtlich leisten die Panels mehr als angegeben.

closed 2023-08-05 by lumapu