lumapu/ahoy #919: feature suggestion: time sync via REST API

ncqgm, 2023-05-01:
It would be helpful if the REST API would allow *setting* the system time from a client.

When run in AP mode, there's no established way for the DTU to sync time:

    NTP is not available
    a web client is only connected sporadically (and then a user still needs to explicitely click [sync with browser])
    Wifi frames only contain ticks, not timestamps as such

The time being wrong, combined with "disable night communication", leads to inverter(s)
not being polled for data.

A REST client could periodically check the DTU time and set it as needed -- if the API would
support getting/setting system time ...

An even more elaborate scheme might be this:

    IF time cannot be established,
    AND night time is subsequently assumed (as per 1.1.1970 00:00 UTC),
    DO poll the inverter at least once
    EVEN IF disable night communications is enabled

Now, if the inverter does answer and does produce it must be daylight
outside :-D     so it would make sense to assume:

    now = "sunrise on 1.1.1970 for the lat/lon set in configuration"    =8-)

Thanks for considering,
Karsten

lumapu, 2023-05-02:
das sollte per API mit 

```
{
    "path": "setup",
    "cmd": "set_time",
    "val": <UNIX_TIMESTAMP_SEC>
}
```

gehen

ncqgm, 2023-05-04:
Also ein POST nach api/setup mit den genannten JSON-Daten ?  Das wäre prima und bislang nicht dokumentiert (?) -- hab's im
Quellcode zumindest nicht finden gekonnt.

lumapu, 2023-05-09:
genau, konntest du es in der Zwischenzeit testen?

ncqgm, 2023-05-10:
Am Tue, May 09, 2023 at 11:53:58AM -0700 schrieb Lukas Pusch:

> genau, konntest du es in der Zwischenzeit testen?

Noch nicht, ich werde aber berichten.

Karsten
--
GPG  40BE 5B0E C98E 1713 AFA6  5BC0 3BEA AC80 7D4F C89B

ncqgm, 2023-05-14:
Am Tue, May 09, 2023 at 11:53:58AM -0700 schrieb Lukas Pusch:

> genau, konntest du es in der Zwischenzeit testen?

Klappt, Issue kann geschlossen werden.

For reference (Python3 snippet):

	data = {
	    "cmd": "set_time",
	    "val": ts				#<UNIX_TIMESTAMP_SEC>
	}
	url = 'api/setup'
	return self.run_POST(url = url, data = data)

for appropriate implementations of run_POST().

Karsten
--
GPG  40BE 5B0E C98E 1713 AFA6  5BC0 3BEA AC80 7D4F C89B

closed 2023-05-14 by ncqgm