lumapu/ahoy #921: HM-1500 kann nur per HTML "Serial / Control" Power Limit Command gesteuert werden NICHT per MQTT und Rest API

Newbie-2, 2023-05-02:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Modelname: HM-1500
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [x] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 6.0.9
Github Hash: 15ec6a0 :: 0.6.9

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [x] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Newbie-2, 2023-05-02:
Ich nutze für die MQTT und Rest API Steuerung Node Red.
Das auslesen aller Daten funktioniert.

Nur folgender Datensatz wird NICHT ausgegeben:
<TOPIC>/<INVERTER_NAME_FROM_SETUP>/ch0/**PowerLimit**

Desweiteren kann ich den HM-1500 nur per HTML Oberfläche korrekt in der Leistung beeinflussen, jedoch nicht per REST API, oder MQTT ?

inverter/ctrl/limit/0 300W
inverter/devcontrol/0/11 300

Beide "MQTT Out" Befehle in Red Node führen zu keinem Ergebnis ?

lumapu, 2023-05-02:
das MqTT topic für die Limitierung heißt `inverter/ctrl/power/0`

Newbie-2, 2023-05-02:
Hallo lumapu Danke für den Tip. Leider ist es jetzt zu dunkel, wodurch der Wechselrichter nicht aktiv ist. Ich werde das Morgen ausprobieren. 

Aber warum steht denn dann im:
**https://github.com/lumapu/ahoy/blob/main/User_Manual.md

Example:           inverter/ctrl/limit/0     600W**

lumapu, 2023-05-03:
Entschuldige, ich meinte `limit` und nicht `power`. Hier nochmal der Codeausschnitt dazu:

https://github.com/lumapu/ahoy/blob/7be4ae4536437852d3c95fba07ee8e083ef75aa1/src/publisher/pubMqtt.h#L310-L327

bassbumper, 2023-05-04:
> Aber warum steht denn dann im: https://github.com/lumapu/ahoy/blob/main/User_Manual.md
> 
> Example: **inverter/ctrl/limit/0 600W**

Hi Newbie, 
habe auch einen 1500er und bei mir funktioniert es via MQTT oder API ohne Probleme

hier z.B. via API
        var xhr = new XMLHttpRequest();
        xhr.open("POST", "http://x.x.x.x/api/ctrl", true);
        xhr.setRequestHeader("Content-Type", "application/json");
        var data = JSON.stringify({"id": "0", "cmd": "limit_nonpersistent_absolute", "val":newLimit});
        xhr.send(data);

Newbie-2, 2023-05-04:
Hallo bassbumper, Danke für deinen Tip.
Ist schon komisch, das Dein HM-1500 mit MQTT unter "RED NODE" angesteuert werden kann?
Aber warum hast Du den bei dem API Befehl die id: mit "0" angegeben. Ist doch eine Zahl wie der Watt Wert und müsste doch Ohne die "" somit nur 0 angegeben werden, oder ?
Könntest Du mir vielleicht den Red Node FLOW kopieren und senden ? 
Den könnte ich dann bei mir reinkopieren und vergleichen warum das bei mir nicht funktioniert. :-)

Newbie-2, 2023-05-05:
[
    {
        "id": "5e54fdd4189b37f8",
        "type": "tab",
        "label": "AhoyDTU",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "3258189372f609b6",
        "type": "inject",
        "z": "5e54fdd4189b37f8",
        "name": "{\"id\":0,\"cmd\":\"limit_persistent_absolute\",\"val\":300}",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "{\"id\":0,\"cmd\":\"limit_persistent_absolute\",\"val\":300}",
        "payloadType": "json",
        "x": 250,
        "y": 180,
        "wires": [
            [
                "f53a095c856ee4ad"
            ]
        ]
    },
    {
        "id": "f53a095c856ee4ad",
        "type": "http request",
        "z": "5e54fdd4189b37f8",
"name": "http://192.168.5.200/api/ctrl",
        "method": "POST",
        "ret": "txt",
        "paytoqs": "ignore",
"url": "http://192.168.5.200/api/ctrl",
        "tls": "",
        "persist": false,
        "proxy": "",
        "insecureHTTPParser": false,
        "authType": "",
        "senderr": false,
        "headers": [],
        "x": 640,
        "y": 180,
        "wires": [
            [
                "3ee06d1435658684"
            ]
        ]
    },
    {
        "id": "3ee06d1435658684",
        "type": "debug",
        "z": "5e54fdd4189b37f8",
        "name": "debug 2",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "x": 1000,
        "y": 180,
        "wires": []
    },
    {
        "id": "86a9bcd74176d3e0",
        "type": "mqtt out",
        "z": "5e54fdd4189b37f8",
        "name": "",
        "topic": "inverter/ctrl/limit/0",
        "qos": "",
        "retain": "",
        "respTopic": "",
        "contentType": "",
        "userProps": "",
        "correl": "",
        "expiry": "",
        "broker": "f07c3e81.157df",
        "x": 650,
        "y": 320,
        "wires": []
    },
    {
        "id": "49f82ff8b02a2a91",
        "type": "inject",
        "z": "5e54fdd4189b37f8",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "300",
        "payloadType": "num",
        "x": 350,
        "y": 360,
        "wires": [
            [
                "86a9bcd74176d3e0"
            ]
        ]
    },
    {
        "id": "e213e73f1f65c8b4",
        "type": "inject",
        "z": "5e54fdd4189b37f8",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "200",
        "payloadType": "num",
        "x": 350,
        "y": 400,
        "wires": [
            [
                "86a9bcd74176d3e0"
            ]
        ]
    },
    {
        "id": "3a22396a05027b43",
        "type": "inject",
        "z": "5e54fdd4189b37f8",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "600",
        "payloadType": "num",
        "x": 350,
        "y": 440,
        "wires": [
            [
                "86a9bcd74176d3e0"
            ]
        ]
    },
    {
        "id": "f07c3e81.157df",
        "type": "mqtt-broker",
        "name": "Inverter",
        "broker": "192.168.5.119",
        "port": "1889",
        "clientid": "InverterNodeRed",
        "autoConnect": true,
        "usetls": false,
        "compatmode": false,
        "protocolVersion": "4",
        "keepalive": "60",
        "cleansession": true,
        "birthTopic": "",
        "birthQos": "0",
        "birthPayload": "",
        "birthMsg": {},
        "closeTopic": "",
        "closeQos": "0",
        "closePayload": "",
        "closeMsg": {},
        "willTopic": "",
        "willQos": "0",
        "willPayload": "",
        "willMsg": {},
        "userProps": "",
        "sessionExpiry": ""
    }
]

Newbie-2, 2023-05-05:
Natürlich muss in der NODE "Inverter" die eigene IP Adresse und der Port des MQTT Server angegeben werden.
Und ebenfalls im Flow "HHTP-Request".

Newbie-2, 2023-05-05:
Natürlich muss in der NODE "Inverter" die eigene IP Adresse und der Port des MQTT Server angegeben werden.
Und ebenfalls im Flow  des "HTTP-Request".

closed 2023-05-05 by Newbie-2

PeterOFre, 2023-10-16:
> Hallo Peter,
>
> leider habe ich nicht mehr den Bezug zu Deiner Information.
>
> Aber so wie es in der Erklärung aufgeführt ist bedeutet "nonpersistent" das
>
> es in den flüchtigen temporären Speicher geschrieben wird, was kein Problem
>
> darstellt.
>
> Dagegen bedeutet "persistent" das es in das dauerhafte Flash EEPROM
>
> geschrieben wird, welches nur selten geschrieben werden sollte.
>
> Somit ist "nonpersistent" für das ständige ändern natürlich korrekt und
>
> persistent" sollte nur für die einmalige Begrenzung von 600Watt oder 800Watt
>
> einmalig festgelegt werden.
>
> Liebe Grüße von der Nordseeküste
>
>                                                                 Bernd
>
>
>
>
>
>
>
>
>
>
>
> Control" Power Limit Command gesteuert werden, NICHT per MQTT und Rest API
>
> (Issue #921)
>
>
>
> Hallo Newbie-2, mich wundert, dass Du "persistent" ausgewählt hast, ich
>
> dachte, das führt über kurz oder lang, dazu, dass das EEPROM den Geist
>
> aufgibt.
>
> Deshalb hab ich "nonpersistent" bei mir gewählt.
>
> Aber ich kann mich auch irren :-)
>
> Sonnige Grüsse aus Freiburg
>
> Peter

Hallo Bernd,Danke für die schnelle Antwort, so sehe ich es auch. Komme gerade vom Kandel, das ist ein Berg nahe Freiburg. War eiskalt und mit Sterne fotografieren war nichts. Trotzdem schöne Grüße Peter