lumapu/ahoy #922: Wechselrichter HM1500 is not yet available !?

Piffekopp, 2023-05-02:
## Hardware
  * [X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: _VENDOR ESP8266MOD
Retailer URL: ______

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [X] external antenna (SMA)

### Power Stabilization:
* [X]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [X] AhoyDTU Webinstaller

Hallo Leute,
meine AhoyDTU empfängt nichts von meinen Hoymiles HM1500. 
Habe die Anschlüsse laut Plan nochmal überprüft und durchgemessen. Hier ist mir nichts aufgefallen bzw. das sollte alles richtig angeschlossen sein. Weis mir momentan keinen Rat mehr.

Habt ihr da vielleicht eine Idee an was es liegen könnte ?

An meinem Wechselrichter sind derzeit nur zwei Module auf Eingang 1 + 2 angeschlossen. Die grüne LED am Wechselrichter blinkt alle 2s grün was laut Beschreibung bedeutet, daß der WR Strom erzeugt jedoch gibt es ein Problem mit einem Anschluss. Ich vermute, daß das von den zwei fehlenden Modulen kommt die auf Eingang 3 + 4 nicht angeschlossen sind. Denke mal die fehlende Kommunikation mit der AhoyDTU dürfte damit nichts zu tun haben. Der WR müsste ja trotzdem Daten senden.

LG Sven

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mroenne2022, 2023-05-02:
In den Einstellungen müssen m.E. bei LED0 und LED1 auf off gestellt werden!? Sonst kollidiert das doch Mut dem IRQ Port

Piffekopp, 2023-05-02:
O.K.
Werde es morgen mal probieren. Danke schon mal für die rasche Antwort.

mroenne2022, 2023-05-02:
Ja genau,  kannst das heute schon so einstellen und wirst morgen früh sehen das es dann sehr warscheinlich läuft sobald der WR erwacht ist 🌞👍

Piffekopp, 2023-05-03:
Hey.
Als ich das Bild gesehen habe dachte ich auch schon daran. Ist aber nur Flussmittel vom Lötzinn.

Piffekopp, 2023-05-03:
@mroenne2022 
Danke mein Freund. 👍  Hat tatsächlich daran gelegen. Funktioniert jetzt.
Hatte ja den Web Installer genutzt und bin davon ausgegangen, daß ich da nix mehr einstellen muss. :-/

Gruß Sven

Piffekopp, 2023-05-03:
Davon abgesehen ist das alles noch Neuland für mich.

mroenne2022, 2023-05-03:
@Piffekopp : Super,  ab jetzt wirst du dich nach und einarbeiten,  hier ständig mitlesen hilft sehr 👌

closed 2023-05-03 by Piffekopp