lumapu/ahoy #923: REST API api/record/live liefert nicht alle Daten für HM-300

marseib, 2023-05-03:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

3 inverter added 2xHM-1500 and 1xHM-300
Inverter interval set to 5 seconds

### Error description

Ich bertreibe die Ahoy-DTU mit 3 Invertern. 2xHM1500 und 1xHM300 und greife auf die Daten per REST API zu. Beim HM-300 sind einige Daten des Inverters (z.B. Temperatur) am Endpoint api/record/live nicht verfügbar. Die Ausgabe der Daten endet nach "F_AC". Siehe anbei meine aktuellsten Werte am Endpoint.

"PF_AC","Temp",... sind für den HM-300 (inverter 2) nicht verfügbar, im Gegensatz zu den HM-1500 Invertern. 
Die Werte sind auf der normalen Ahoy-DTU UI Live Seite zu sehen.

Wenn der HM-300 in Zustand "not producing" ist (nachts), dann sind auch die Felder im REST API sichtbar.

Ansonsten funktioniert die Software in Version 0.6.9 super stabil!

```
{"inverter":[[{"fld":"U_DC","unit":"V","val":"33.80"},{"fld":"I_DC","unit":"A","val":"1.00"},{"fld":"P_DC","unit":"W","val":"33.60"},{"fld":"YieldDay","unit":"Wh","val":"62.00"},{"fld":"YieldTotal","unit":"kWh","val":"299.82"},{"fld":"Irradiation","unit":"%","val":"8.51"},{"fld":"U_DC","unit":"V","val":"33.80"},{"fld":"I_DC","unit":"A","val":"0.96"},{"fld":"P_DC","unit":"W","val":"32.40"},{"fld":"YieldDay","unit":"Wh","val":"60.00"},{"fld":"YieldTotal","unit":"kWh","val":"283.53"},{"fld":"Irradiation","unit":"%","val":"8.20"},{"fld":"U_DC","unit":"V","val":"33.90"},{"fld":"I_DC","unit":"A","val":"1.07"},{"fld":"P_DC","unit":"W","val":"36.30"},{"fld":"YieldDay","unit":"Wh","val":"64.00"},{"fld":"YieldTotal","unit":"kWh","val":"315.14"},{"fld":"Irradiation","unit":"%","val":"9.19"},{"fld":"U_DC","unit":"V","val":"33.90"},{"fld":"I_DC","unit":"A","val":"1.07"},{"fld":"P_DC","unit":"W","val":"36.20"},{"fld":"YieldDay","unit":"Wh","val":"64.00"},{"fld":"YieldTotal","unit":"kWh","val":"299.25"},{"fld":"Irradiation","unit":"%","val":"9.16"},{"fld":"U_AC","unit":"V","val":"241.80"},{"fld":"I_AC","unit":"A","val":"0.55"},{"fld":"P_AC","unit":"W","val":"131.70"},
{"fld":"Q_AC","unit":"var","val":"22.80"},{"fld":"F_AC","unit":"Hz","val":"49.96"},{"fld":"PF_AC","unit":"","val":"0.99"},{"fld":"Temp","unit":"°C","val":"15.20"},{"fld":"ALARM_MES_ID","unit":"","val":"1.00"},{"fld":"YieldDay","unit":"Wh","val":"250.00"},{"fld":"YieldTotal","unit":"kWh","val":"1197.74"},{"fld":"P_DC","unit":"W","val":"138.50"},{"fld":"Efficiency","unit":"%","val":"95.09"}],[{"fld":"U_DC","unit":"V","val":"33.80"},{"fld":"I_DC","unit":"A","val":"0.80"},{"fld":"P_DC","unit":"W","val":"27.10"},{"fld":"YieldDay","unit":"Wh","val":"46.00"},{"fld":"YieldTotal","unit":"kWh","val":"4.38"},{"fld":"Irradiation","unit":"%","val":"7.32"},{"fld":"U_DC","unit":"V","val":"33.80"},{"fld":"I_DC","unit":"A","val":"1.00"},{"fld":"P_DC","unit":"W","val":"33.70"},{"fld":"YieldDay","unit":"Wh","val":"60.00"},{"fld":"YieldTotal","unit":"kWh","val":"50.23"},{"fld":"Irradiation","unit":"%","val":"9.11"},{"fld":"U_DC","unit":"V","val":"33.50"},{"fld":"I_DC","unit":"A","val":"0.89"},{"fld":"P_DC","unit":"W","val":"29.90"},{"fld":"YieldDay","unit":"Wh","val":"53.00"},{"fld":"YieldTotal","unit":"kWh","val":"55.15"},{"fld":"Irradiation","unit":"%","val":"8.93"},{"fld":"U_DC","unit":"V","val":"33.50"},{"fld":"I_DC","unit":"A","val":"1.00"},{"fld":"P_DC","unit":"W","val":"33.50"},{"fld":"YieldDay","unit":"Wh","val":"58.00"},{"fld":"YieldTotal","unit":"kWh","val":"61.35"},{"fld":"Irradiation","unit":"%","val":"10.00"},{"fld":"U_AC","unit":"V","val":"238.10"},{"fld":"I_AC","unit":"A","val":"0.50"},{"fld":"P_AC","unit":"W","val":"118.00"},{"fld":"Q_AC","unit":"var","val":"22.10"},{"fld":"F_AC","unit":"Hz","val":"49.96"},{"fld":"PF_AC","unit":"","val":"0.98"},{"fld":"Temp","unit":"°C","val":"15.40"},{"fld":"ALARM_MES_ID","unit":"","val":"1.00"},{"fld":"YieldDay","unit":"Wh","val":"217.00"},{"fld":"YieldTotal","unit":"kWh","val":"171.11"},{"fld":"P_DC","unit":"W","val":"124.20"},{"fld":"Efficiency","unit":"%","val":"95.01"}],[{"fld":"U_DC","unit":"V","val":"34.60"},{"fld":"I_DC","unit":"A","val":"0.75"},{"fld":"P_DC","unit":"W","val":"25.80"},{"fld":"YieldDay","unit":"Wh","val":"45.00"},{"fld":"YieldTotal","unit":"kWh","val":"5.44"},{"fld":"Irradiation","unit":"%","val":"6.97"},{"fld":"U_AC","unit":"V","val":"239.20"},{"fld":"I_AC","unit":"A","val":"0.10"},{"fld":"P_AC","unit":"W","val":"24.70"},{"fld":"Q_AC","unit":"var","val":"0.00"},{"fld":"F_AC","unit":"Hz","val":"49.96"},{"fld":"PF_AC","unit":""}]]}
```

marseib, 2023-05-03:
Hi Lukas, danke für die Info. Das geht so aus der Dokumentation (noch) nicht hervor 😉 .
Es ist aber definitiv die Version 0.6.9. Das Problem tritt am Endpoint `api/record/live` auf und nicht an `api/live`. Soll man `api/record/live` nicht benutzen obwohl er in der Liste unter `api` aufgeführt ist?

lumapu, 2023-05-03:
der Grund für die fragmentierten API requests ist der ESP8266. Genau wie du es beschrieben hast, fehlen bei den ursprünglichen requests oft die Daten am Ende.
Mit dem ESP32 hat der Endpunkt `/api/record/live` seine Berechtigung, da dieser mehr Speicher hat. Der Endpunkt ist eher aus der Einfachheit entstanden, da er schon quasi schon fast da war.

Bitte nutze für deine API requests die oben genannten Endpunkte, die leider noch nicht vollständig dokumentiert sind.

closed 2024-01-02 by lumapu