lumapu/ahoy #924: Ist das mit dieser Lizenz legal?? --> ahoy DTU Cloud

Mart-Muc, 2023-05-04:
Hallo Lukas, hallo Team,

bin gerade über diese Seite "gestolpert", und frage mich nun, ob es moralisch vertretbar und legitim ist, Eure Arbeit kommerziell zu nutzen???
Also ich würde mich ausgenutzt fühlen - oder ist das einer von den Entwicklern hier?

Ich kenn mich aber mit der Gesetzgebung Null aus, deshalb auch nicht der direkte Link:
googelt mal nach: ahoy DTU Cloud

Danke für Eure Arbeit, lg Mart

rmayergfx, 2023-05-04:
Ich weiss nicht warum du nicht den Link direkt hier rein kopiert hast, aber egal. 
Hast du mal genau gelesen was dort angeboten wird? Cloudspeicher:

> Diese Cloud übernimmt als günstige Alternative zur DTU pro das Monitoring, die Speicherung, und Visualisierung der ahoy DTU PV-Daten. Die open DTU wird ebenfalls unterstützt

knickohr, 2023-05-04:
Ich glaube nicht das man dagegen was unternehmen kann oder auch soll. Schließlich bietet der Anbieter ja nur an die Daten aus Ahoy in eine Cloud abzulegen, so wie es viele auch mit HA oder sonstige MQTT Broker machen. Solange er nicht die Software Ahoy auf irgendwelchen DTU vertreibt ist es legal.

Ollipop030, 2023-05-04:
Die vermieten nur eine Datenbank. Die DTU soll man sich ja auch selbst beschaffen, auf deren Homepage ist sogar eine Bauanleitung (inkl. Link zu diesem Repo). Unterstützt meiner Meinung nach sogar dieses Projekt.

Harisjems, 2023-05-07:
Ich würde die Thema hier missbrauchen,ohne neue anzufangen.
Wie kann man die daten Lokal speichern.inkl. Monitoring usw.!?
Synology DS ist vorhanden und paar Rasperrys die sich langweilen.
Über Begriff Grafan,Home Asi usw.bin ich gestolpert,ist des einer von den richtige Ansatz!?

knickohr, 2023-05-07:
MQTT, Grafana, HomeAssistant, Mosquitto, über REST-API auslesen …

Es gibt viele Möglichkeiten.

Harisjems, 2023-05-07:
O.K. Danke

mager33, 2023-05-13:
https://www.pv-argus.de/

mager33, 2023-05-13:
PS: ich würde das NICHT so machen, da sie einen Portweiterleitung zur AhoyDTA/OpenDTU einrichten. Wehe, da ist in eigendeiner Libary eine noch nicht entdeckte Lücke, dann ist das ein prima Einfallstor. Deren Lösung ist ja für Leute ohne techn. Ahnung gedacht - die aktualisieren ihre IOT-devices auch nicht. Und die, die Ahnung haben brauchen sowas eh nicht.

closed 2023-08-05 by lumapu