lumapu/ahoy #925: Sprachen

HorstyS, 2023-05-05:
hi,

ich habe zu dem thema hier noch nichts gefunden und wollte einfach mal nachfragen:

gibt es denn eine intention das UI auch in andere sprachen zu übersetzen? ich würde mein englisch als doch ziemlich gut bezeichnen, aber es gibt hier einige spezielle fachausdrücke die ich mir habe auf deutsch übersetzen müssen, damit ich weiß was die werte bedeuten, bzw. wie ich sie zu interpretieren habe. des weiteren scheint es sich ja hier auch um einen deutschen ableger zu handeln, wenn ich das richtig verstehe ;)

das soll nur eine allgemeine frage sein, und erst mal kein FR/CR. vielleicht habe ich das thema ja an einer anderen stelle übersehen...dann sorry dafür.

gruß

rmayergfx, 2023-05-05:
Das mit den Fachbegriffen und Übersetzungen ist so ein Thema, nimmst du eine passende Übersetzung und machst dich auf die Suche findet man dann leider oft nur sehr wenige deutsche Übersetzungen und Erklärungen zum Fachbegriff, hingegen beim englischen ist das Netz komplett voll. Schaut man sich die dazugehörigen Werte an, sind die Fachbegriffe auch fast schon wieder selbsterklärend.

tqma1, 2023-05-07:
Ich hätte jetzt auf die Schnelle keinen Begriff gefunden bei welchem ich nicht die deutsche Übersetzung kenne.
Beim Schwester-Projekt gibt es ja auch unterschiedliche Spracheinstellungen.

Außerdem muss ich sagen das ich persönlich kein Fan davon bin alles nur noch in Englisch zu haben.
Dann bitte führen wir gleich Englisch als Amtssprache ein und Deutsch ist nur noch ein lokaler Dialekt.

rmayergfx, 2023-05-07:
Sprachdateien belegen aber auch wieder Resourcen, könnte dann beim ESP8266 knapp werden. Oder man hääte wieder für jede Sprache eigene Installer. FInde das jetzt nicht so schlimm das es "nur" auf english verfügbar ist. Wo ist da nun das Problem? Wichtig sind doch das die Werte stimmen und das Teil soweit funktioniert.  Mit einer passenden Auswertung dahinter oder einem Display hat man doch maximal noch beim Update das Webinterface oder nutzt ihr das so intensiv?

HorstyS, 2023-05-08:
das mit den ressourcen verstehe ich und ist vermutlich ein valides argument (ich kenne die kapazitäten der beiden boards nicht).
dass die werte stimmen (da hat Ahoy vermutlich wenig einfluss drauf, außer bei formeln sofern es welche gibt) und es stabil läuft, sind natürlich die hauptkriterien. aber als ich mit dem thema vor ein paar wochen begonnen habe musste ich mich schon ziemlich "reinfuchsen" was denn z.B. "n var" und "Reactive Power" sein sollen. bei den power limits (okay, ich nutze sie nicht) muss ich schon zwei mal überlegen welche option welche auswirkung hat.
ich habe zwei DTUs mit displays, das ist absolut top. aber wenn ich unterwegs bin schaue ich mir die werte eben auf dem handy an. auch wenn ich zu hause im homeoffice bin nutze ich nur das webinterface. also doch ziemlich oft in meinem fall.

ich weiß durchaus, was es bedeutet mehrere sprachen in eine applikation zu implementieren, zu warten und zu prüfen...und übersetzen zu lassen. wir haben das in der firma auch für einige applikationen. daher ist mein post mehr als frage bzw. diskussionsgrundlage zu verstehen :)

rmayergfx, 2023-05-09:
Schieb die täglichen Daten doch per MQTT auf enen Raspi mit HomeAssistant oder Solaranzeige und dann kannst du die Werte jederzeit aufhübschen, d.h. Sprache ändern oder die für dich passenden Informationen aufbereiten. Zudem hast du dann ein Langzeit archiv. Dieses bitte nicht ins Internet lassen, auch nicht AhoyDTU, immer per VPN nach Hause verbinden und dann die Daten ablesen! IOT sind gerne ein Angriffsvektor.

HorstyS, 2023-05-09:
ne....MQTT ist aktuell noch nichts für mich. ich hab das mal "auf die schnelle" versucht und bin gescheitert. vielleicht irgendwann mal wieder wenn ich muse habe ;)
VPN sowieso. das ist mM nach sowieso ein muss wenn man seine daten zu hause lassen will und schützen will.

die übersetzungsfunktion von Chrome ist übrigens nicht schlecht. damit werden die UI strings meiner meinung nach ziemlich gut bezeichnet. das war am anfang so mein kleiner workaround ;)

DanielR92, 2023-08-25:
Zu dem Thema mit Sprachen, das könnte früher oder später Interessant werden. 
Ich gedenke schon seit langer Zeit den Code auf eine "Bibliothek" für alle Texte zu bauen. 

Dies würde uns auch zum Thema größe des Speichers unter die Arme helfen. Da mir aufgefallen ist das es doch oft mit doppelten Wörtern im Code auftauchen. 

@lumapu, du hättest wahrscheinlich nichts dagegen? 
Dann könnte man auch Sprachen mit einbauen, erstmal Deutsch & Englisch.

Da ich mittlerweile mitbekommen habe das es auch welche gibt die in anderen Ländern Ahoy nutzen. :D

lumapu, 2023-08-27:
ich würde die Sprache auf Compiler Ebene anlegen.
Aus meiner Sicht sollte nur das webUI von der veränderten Sprache betroffen sein.

lumapu, 2024-02-17:
ist eingebaut 😊

closed 2024-02-17 by lumapu