lumapu/ahoy #926: [Bug]

sri-elgora, 2023-05-05:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

.

### Error description

Hallo,

Ein Reset des WR funktioniert bei mir nicht. Habe einen HM-1500 und einen HM-700 getestet. Weder über das Web-UI noch über die REST-API lässt sich ein Reset auslösen, die YieldDay-Werte bleiben stehen und die WR produzieren durchgängig weiter. Power Off und Power On funktioniert.

Grüße SRI

rmayergfx, 2023-05-07:
Bitte in Zukunft auch einen passenden Titel für das Thema wählen, mit BUG alleine kann man nicht viel anfangen und dann wird es auch von einigen Leuten wegignoriert. Bitte etwas mehr Mühe geben!

sri-elgora, 2023-05-07:
Sorry, übersehen, geändert.

sri-elgora, 2023-05-08:
Fehler liegt vermutlich hier, RestApi.h Zeile 556:

`            if(F("power") == jsonIn[F("cmd")])
                accepted = iv->setDevControlRequest((jsonIn[F("val")] == 1) ? TurnOn : TurnOff);
            else if(F("restart") == jsonIn[F("**restart**")])
                accepted = iv->setDevControlRequest(Restart);`

closed 2023-08-05 by lumapu