lumapu/ahoy #927: [Bug] MQTT sendet 0 bei YieldDay und YieldTotal

knickohr, 2023-05-05:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.12

### Github Hash

f67a68e

### Build & Flash Method

ESP Tools (flash)

### Setup

Reset values when inverter status is 'not available - disabled

### Error description

MQTT sendet immer eine 0 bei YieldDay und YieldTotal wenn Inverter abends auf offline gehen, egal ob „ Reset values when inverter status is 'not available“ enabled oder disabled ist.

Fehler trat zum ersten Mal bei 0.6.10 auf.

amzrhh, 2023-05-13:
Hi,Experts, my observations are the following:
 YieldDay and YieldTotal are sent by two topics: 1. inverter/total and 2. inverter/tsunny/ch0/ where tsunny is the given inverter name. Those sent by the 1. are 0 when the inverter goes offline. Topic 2. keeps the last values.
rgs amzrhh

knickohr, 2023-05-13:
Thats correct. The second one is send for every inverter, thats OK. But the total Values should not redettet if not selected in the settings.

lumapu, 2023-05-16:
@knickohr für mich klingt das eher nach der `neuen` Art des MqTT publish. Siehst du, ob alle Werte des Inverters nochmal geschrieben werden?
Ich habe gerade den Code geprüft und keinen Punkt gesehen, an dem 'versehentlich' der Reset ausgeführt wird.

knickohr, 2023-05-17:
@lumapu 
Ja, das hat definitiv mit dem neuen zu tun. In 0.6.8 war das ja noch nicht drin. Hat auch sicher was mit dem zweiten Issue zu tun.

Ist  halt doof da dann die Settings wertlos sind.

Stört mich persönlich recht wenig da ich es in Node-Red unterbinde, aber andere werden früher oder später drüber stolpern.

lumapu, 2023-05-17:
ich kann es bei mir nicht nachvollziehen. Ich denke der Inverter geht kurz offline da zu wenig Spannung. Kurz später hat er nochmal genug Spannung, aber leider sind zu diesem Zeitpunkt alle Zähler schon 0. Kannst du dieses Szenario bei dir ausschließen?
Bei allen drei Invertern bei mir sind sowohl die YieldDay als auch die YieldTotal Werte noch vorhanden (jetzt um 23:23 Uhr). Interessant wäre zB. die Temperatur in deinem Fall, bei mir steht sie zB. noch auf 10,9°C und 11.1°C.

knickohr, 2023-05-18:
Dieses blöde Total 😅

Ich meine nicht das Total der einzelnen Inverter sondern das Totale aufsummierte. Scheint auch nur ein Problem beim der Ausgabe von MQTT zu sein, das braune Feld im Webgui macht es nicht.

Ich umschiffe halt den Bug indem ich mir immer den höchsten Wert im Node-Red halte und ihn um Mitternacht selbst resette.

Und ja, ich kann das Szenario ausschließen da es jeden Tag mit jedem Inverter passiert. Wäre schon ein komischer Zufall. Ab und zu geht wirklich mal ein Inverter Off und dann wieder on. Dann ist er aber erst im Status Partial und das würde ich merken.

Wie gesagt, seit dem „Neuen“ sind einige so „Schmankerl“ reingekommen. Hab ja schon die Issues darauf untereinander verlinkt.

Mach Dir erst mal darüber nicht zuviel Kopf. Fällt wahrscheinlich auch nur denjenigen auf die pausenlos auf das Dashboard starren. Im Moment alles noch mit Node-Red vertuschbar 😉

Und es scheint auch nur ein Problem bei mehreren Invertern zu sein. Bei 9 fällt das halt schon gleich heftig auf 😅

knickohr, 2023-05-21:
Ich hab’s gefunden 😲 Offensichtlich donnert MQTT nochmal alle Werte raus wenn der globale Status auf „offline“ geht. Und ich hab mich schon gewundert warum das mal früher oder später passiert.

lumapu, 2023-06-09:
heute getestet und hat funktioniert - was sagt ihr?

knickohr, 2023-06-10:
Geht nicht, der Build ging in die Hose 😲 Außerdem kann ich erst testen wenn es auch wieder mit dem S3 Fusion klappt 😇

knickohr, 2023-06-10:
OK, ich werde es heute Abend sehen 👍

knickohr, 2023-06-11:
👍

knickohr, 2023-06-16:
Ich mach den mal wieder auf, weil das Problem wieder da ist ☹️

[image]

teletapi, 2023-06-16:
ja hab ich jetzt gerade auch mit meinem Issue gemacht habs auch gerade eben bemerkt

MiniOh, 2023-06-18:
Ja, kann ich bestätigen.
Sowohl mit der 0.7.4, also auch 0.7.6 stimmen die Werte in der GUI aber nicht via MQTT.

knickohr, 2023-06-18:
Er sendet jetzt schon wieder eine Null bei YieldDay 😢

[image]

Es ging, solange ein Inverter noch online war. Jetzt wo sie alle im Status 1 sind, geht der Wert wieder auf Null.

teletapi, 2023-06-18:
Ja so ähnlich sieht das bei mir auch aus, YieldTotal scheint aber zu senden aber YieldDay geht wenn die Produktion auf 0 geht ebenfalls auf null
[image]

knickohr, 2023-06-18:
Es geht noch weiter :  YieldDay wird teilweise auch noch falsch berechnet (bei mehreren Inverteer), siehe Issue #929

knickohr, 2023-07-04:
Yeahhh !  Das funktioniert jetzt (endlich) 😎 Am Ende werden auch nochmal schön alle Werte von allen Invertern raus gegeben 👍

Zu das Ding !

closed 2023-07-04 by knickohr

knickohr, 2023-07-05:
Ist mir gestern Abend nicht aufgefallen. Wenn das aber nur einmal kurz kommt, dann kann gut sein das ich es übersehen habe.

knickohr, 2023-07-05:
Hab’s eben nochmal beobachtet. Ich kann zu diesem Issue kein Problem mehr feststellen. Es kommt nur dann eine Null wenn die Inverter auch aus gehen. Nun ja,  es kommt eine Null bei /total/P_AC etwas zu früh, das ist aber Problem vom Issue #929

lumapu, 2023-07-05:
> Zudem habe ich mit der 0.7.7 schon diverse DTU Reboots innerhalb eines Tages gehabt. Ich bin wieder zurück zur aller ersten 0.6.10 HMS, die, zumindest bei mir, am besten läuft, abgesehn davon, dass abends hin und wieder falsche YieldDay liefert.

Reboots beim ESP32? Das hatte ich noch nie

lumapu, 2023-07-05:
@MiniOh bitte einen neuen Issue aufmachen, falls das Problem mit den Yields wie du beschrieben hast nochmal auftreten solte.

MiniOh, 2023-07-06:
> > Zudem habe ich mit der 0.7.7 schon diverse DTU Reboots innerhalb eines Tages gehabt. Ich bin wieder zurück zur aller ersten 0.6.10 HMS, die, zumindest bei mir, am besten läuft, abgesehn davon, dass abends hin und wieder falsche YieldDay liefert.
> 
> Reboots beim ESP32? Das hatte ich noch nie

Ich auch nur mit der 0.7.7 gestern. Da kam ich nicht über 5min hinaus.
Kann es für die Software ein Problem sein, wenn ein Inverter einen Status lliefert, mit dem du nicht rechnest?
Gestern hatte ich einige Male bei dem der NA Schutz greift.
Also Spannung recht hoch, Inverter bleibt online, hört aber auf zu produzieren.
Ich weiß nicht ob dann ein anderer Status übermittelt wird.
In der GUI steht dann, Inverter online aber keine Erzeugung.

lumapu, 2023-07-07:
nein der Inverter Status hat keinen Einfluss auf die Stabilität von Ahoy