lumapu/ahoy #930: Hoymiles Crashes

IngoEF, 2023-05-07:
Running hoymiles modul in RasPi suffering several crashes per day, alway getting two lines like following in logfile:
2023-05-07 19:43:22 CRITICAL: Exception catched: unpack requires a buffer of 2 bytes
2023-05-07 19:43:22 CRITICAL: None

What does it mean and how can I repair?

Am I right in this branch?

IngoEF, 2023-05-09:
we are talking about ahoy/tools/rpi/hoymiles
the software starts with "python -um hoymiles" in the newest RasPi OS.
I have no idea, what more information I could provide.
The software feeds Volkszähler and was running stable since October last year but about two weeks ago reliabilty got lost without doing anything from my side.

rmayergfx, 2023-05-10:
take a new SD Card and make a fresh setup and test again.

lumapu, 2023-07-23:
is the issue still there?

IngoEF, 2023-07-24:
I do not use it anymore

closed 2023-07-24 by IngoEF