lumapu/ahoy #931: [Bug] v0.6.9 doesn't boot on ESP8266 (D1 Mini Pro)

tomquist, 2023-05-08:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

v0.6.9 (non-prometheus)

### Github Hash

15ec6a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Fresh-Install

### Error description

1) I went into the perfectly working Web UI of a AhoyDTU v0.5.66 and uploaded 230419_ahoy_0.6.9_15ec6a0_esp8266.bin as a OTA update
2) It uploaded the firmware & restarted AhoyDTU. However, the device never logged into my Wifi. When resetting the device the LED turns on for a very short moment and then stays off.
3) Used the Web Installer to manually flash the firmware to v0.6.9 again (without erasing settings) with the same result (using console of the Web Installer to see the serial log output shows nothing)
4) Retried flashing the firmware through the Web Installer but this time with erasing settings -> same result
5) Used the [nodemcu-pyflasher](https://github.com/marcelstoer/nodemcu-pyflasher) to flash the firmware again -> same result
6) Downloaded v0.6.0 and flashed it using the nodemcu-pyflasher -> everything starts to work again

It seems v0.6.9 isn't working at all on a D1 Mini Pro.

allesgutewarweg, 2023-05-12:
I had a perfectly running D1 Mini Pro v0.6.9 until i've changed MQTT password-less to password-protected. When i switch back to my password-less broker everything is working fine again. It seems that there's an issue with MQTT and passwords.

Can you confirm this?

tomquist, 2023-05-13:
My MQTT settings were always password-less and I didn't change them to password protected. I also tried flashing 0.6.9 after resetting the device but that didn't help. I wouldn't even be able to change the setting because the device doesn't boot at all with 0.6.9.

Andy811963, 2023-05-14:
I have the same problem.
069 and 0612 not boot.

lumapu, 2023-05-16:
is there no console output at all? What happens more detailed? Does the blue LED flash realy quick?

tomquist, 2023-05-16:
No, there's no console output at all. When the device gets powered on or after it gets reset the blue LED shortly flashes once and then stays off forever. I just reproduced the exact same issue with another AZDelivery D1 Mini Pro I had been laying around and got the exact same issue. Please let me know if I should do more tests.

lumapu, 2023-05-16:
is it possible to install any other ESP project to this ESP and then flash Ahoy again?
Did you checked the erase option during installation with our Webinstaller?

tomquist, 2023-05-16:
The ESP is fine. I can successfully install v0.6.0 and I was able to reproduce the issue on another brand new ESP where I was able to install and boot 0.6.0 but not 0.6.9.
I just tried all the dev builds and found that the last working version is 0.6.8.

tomquist, 2023-05-16:
Here's a list of firmwares I tried:
- 0.6.0: ✅
- 0.6.7: ✅
- 0.6.8: ✅
- 0.6.9: ❌
- 0.6.11: ❌
- 0.6.12: ❌

lumapu, 2023-05-17:
do you know the flash size of the ESP? Have you tried to install the ESP8285 binary - it is for small flash devices. Otherwise you can try to zip the firmware before uploading it via OTA.

allesgutewarweg, 2023-05-18:
I have checked my unit again. Neither reflashing nor reloading the factory settings worked. The chip seemed to be completely broken.

Only a new flash directly from the website made it work again. I suspect there were problems with the file system as some settings were not saved correctly, but now everything is working as expected and has been for several days.

tomquist, 2023-05-18:
> do you know the flash size of the ESP?
It's 4MB ([this one](https://www.azdelivery.de/products/d1-mini-pro))

> Have you tried to install the ESP8285 binary - it is for small flash devices.
Does that work on a ESP8266 chip? I only ever tried the ESP8266 binaries.

> Otherwise you can try to zip the firmware before uploading it via OTA.
In my most recent tests I flashed the firmware directly using the serial port using nodemcu-pyflasher.

lumapu, 2023-05-25:
> Does that work on a ESP8266 chip? I only ever tried the ESP8266 binaries.

I think so - there is no difference except the flash from my sight.

closed 2023-07-23 by lumapu

killah78, 2023-08-31:
As a final comment: Try to flash via NodeMCU with 57k baud and DOUT flash mode. That worked for me on a D1 mini pro with same issues.