lumapu/ahoy #932: [Question] Anybody else have trouble with adding HM-1500 Inverters?

DB1BMN, 2023-05-10:
## Hardware
  * [x] ESP8266

Modelname: WeMos D1 mini Clone
Retailer URL: eBay / AliExpress

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* [x] circuit board

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module

### Version / Git SHA: 
Version: 0.6.9

### Build & Flash Method:
* [x] AhoyDTU Webinstaller

Hi,

has anybody else trouble adding HM-1500 inverters?

Several customers reported me that they couldn't find their HM-1500.
The nRF24L01+ module was checked before on my HM-1200 at home and after returning and worked fine.
250kbps Mode, Pin Configuration, Power Level, Cold-Start etc. was checked several times.
Serial Number 1161x each.
Production (green LED on inverter) seems to work.

My theories so far:
- They have an very old Firmware in the inverter incompatible to Ahoy.
- They have a very new Firmware in the inverter incompatible to Ahoy. Hopefully HM has not changed the communication protocol / routines to stop Our Project.
- They have a counterfeit inverter with different communication or even without a radio module at all.

Can anybody report who observed similar problems? Thanks!

Best Regards, Mark

Argafal, 2023-05-12:
What do you mean with "several of my customers"? Are you aware that Ahoy is distributed under a "no commercial use" license?

DB1BMN, 2023-05-13:
Hi, don't bother, I'm not going to hijack your Project.

I've designed [a PCB](https://github.com/lumapu/ahoy/issues/911)
which is capable to work as a host for hardware needed by Ahoy-DTU but beyond this also offers more additional peripherials.
So in general it is only an experimenting platform wo WeMOS D1 mini-Modules.

I've ordered the PCBs from JLCPCB and populated them.
Some I gave away to my friends or collegues. The rest I sold on eBay for an overall amount not even covering the prototype end engineering samples.

In this way I made some people happy to see what their solar cells are producing who are neither into Programming nor into Electronics so they would never build an Ahoy!-(or what ever)-DTU on their own.

Henning256, 2023-05-26:
I have an HM-1500 and I had the problem that I stopped getting packets after a while using OpenDTU.
Then I switched to AhoyDTU and it was a lot better but still not stable. I upgraded from an ESP8266 to an ESP32 which improved performance (especially when using MQTT) but did not improve the connection.
I added a capacitor to the module and switched back to the one without an external antenna and now it is stable.
Maybe the 3.3V converter on the ESP got unstable because of the power draw of the module...

Suschi64, 2023-05-28:
This issue belongs to Hoymiles HM-1200 inverters too.
Did anybody know how Maxxisun provided inverters handle the 600 watts limitation? They claimed it could be changed only by them (written on there “certificate”).
There are also rumours that the coexistence of 2.4ghz and 5ghz could cause some trouble in communication. Any thoughts on this?
Cheers Arne

rmayergfx, 2023-05-28:
@Suschi64 
plz open a new question about limitation and this firm with certificate. do not forget to place a link also to the website, so everyone knows what you are talking about. 
As soon as I have to search for something, I don't feel like helping others. You should always have these basics in the article, so that it is also clear and does not have to be guessed what is perhaps meant.

closed 2023-07-23 by lumapu