lumapu/ahoy #936: Keine automatische Verbindung zum NTP-Server nach reboot

mueller-esch, 2023-05-12:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

board

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

CC BY-NC-SA 4.0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MqTT ist eingerichtet.
Intervall 30sec

### Debug Serial Log output

```shell
I: (#0) resetPayload
I: (#0) prepareDevInformCmd 0x0b
15 pid: 80
I: TX 27B Ch61 | 15 84 82 28 57 89 95 34 32 80 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 ca a1 96 
I: (#0) nothing received
```

### Error description

Nach einem Reboot startet die Verbindung zum NTP-Server nicht automatisch.
Nur durch drücken von Sync by browser verbidet sich die DTUmit demNTP-Server. Soll das so sein?
Danke im Voraus für ein Feeedback.

rmayergfx, 2023-05-14:
NTP=Zeitserver, das hat nichts mit dem Wechselrichter zu tun. Wenn er sich nicht mit dem NTP vernindet (dazu benötigt er eine Vernindung ins Internet oder er zieht sich die Zeit von einem lokalen Server der bereits im Netz vorhanden ist) würde u.U. die Uhr permanten falsch gehen und das Datum nicht stimmen.

mueller-esch, 2023-05-14:
Hallo, zuerst einmal Danke für dein schnelles Feedback.

Ich glaube ich habe mich in meiner Anfrage nicht klar genug ausgedrückt.

Es geht um folgendes:

Nach einem Stromausfall oder einer Trennung der Ahoy-DTU vom Stromnetz
startet diese ja neu. Sie kann sich aber dann nicht mit dem WR connecten,
solange nicht ein Zeitsignal vorhanden ist. 

Dies bezieht aber meine DTU nicht automatisch, sondern muss immer zuerst
über die Funktion Sync from Browser verbunden werden. 

Erst dann startet die Datensammlung in der LIVE-Seite. Auch das
MqTT-Protokoll startet erst dann.

Ich verstehe es bisher so, dass nach einem Reboot doch auch eine
automatische Verbindung zum NTP-Server stattfinden sollte.

Was meinst du dazu?

 

 

Gruß Paul Müller

reboot (Issue #936)

 

<https://github.com/lumapu/ahoy/issues/936#issuecomment-1546875050> , or
unsubscribe
C5JHANCNFSM6AAAAAAX7FG4ZM> .
<https://github.com/notifications/beacon/AENMKIHGJWRKP64GT7VALVTXGC5JHA5CNFS
M6AAAAAAX7FG4ZOWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTS

rmayergfx, 2023-05-15:
> Dies bezieht aber meine DTU nicht automatisch, sondern muss immer zuerst
> über die Funktion �Sync from Browser� verbunden werden.

Und das ist der Punkt an dem wir eine Lösung benötigen. Warum ist das so? AhoyDTU ist doch im WLAN verbunden, die meisten Router geben doch auch NTP mit in das Netzwerk. 
Geht z.B. bei einer FRITZ!Box recht einfach in den Settings. Dann setzt du einfach die Router IP als NTP Server und schon bist du deine Probleme los.
Alternativ kannst du auch auf deinem WindowsPC/Server/Synology/Linux einen NTP Server aktiv schalten.

mueller-esch, 2023-05-15:
Hallo, das war der Tipp des Tages. Jetzt funktioniert nach einem Reboot alles wie es soll. Vielen Dank für deine Hilfe.

Gruß Paul

Blueey3, 2023-07-15:
> Und das ist der Punkt an dem wir eine Lösung benötigen. Warum ist das so? AhoyDTU ist doch im WLAN verbunden, die meisten Router geben doch auch NTP mit in das Netzwerk. Geht z.B. bei einer [...]

Danke für den Tipp. Hatte das gleiche NTP Problem.

closed 2023-07-22 by lumapu

Blaubaer9, 2024-01-30:
Hallo ich habe das gleiche Problem.
Leider hat das Eintragen der IP-Adresse der Fritzbox nicht geholfen.
Muss ich noch Einstellungen an der Fritzbox vornehmen?
[image]
[image]

lumapu, 2024-01-30:
Port 0 ist glaube ich falsch, es muss Port 123 sein

Blaubaer9, 2024-01-31:
mit Port 123 funktioniert es leider auch nicht :(
(ich muss immer noch über den Browser die Zeit setzen.)

blueline13, 2024-01-31:
@Blaubaer9 
In der Fritzbox muss Der Dienst auch aktiv sein. Zu finden in den Netzwerkeinstellung. [https://avm.de/service/wissensdatenbank/dok/FRITZ-Box-7590/336_Zeitsynchronisation-NTP-fur-FRITZ-Box-und-Netzwerkgerate-einrichten/](url)

Dann mal prüfen ob der Dienst auch wirklich läuft, mit Windows Client wäre es das hier: 
w32tm /stripchart /computer:<IP-Adresse Fritz Box>

Blaubaer9, 2024-01-31:
Danke für den Hinweis,
ich hatte das eigentlich schon eingestellt und jetzt noch einmal überprüft..
[image]

sorry mit "w32tm /stripchart /computer:" kann ich nichts anfangen

blueline13, 2024-01-31:
@Blaubaer9 
War mein Fehler. Da fehlte noch vorher eine "Command Schell" öffnen (Windows Startbutton - ausführen - und dann CMD eingeben und ausführen.) im CMD Fenster dann den richtigen Befehl eingeben: 

w32tm /stripchart /computer:"IP-Adresse der Fritz Box"

Ich hoffe so ist das besser zu verstehen. Sollte die Verbindung nicht klappen dann gibt es einen Fehler. Damit wäre dann sichergestellt das der NTP Dienst auch funktioniert.

Blaubaer9, 2024-01-31:
ok habe ich gemacht und das ist das Ergebnis:
[image]

Die Zeit stimmt allerdings passt irgendetwas nicht.

blueline13, 2024-01-31:
Die Zeit stammt von deinem lokalen Rechner und ist deine IP nicht 192.168.0.1?

Blaubaer9, 2024-01-31:
mmm na klar copy paste falsch ausgeführt

w32tm /stripchart /computer:192.168.0.1
192.168.0.1 wird verfolgt [192.168.0.1:123].
Es ist 31.01.2024 11:48:44.
11:48:44, d:+00.0009713s o:-05.9787924s  [           *               |                           ]
11:48:46, d:+00.0012826s o:-05.9788435s  [           *               |                           ]

blueline13, 2024-01-31:
Jepp so sieht das gut aus und damit sollte die DTU sich die aktuelle Zeit holen. Du hast auch wie @lumapu schrieb den Port auf 123 bei NTP in der DTU gestellt und gespeichert, dann sollte es normal funktionieren. Sonst mal die DTU Stromlos machen und nochmal schauen ob es dann geht. Bitte auch mal den Browser Cache löschen bevor du nach dem Neustart der DTU das erste mal darauf zu greifst. Danach nochmal die Einstellungen prüfen ob alles richtig ist.

Blaubaer9, 2024-01-31:
Den Port habe ich auf 123 gelegt
Manchmal allerdings schafft es die DTU die Tages Wh und die gesamt kWh vom Inverter zu bekommen.
Die statische vergebene IP der DTU funktioniert auch.

Ich bin leider noch nicht weiter gekommen :(

Ich werde mal den ESP tauschen (was auch nichts gebracht hat).

Es funktioniert einfach nicht  :(

lumapu, 2024-01-31:
> ok habe ich gemacht und das ist das Ergebnis:
> [image]
> 
> Die Zeit stimmt allerdings passt irgendetwas nicht.

ich würde die empfehlen die IP zu prüfen 192.16.x.x gehört mWn. nicht zu den Bereichen, die man privat nutzen sollte. Der Bereich von Claas C Netzwerken geht von 192.168.0.0 bis 192.168.255.255
Das soll nur ein Hinweis sein, da ich selbst schon mal Probleme mit verschiedenen Netzwerkgeräten hatte. Mein Bereich war deutlich schlimmer (150.100.7.x), frage nicht was mich getrieben hat.

rmayergfx, 2024-01-31:
Wenn ihr schon eine Fritzbox habt, dann lasst unbedingt bei AhoyDTU das System auf DHCP und vergebt in der Fritzbox die IP und setzt den Haken bei "Diesem Netzwerkgerät immer die gleiche IPv4-Adresse zuweisen".
Ich erlebe es immer wieder das bei der Vergabe der IP Adresse fehler gemacht werden bei Subnetmask und Gateway, was dann dazu führt, das der ntp-Dienst nicht mehr erreichbar ist. Je nach Router darf auch die feste IP nicht im Segment der DHCP Adressen liegen, also wenn der Router es anbietet, setzt es im Router dann funktioniert es auch.

blueline13, 2024-01-31:
@Blaubaer9 
So wie sich das nun anhört scheint das wohl mit dem NTP-Server und der Uhrzeit zu klappen. Ist das so? Wenn nein dann brauchen man mal ein paar Infos. Dann zeige uns doch bitte was du für Netzwerkeinstelungen auf deiner DTU und Fritzbox hast. Ich kann dir auch nur @rmayergfx Hinweise empfehlen.

Blaubaer9, 2024-02-01:
Habe jetzt lange hin und her probiert.
und es scheint so, als wenn sich der ESP fast nie bei der FRITZBOX  anmeldet.
Man kann sich auch nicht  100% zuverlässig über http://192.168.4.1/ in den ESP anmelden, aber irgendwann geht es halt.
Vielen Dank auch an @rmayergfx ich habe das jetzt in der FRITZBOX eingestellt wie du es vorgeschlagen hast.
Die Frage ist jetzt warum sich das Teil weigert sich bei der FRITZBOX  anzumelden.

Ich versuche morgen nochmal einen neuen ESP ob es besser wird.

rmayergfx, 2024-02-02:
Die Frage die sich hier bei mir wieder sofort stellt, ist wie die FB konfiguriert wurde und welches Modell es ist. Bei mir laufen alle IoT im GuestWLAN das auf 2,4Ghz konfiguriert wurde und der Internetzugang ist gesperrt, da ich nur lokal mit den Devices arbeite.
Habe leider schon oft genug erlebt, das wenn 5GHz und 2.4GHz die gleiche SSID nutzen viele Geräte aus dem IoT Bereich keine oder keine zuverlässige Verbindung aufbauen (können) erst mit der Trennung von 5GHz und 2.4 GHz haben diese dann sauber kommuniziert.

Blaubaer9, 2024-02-02:
@rmayergfx  danke für die Hinweise.
Ich habe eine 6490cable FB und habe das 5Ghz abgeschaltet :) und ich hatte auch nur eine feste IP vergeben für meinen Shelly EM (allerdings vom Shelly her was ich jetzt auch ändere und die IP über die FB vergebe)

Habe erst einmal den ESP neu geflashed und dann die Konfiguration(identisch mit den vorherigen Settings) mit dem Handy gemacht. 
Siehe da es funktioniert auf Anhieb(eventuell hat sich mein NB WLAN nicht wirklich mit dem ESP verstanden)

So jetzt funktioniert alles so wie es soll.
Vielen Dank für die vielen hilfreichen Hinweise.