lumapu/ahoy #937: [Bug] nRF-Module not reinitialized to 250 kbps Mode

DB1BMN, 2023-05-12:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Freshly installed Firmware.
New components from box / ESD bag

### Error description

Hi,
since I was investigating a little on https://github.com/lumapu/ahoy/issues/932#issue-1703389690 and checked about a dozen modules
I realized that when I remove the nRF-Module from my PCB the Firmware correctly says under 'System' -> 'Radio' page that "nrf24l01+ is not connected".
When I plug it back it says that is connected but in 2 MBps mode ant stays so thus communication with the inverter is no more possible until you reboot.

Please, could you check before communication to inverter if correct Datarate / Speed is set correctly and readjust it?

I'think this could prevent some communication errors some people expirienced when conntecting their modules with wobbly jumper cables etc.

Thanks, Mark
[image]

Argafal, 2023-05-12:
I have only seen the 2MBps mode when the SPI Pins weren't configured correctly. Do you have a screenshot of your configuration and some pictures of your hardware setup?

DB1BMN, 2023-05-13:
Hi,
thank you for your quick answer.
Not to misunderstand me!
All tested nRF-Modules work so far, *when* in 250 kbps Mode after Booting. If put in my inverters (HM-1200) S/N they easily connect to it.
But if I remove the nRF Module from PCB and plug it (or an other) back then the DTU Firmware sticks to 2 Mbps mode and I have to reboot.
You can believe me that the PCB is correct: 13 x 100 nF MLCC in total at 3V3 and 5 V + 100µF at 3v3 directly near to the nRF. GND and 3V3 over planes at top and bottom.

If you like I can also catch the SPI signals with a Scope when I'm later in the lab.

Please find attached the screenshots of the DTU as well as the relevant schematic diagrams.

[image]

[image]

[image]

[image]

[image]

[image]
[image]

rmayergfx, 2023-05-14:
Again to your 2MBps problem.
Should not be the Problem if you power off during the replacement.

lumapu, 2023-05-17:
@DB1BMN is the problem still existing?

closed 2023-07-23 by lumapu

DB1BMN, 2024-06-01:
Hallo, ich möchte noch mal auf das Problem zurück kommen, da es sicher anderen Leuten auch nützt.

Ich habe bemerkt, dass mittlerweile (V 0.8.122) erkannt wird, ob das nRF-Modul angesteckt ist oder nicht, durch ein gestricheltes Antennen-Symbol auf der linken Display-Seite und im "System"-Menü mit einem roten "not connected". Das ist gut!

Nach anstecken, wird das nRF-Modul wieder erkannt und im Display erscheint die vollständige Antenne und das grüne "connected"-Symbol im Menü. Leider wird die Datenrate nicht erneut initialisiert, sondern verbleibt auf 2 Mbps.
[image]

Ich denke, es gibt genügend Leute, die ihre DTU immer noch mit Jumper-Kabeln zusammengesteckt haben und bei einem Wackelkontakt sich vom nRF-Modul aussperren, obwohl es irgendwann wieder kommt, aber mit der falschen Datenrate.

Darum mein Vorschlag/Bitte: Wenn das nRF-Modul erneut gefunden wird (die Firmware merkt es jae eh, um die korrekten Symbole anzuzeigen) dann auch noch mal neu auf 250 kbps initialisieren.

rmayergfx, 2024-06-05:
Nochmals zur Info, das ist ein Workaround für ein Problem das so eigentlich nicht existieren dürfteund seltenst im laufenden Betrieb passiert. Die Module sind wie bereits oben beschrieben nicht HotPlug fähig und dementsprechend nicht dafür gedacht im laufenden Betrieb ausgetauscht zu werden! Immer ausschalten. Eine Erkennung und Reinitalisierung des Modules bedeutet wieder unnötige Routinen die permanent mit im Hintergrund laufen.