lumapu/ahoy #938: [Question] How to connect Ahoy! to Home Assistant? (For Dummies please!)

DB1BMN, 2023-05-12:
## Hardware
  * [X] ESP8266
  * [X] Raspberry Pi

### Version / Git SHA: 
Version: 0.6.9
Github Hash: 15ec6a0

### Build & Flash Method:
* [X] AhoyDTU Webinstaller

Hello,

I'm trying to connect Ahoy!-DTU to Home Assistant because I want to plot my solar data.
I've freshly installed Home Assistant 2023.5.2 on my Rasberry Pi 4 (4 GB) following [this instruction](https://www.youtube.com/watch?v=KKj9V-tbLPY) but it seems not fo find my Ahoy-DTU :-(
Both are of course in the same network. The rasberry is connectet via RJ45 cable to a switch which is contected to a Speedport WLAN router. The Ahoy-DTU is logged in to the same WLAN and its Web-UI is visible in the Browser.

But when I click at 'Settings' -> 'MQTT' -> 'Discovery Config (homeassistant)' on 'SEND' it says 'command excuted' but I can not find it in the Home Assistant Window. It shows me approx 20 different other devices that *I did not add* and rather would like not to see in HA but the Ahoy-DTU is not shown :-(

[image]

Please, can you explain me how to add the Ahoy-DTU and plot the diagrams? In Englisch or German - or Polish ;-)
I'm rather the Electronics Guy but when it comes to Informatics I habe two left hands with ten left thumbs...

Thanks in Advance,
Mark

P.S. any idea why my both WLAN-Radios are shown twice each and my both Synology NASes even 4 + 5 times?

rmayergfx, 2023-05-13:
Hallo Mark,
du benötigst zwindend die MQTT integration:
[image]

Hast du auch schon einen MQTT Broker auf dem RasPi laufen, wenn nein, den benötigts du auch noch!

In Ahoy DTU setzt du dann in den Settings deine persönlichen Vorlieben im Bereich MQTT:
[image]

Damit du auch gleich prüfen kannst, ob die Kommunikation mit MQTT sauber läuft, hole dir den http://mqtt-explorer.com/ damit kannst du gleich prüfen ob eine Kommunikation überhaupt stattfindet.
[image]

Zur leichteren config empehle ich noch ein
`mqtt: !include mqtt.yaml`

in der configuration.yaml

mit den Settings die du  mit dem MQTT Explorer auslesen kannst, z.B.
`# Loads default set of integrations. Do not remove.
 sensor:
 - name: "Invertername Wifi"
   unique_id:  inverterwifi
   state_topic: "nameausdenMQTTSettingsAhoyDTU/wifi_rssi"     
   icon: mdi:wifi`

rmayergfx, 2023-05-13:
Since in their other issue:
https://github.com/lumapu/ahoy/issues/932
it was already written that they earn money with AhoyDTU, it would be more than unfair to also create a tutorial for dummies for using, so that you can also sell this to your customers.

mager33, 2023-05-13:
AhoyDTU is a MQTT device, so you need the MQTT integration. Clicking "Discovery" will trigger MQTT discovery in HA.

DB1BMN, 2023-05-13:
LOL! Whats the Problem anyhow?

Yes, I've created a PCB and ordered it from JLCPCB and populated it.
Some I gave to my friends and collegues the rest I sold on eBay.
To make the first Prototype working I spent approx. 500 €. The "earnings" from eBay are approx 200 € so my "winnings" are in the **-**300 € region. - But I made a few people happy at last.

As far as I eliminate the mechanical Issiues (Have you ever tried to understand the "drawing" of the 1.3" display? - Good Luck putting it to a layout! - Not no mention that there are at least two versions out there with swapped Vdd/GND-Terminals and/or shiftet mounting holes) I will publish the KiCad Project to GitHub as mentioned [here ](https://github.com/lumapu/ahoy/issues/911)

Besides this I did also report some Bugs and made suggestions how to fix them. In my eyes also a valuable contribution to the Project and I also offered my help when it comes to measurements on the hardware. 
I'm so sorry, that I'm not so deep into the programming stuff! - I'm just a thumb Electronics Engineer obviously to stupid to click on a blue button.

So, please could we now focus back again on technical problems and please would you like to help me, how to connect the Ahoy-DTU to Home Assistant? - I've heard I need something like an IO-Broker, but I can not find it in HA.

DB1BMN, 2023-05-13:
> AhoyDTU is a MQTT device, so you need the MQTT integration. Clicking "Discovery" will trigger MQTT discovery in HA.

Hi Thanks!
Thats exactly what I'm trying.

I have 2 Browser Windows side-by-side. One with the Ahoy MQTT-Panel where I click on "Discovery Config (homeassistant) SEND"  (it says then "command excuted").
In the other other Window I have the Home Assistand listed all(?) devices but the Ahoy-DTU is not listed. Very strange.

rmayergfx, 2023-05-13:
HomeAssistant has still bugs to find MQTT Devices, i don´t know why, sometimes it works, sometimes not and when i discovered the devices, it doesn´t integrate all available values. Thats why i integrated a mqtt.yaml with manual settings, this will force to use the data after you reboot HA. 
Nice benefit, it will not show you "unavailable" on the Fields, if the WR ist going to sleep on sunset :-)

DB1BMN, 2023-05-13:
Hi OK, jetzt auf Deutsch, ist eh einfacher für mich ;-)

Danke, dass wir uns so schnell wieder "vertragen" konnten.
Wie gesagt, ich wollte euer Projekt nicht kapern, sondern ganz im Gegenteil ich wollte das Projekt für möglichst viele Leute zugänglich machen, indem ich eine zuverlässige und einfach zu handhabende Hardware entwickle und vorstelle. - Gibt halt eben viele Leute, die nicht löten können oder wollen oder denen so ein gedrucktes Gehäuse nicht zusagt.
Der Aufbau mit den Jumper-Kabeln ist ja auch nicht so zuverlässig auf Dauer, wie man mit der [fehlerhaften Re-Iinitialisierung](https://github.com/lumapu/ahoy/issues/937) sieht (BTW: Scope-Bilder sind in Vorbereitung).

Back on Topic:
- Also der Raspberry ist frisch aufgesetzt mittels RPi Imager auf folgende Version(en):
  -  Home Assistant 2023.5.2 
  - Supervisor 2023.04.1 
  - Operating System 10.1
  - Frontend 20230503.3 - latest 

Ich habe fast befürchtet, dass ich noch Sachen nachinstallieren muss, aber die Tutorials sind entweder so veraltet, dass man sich nicht mehr zurechtfindet in der Oberfläche oder setzen auf fortgeschrittenes Vorwissen auf, das ich leider (noch) nicht habe.

Danke für die Screenshots! Doch schon beim ersten hapert es bei mir.
Wenn ich auf der Startseite in das Suchfeld ""MQTT eingebe, findet es das nicht.

rmayergfx, 2023-05-13:
MQTT muss im HA nachinstalliert werden:
Einstellungen > Geräte & Dienste dort unten rechts auf den blauen "+ INTEGRATION HINZUFÜGEN"
Seutp:
https://www.youtube.com/watch?v=1FLY0n0Aa-Q

Das mit der Platine ist zwar nett, aber es gibt leider so viele verschiedene Optionen und die Hersteller haben alle unterschiedliche Maße das es aus meiner Sicht keinen Sinn macht im großen Stil da eine Platine zu entwerfen oder zu vertreiben. Ja mit Platine ist das kompakter, aber leider auch nicht fehlerfrei. Hatte selbst schon Platinen in der Hand die n.i.O. waren da die Durchkontaktierung mies war und nachgearbeitet werden musst.
Nach wie vor das größte Manko aus meiner sicht ist, das jedes Board und jedes Wlan Modul unterschiedliche Maße haben, mal sitzt die Bohrung anders, dann wurden beim Display VCC und GND vertauscht wo man sich nur an den Kopf fasst und denkt, hey ihr seid doch Hersteller, wieso macht ihr beim kleinen Display die Anschlüsse anders als beim größeren Display, da ist der Ärger doch vorprogrammiert.

mattt3, 2023-05-16:
> HomeAssistant has still bugs to find MQTT Devices, i don´t know why, sometimes it works, sometimes not and when i discovered the devices, it doesn´t integrate all available values. Thats why i [...]

could you supply your mqtt.yaml for download, please?

rmayergfx, 2023-05-18:
no, as this has my private Settings i don´t wan´t to share, see a few posts above the sample!

austria89, 2023-05-31:
Here are samples I've created for my HM800.
example topic path: `mqtt/solar/inverters/hm800`
sample inverter name (configured in Ahoy DTU settings): `Hoymiles-HM800`

configuration.yaml: (add line)
`mqtt: !include mqtt.yaml # Hoymiles-HM800`

mqtt.yaml: (add new file in same directory as configuration.yaml if not existing yet)
```
sensor:
    # General data
  - name: "Hoymiles-HM800 Status"
    unique_id: inverter_status
    icon: mdi:information-outline
    state_topic: "mqtt/solar/inverters/hm800/status"
  - name: "Hoymiles-HM800 MQTT Status"
    unique_id: inverter_mqttstatus
    icon: mdi:information-outline
    state_topic: "mqtt/solar/inverters/hm800/mqtt"
  - name: "Hoymiles-HM800 Uptime"
    unique_id: inverter_uptime
    unit_of_measurement: "s"
    icon: mdi:clock-time-eight-outline
    state_topic: "mqtt/solar/inverters/hm800/uptime"
  - name: "Hoymiles-HM800 Sunrise"
    unique_id: inverter_sunrise
    state_topic: "mqtt/solar/inverters/hm800/sunrise"
    icon: mdi:weather-sunset-up
  - name: "Hoymiles-HM800 Sunset"
    unique_id: inverter_sunset
    icon: mdi:weather-sunset-down
    state_topic: "mqtt/solar/inverters/hm800/sunset"
    
    # Channel 0 - aggregated inverter data
  - name: "Hoymiles-HM800 WiFi"
    unique_id: inverter_wifi
    state_topic: "mqtt/solar/inverters/hm800/wifi_rssi"
    icon: mdi:wifi
  - name: "Hoymiles-HM800 Channel 0 - Voltage"
    icon: mdi:power-socket-de
    unit_of_measurement: "V"
    unique_id: inverter_ch0_voltage
    state_topic: "mqtt/solar/inverters/hm800/Hoymiles-HM800/ch0/U_AC"
  - name: "Hoymiles-HM800 Channel 0 - Current"
    icon: mdi:current-dc
    unit_of_measurement: "A"
    unique_id: inverter_ch0_current
    state_topic: "mqtt/solar/inverters/hm800/Hoymiles-HM800/ch0/I_AC"
  - name: "Hoymiles-HM800 Channel 0 - Power"
    icon: mdi:solar-power
    unit_of_measurement: "W"
    unique_id: inverter_ch0_power
    state_topic: "mqtt/solar/inverters/hm800/Hoymiles-HM800/ch0/P_AC"
  - name: "Hoymiles-HM800 Channel 0 - Reactive Power"
    unit_of_measurement: "var"
    unique_id: inverter_ch0_reactivepower
    state_topic: "mqtt/solar/inverters/hm800/Hoymiles-HM800/ch0/Q_AC"
  - name: "Hoymiles-HM800 Channel 0 - Frequency"
    icon: mdi:sine-wave
    unit_of_measurement: "Hz"
    unique_id: inverter_ch0_frequency
    state_topic: "mqtt/solar/inverters/hm800/Hoymiles-HM800/ch0/F_AC"
  - name: "Hoymiles-HM800 Channel 0 - Power Factor"
    unique_id: inverter_ch0_powerfactor
    state_topic: "mqtt/solar/inverters/hm800/Hoymiles-HM800/ch0/PF_AC"
  - name: "Hoymiles-HM800 Channel 0 - Temperature"
    icon: mdi:temperature-celsius
    unit_of_measurement: "°C"
    unique_id: inverter_ch0_temperature
    state_topic: "mqtt/solar/inverters/hm800/Hoymiles-HM800/ch0/Temp"
[... 65 lines cut ...]
    icon: mdi:current-dc
    unit_of_measurement: "A"
    unique_id: inverter_ch2_current
    state_topic: "mqtt/solar/inverters/hm800/Hoymiles-HM800/ch2/I_DC"
  - name: "Hoymiles-HM800 Channel 2 - Power"
    icon: mdi:solar-power
    unit_of_measurement: "W"
    unique_id: inverter_ch2_power
    state_topic: "mqtt/solar/inverters/hm800/Hoymiles-HM800/ch2/P_DC"
  - name: "Hoymiles-HM800 Channel 2 - YieldDay"
    icon: mdi:sigma
    unit_of_measurement: "Wh"
    unique_id: inverter_ch2_yieldday
    state_topic: "mqtt/solar/inverters/hm800/Hoymiles-HM800/ch2/YieldDay"
  - name: "Hoymiles-HM800 Channel 2 - YieldTotal"
    icon: mdi:sigma
    unit_of_measurement: "kWh"
    unique_id: inverter_ch2_yieldtotal
    state_topic: "mqtt/solar/inverters/hm800/Hoymiles-HM800/ch2/YieldTotal"
  - name: "Hoymiles-HM800 Channel 2 - Irradiation"
    icon: mdi:solar-power-variant
    unit_of_measurement: "%"
    unique_id: inverter_ch2_irradiation
    state_topic: "mqtt/solar/inverters/hm800/Hoymiles-HM800/ch1/Irradiation"
```

mattt3, 2023-06-02:
Thank you for the effort. Very helpful

austria89, 2023-06-02:
adding
`device_class: power`
as attribute to the power sensors, e.g.
```
  - name: "Hoymiles-HM800 Channel 0 - Power"
    device_class: power
    icon: mdi:solar-power
    unit_of_measurement: "W"
    unique_id: inverter_ch0_power
```

gets rid of the warning message when adding the sensor to the Home Assistant energy dashboard.

this, btw, is done by creating a new helper: Integration - **Riemann sum integral sensor** (easy to find with Google) - that uses Channel 0 Power as input sensor (trapezoidal rule). 
In the end it looks something like this:
[image]

kmarek1, 2023-07-07:
adding
```
  - name: "Hoymiles-HM800 Channel 1 - YieldTotal"
    state_class: total
    device_class: energy 
    icon: mdi:solar-power-variant
    unit_of_measurement: "kWh"
    unique_id: inverter_ch1_yieldtotal
    state_topic: "mqtt/solar/inverters/hm800/Hoymiles-HM800/ch1/YieldTotal"
```
You can add the sensor to the Home Assistant energy dashboard.

closed 2023-08-05 by lumapu