lumapu/ahoy #939: ! Important ! Official information on the removal of the restriction on the performance of TSUN WR !

rmayergfx, 2023-05-13:
Habe vor einigen Tagen TSUN eine Anfrage gestellt, ob und wie es möglich ist, nach der hoffentlich bald kommenden Gesetzesänderung die bereits gekauften WR offiziell upzugraden, darauf habe ich heute diese Antwort erhalten:

Hallo ****,

ja, es ist möglich. Aber die Entdrosselung ist nur gültig für Generation 3. (Anfang der SN mit "R")

Wenn gemäß des Gesetzes in Deutschland die Brenzung auf 600 W auf 800 W entdrosselt wird, könnten Sie direkt mit der Talent Home APP Update-Service für Ihr Gen 3 Produkt bezahlen. Es kostet 24 Euro.

Mit besten Grüßen,

********
Marketing Specialist/Marketing Spezialistin
TSUNESS GmbH
Website: https://www.tsun-ess.com/

stefan123t, 2023-12-21:
Possible duplicate of #909

rmayergfx, 2023-12-22:
@stefan123t 
Can you plz stop on this topic? This is not a duplicate as it was an official statement posted on TSUN homepage, soon after release of the new inverterst! it was just a documentation and information for all users here on git to know what is going on.

stefan123t, 2024-10-27:
I marked this issue as a possible duplicate of #909 because we currently do not support TSUN inverters of the third generation, i.e. Serial numbers with 16 alpha-numeric characters starting with an `R`.

@rmayergfx this issue is now almost one year old, you yourself added a reference to an other issue regarding the Serialnumber #1018. brunsheinz got a replacement according to #950 and #1082 is closed as will not be implemented.

So do we have to keep this open as an issue for AhoyDTU ? You could simply add it to the documentation / wiki if this is necessary.