lumapu/ahoy #940: No Rx from Inverter, please help

walter-in-space, 2023-05-13:
## Hardware
ESP8266 D1 Mini Clone from Amzn
Inverter: HM-800

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [?] **square dot** indicates original Nordic Semicon chip 
* [?] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board

### Power Stabilization:
* [x]  330uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [-] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.6.9
Github Hash: 16ec6a0

### Build & Flash Method:
* [x] AhoyDTU Webinstaller

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Module seems to be online, but Never received a single byte.
250 kBit 
I tried four equal RF-modules, three in the picture. No success.
Please help, what can I test?

[image]
[image]
[image]
[image]

rmayergfx, 2023-05-13:
Please show us the settings from System Config.
Are both LED to set off/default?
SCLK/MOSI/MISO with valid data or 0?

walter-in-space, 2023-05-13:
All left default.
No extra LEDs connected.

[image]

rmayergfx, 2023-05-13:
Plz check wiring and GPIO like described here:
https://github.com/lumapu/ahoy/issues/36

kkro, 2023-05-13:
As #36 says: 
Hardware: Changed D4/D3 and in software config too.
"Inverter #0: HM-800 (v0) is available and is producing"
**It works.**

Thank you very much!

May be a little hint on the main-page could help others.

walter-in-space, 2023-05-13:
Solution is #36 

Thanks to rmayergfx

closed 2023-05-13 by walter-in-space