lumapu/ahoy #941: No wiring scheme for "NRF24L01+ with antenna"

KurtPfeifle, 2023-05-13:
## Hardware
  * [+] ESP8266

Modelname: "Model Vendor"
Retailer URL: ?

### nRF24L01+ Module
* [+] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [?] **square dot** indicates original Nordic Semicon chip 
* [?] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [-] circuit board
* [+] external antenna (SMA)

The AhoyDTU website (German version) lists two different NRF24L01+ boards which may be used:

1. NRF24L01+ ***without*** external antenna
2. NRF24L01+ ***with*** an external antenna 

The wiring diagram then only shows the wiring for the NRF24L01+ version ***without*** the antenna:
[image]

I am missing the wiring for the version ***with*** external antenna. The issue is that this SMD board does not have any identifiers printed on it such as the other one. So I'm quite lost...

[image]

rmayergfx, 2023-05-13:
Strange, could possible be a fake module. Where did you buy it? No description on the seller site with a picture?
Normally should work like this: https://hallroad.org/images/thumbnails/512/512/detailed/37/HW-237_2.4GHz_NRF24L01_PA_LNA_SMA_Wireless_Transceiver_Module_With_Antenna_2.jpg
Very suspect Modul as ther is not print around the parts.

KurtPfeifle, 2023-05-13:
Thank you for your reply.

> Strange, could possible be a fake module. Where did you buy it? No description on the seller site with a picture? Normally should work like this: https://hallroad.org/images/thumbnails/512/512/detailed/37/HW-237_2.4GHz_NRF24L01_PA_LNA_SMA_Wireless_Transceiver_Module_With_Antenna_2.jpg Very suspect Modul as there is not print around the parts.

I bought the module from the the eBay link provided on the AhoyDTU.de website (the makershop.de had it sold out). I do not think it is a fake one, since eBay says the provider has sold 2100+ of these cheap (3,xx€) boards already.

No, there was no detailed description or hi-res pic on the seller site. I didn't find any hi-res pics on the wider internet either.

The one you provided in your answer isn't much better -- however, the breakout sub-pic is somewhat helpful since it annotates some IDs for the pins.

>  NRF24L01+ should have all the same pinout, equal what kind of antenna type.

This is just a guess, right? However, if guessed or not: it's still the question, from which side of the board you look at it? The long pins (for the jumper wires) are on the backside. So is the backside to be represented by your sub-pic? The sub-pic provides these IDs:

- GND (clear), ground
- VCC
- CE
- CSN
- SCK
- MISO
- IRQ

What do they mean? How are these to be connected to the pins on the ESP8266?

There the pin IDs to be used for AhoyDTU are given as:
- G (ground?)
- D4
- D3
- 3V3
- D8
- D7
- D6
- D5

As you can see, there is no match in the IDs. It may be obvious to electronics pros, but it isn't for a rookie like me.

Independent of the question wether I have a fake hardware module, the question still remains and isn't answered by the AhoyDTU website or any docu I found:

**How is the wiring to be arranged for the *NRF24L01+ PA LNA SMA with Antenna* variant?**

Solarteur, 2023-05-14:
The wiring for nRF24RF01+ with printed antenna and external antenna is identical. I used both on **one** board in exchange.

rmayergfx, 2023-05-14:
@KurtPfeifle 

> You may notice on the website (https://ahoydtu.de/getting_started/) that the pic there doesn't show any labels on the SMD board either. This could be the indication that there indeed isn't any on the original board.

No need for this at is totaly clear how to connect. On the picture you see WEMOS D1 mini with all necessary wiring and also the  NRF24L01+, as they all have the same wiring there is nothin unclear at this Point. You see on the picture that all components are visible, so it is very easy to know the right directions and pins. Just take your  NRF24L01+ and place it so you can see the soldered components and you´re right.

DanielR92, 2023-08-25:
Solarteur already mentioned it.

closed 2023-08-25 by DanielR92