lumapu/ahoy #942: Integrate Serial to Net function (Serial2Net)

Tool-use-today, 2023-05-13:
## Hardware
  * [X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: Wemos mini
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [x] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Would it possible to add  a serial to net function to the firmware? 

Like [https://github.com/soif/Serial2Net_ESP8266](url)

Would make it possible to use it on hoymiles inverters while talking with serial devices as well (i.e. modbus adapters for chargers)

lumapu, 2023-05-16:
Please think of the feature again. Is it useful for any other user out there or is it realy specific to your szenario?
My idea is to build a bridge between all generations of Hoymiles inverters to Wifi / LAN.
Maybe you can encapsulate your idea into some kind of `plugin` and add another compile option to `platformio.ini`.

stefan123t, 2023-12-21:
@Tool-use-today what is the idea of a Serial 2 Network adapter for Ahoy DTU ?
You can either use the project Serial2Net on an ESP8266 and hook a modbus adapter via Wireless Network to your system.
But why would you need that option in Ahoy DTU ?

Tool-use-today, 2023-12-22:
The idea behind that was to use just one ESP for controlling the Hoymiles Inverters and a modbus controlled solar battery charger at the same time. I am setting up a selfmade battery solar system with an EPEVER MPPT Charger and want to use that with a controlled Hoymiles Inverter for a zero Grid feeding systems. Till today i use two ESP. One for the Serial2Net combined with a modbus to serial adapter and the other for the Hoymiles.

But, as lumapu answered before: the Ahoy DTU should maybe remain a clean communication brige for the Hoymiles Inverters