lumapu/ahoy #943: [Bug]

juvi007, 2023-05-13:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT with 0s Inervall

### Error description

when Inverse stops receiving Data ist sets the Home Assistant Entries to Unavailable wich is displayed in my Dashboard but I think it should display 0.
[image]

rmayergfx, 2023-05-14:
@juvi007 
This is totally normal with this MQTT Settings. you can change this by adding the sensort in a mqtt.yaml and then include this in your configuration.yaml. Then the values will stay at the last received and not change to unavailable until the System is online again.

ottelo9, 2023-06-20:
@rmayergfx 
Doesnt work for me. I tried it with this config:

mqtt.yaml
`sensor:
  - name: "Leistung Schuppen (800W)"
    unique_id: hm800_schuppen_p_ac
    state_topic: "inverter/HM800-Schuppen/ch0/P_AC"
  - name: "Leistung Carport (800W)"
    unique_id: hm800_carport_p_ac
    state_topic: "inverter/HM800-Carport/ch0/P_AC"`

ottelo9, 2023-06-27:
@rmayergfx Can you help me with this problem?

rmayergfx, 2023-06-27:
Which version of HA is running? Raspberry, Docker? How is MQTT enabled. It seems to me that the system tries to get valid values but the system is not answering. This will lead in "unavailable". 

> when Inverse stops receiving Data ist sets the Home Assistant Entries to Unavailable wich is displayed in my Dashboard but I think it should display 0.

No, Zero (0) is no option, as it could be a different kind of data send by MQTT, so unavailable is right.

ottelo9, 2023-06-27:
@rmayergfx 
Running on a RPi4. I use the Mosquitto broker. I‘ve lot of these unavailable gaps because the signal quality is quite poor. But I cannot change this.

rmayergfx, 2023-06-28:
RPi4 ist connected with Wireless or Ethernet? If Wireless, plz do a change that RPi is only connected to 2.4GHz, this is more stable also for longer distance than 5GHz. Disable 5GHz and try again.

ottelo9, 2023-06-29:
> RPi4 ist connected with Wireless or Ethernet? If Wireless, plz do a change that RPi is only connected to 2.4GHz, this is more stable also for longer distance than 5GHz. Disable 5GHz and try again.

No it’s connected over Ethernet.

rmayergfx, 2023-06-29:
What kind of nRF24L01+ is attached? With external antenna? Is a 100μF Elko applied to stabilize the WLAN Modul? How big is the distance between HM800 and AhoyDTU?

ottelo9, 2023-06-29:
nRF24L01+ With ext antenna. 100uF not applied. I talk to 2 PV inverter. I have only with one of them problems. But this is actually not my question. The question was how to configure mqtt in HA correctly, that in case of one inverter is getting not available HA don’t touch the values (old values).

rmayergfx, 2023-06-30:
That is a good question. 
Hello @lumapu, if the second inverter is going offline, what state is send with mqtt? Is the status unavailable? if so is it possible to send offline or Zero? Last value seems wrong to me, as the system behind will then add wrong values. Offline or Zero should be better at this point.

How is the setup done with the second inverter and why is the system going offline?

ottelo9, 2023-06-30:
Actually its correct in HA, the inverter values are all unavailable if the AHOY DTU cant get an answer from it. But because I have some connection problems at some time I want to set the time when HA detects that the values of the mqtt member is old and so unavailable. I think there is a config / flag that I can set in MQTT to set up when a value is flagged as old/unavailable.

ottelo9, 2023-08-06:
Why closed?

lumapu, 2023-08-06:
I thought that the newest release `0.7.25` fixes this issue. Have you tested this build?

ottelo9, 2023-08-06:
I will give it a try. Thx for answer.

ottelo9, 2023-08-18:
@lumapu 
Sorry I forget to answer. I have AhoyDTU cbdb150 :: 0.7.26 and still have unavailable entities when inverter not responding after few minutes. I haven’t found any setting for that in ahoy or home assistant.

stefan123t, 2024-10-27:
@juvi007 @ottelo9 can you double check whether this issue is still pending with v0.8.152 ?

ottelo9, 2024-10-27:
I moved to opendtu-onbattery

juvi007, 2024-10-28:
@stefan123t Yes everything is working as intended, thanks for the Help

closed 2024-10-28 by juvi007