lumapu/ahoy #946: feature suggestion: assume daylight when NTP unavailable

ncqgm, 2023-05-14:
May I suggest that

- upon startup
- if NTP is unavailable

AhoyDTU should attempt to retrieve data from inverters at least once,
_regardless_ of night-communication-disabled.

If data is received and the panels are producing time could be set to 12 o'clock
noon (UNIX timestamp 39600). If there is no production UNIX timestamp 0 can
be used, as is done currently.

This would overcome the following problem: if NTP is not available and
night communication is disabled AhoyDTU will not contact the inverter
for a few hours after startup because it thinks current time is just after
midnight (on Jan 1st 1970, UNIX timestamp 0).

Thanks,
Karsten

rmayergfx, 2023-05-15:
Why should NTP be unavailable, there are many options to integrate NTP:
https://github.com/lumapu/ahoy/issues/936#issuecomment-1547350547

btw.  An option to set a Backup Server would be nice

tqma1, 2023-05-15:
> Why should NTP be unavailable,

Because no internet is actually available or someone can not connect the dtu to a wifi becouse there is no permanet wifi available.

ncqgm, 2023-05-15:
Am Mon, May 15, 2023 at 04:54:30AM -0700 schrieb tqma1:

> > Why should NTP be unavailable,
>
> Because no internet is actually available or someone can not connect the dtu to a wifi becouse there is no permanet wifi available.

Well, and if the dtu is used/has to be used in AP mode
because of, say, connectivity issues ...

This is all about resilience.

Sure, this would be a somewhat exotic "feature" but such
things separate nice projects from really nice projects ;-)

Thanks,
Karsten
--
GPG  40BE 5B0E C98E 1713 AFA6  5BC0 3BEA AC80 7D4F C89B

lumapu, 2023-06-17:
check issue #951 

I've added a new feature: in the section of inverter-settings there is a new checkox which configures Ahoy to communicate with inverters even if there was no time set. Special feature: Once you hit the index page of Ahoy the time will then synchronized automatically (but only if the setting inside the inverter settings is true)

ncqgm, 2023-06-18:
Sounds fixed to me, when working as advertised. Maybe mark the checkbox by default, such that one sees data without having to _enable_ timeless communication.

closed 2023-06-18 by ncqgm