lumapu/ahoy #947: [BUG] Only showing bootscreen on Display if DTU is in AP-Mode and NTP is not set/synced from browser

McMakeIT, 2023-05-15:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9; 0.6.0

### Github Hash

15ec6a0, 9c7d7a0

### Build & Flash Method

ESP Tools (flash)

### Setup

NRF24L01+ with external antenna
ESP32 D1 Mini 
SH1106 1,3" OLED

DTU in AP Mode
One Inverter are configured (HM600)
NRF power set to max (doesn't matter, other values are tested)
Communication / connection OK

### Error description

After flashing and configuration of nRF, Display & Inverter only bootscreen is shown on Display (Ahoy, Version, bla,bla...).  No Inverter data is show on display. Serial Debug and Live Data are showing the right Inverter data in the WebUI. At this state the DTU is in AP-Mode (not connected to any Wifi).

Workaround:

Sync NTP "FROM BROWSER" and start "SYNC NTP". After that the Inverter Data is showing up on Display.  You have to repeat this process after every reboot of the DTU.

I'm a bit confused why the right Time (NTP) is needed for the Display.

ncqgm, 2023-05-16:
Initially, UNIX timestamp 0 is assumed as long as NTP ist not available and
time has not been synced from the browser yet. Effectively, this means the
DTU assumes night time. In case communications at night is disable it will
not try to connect to an inverter. Hence no inverter data.

Once you sync time the dtu will learn it is daytime and retrieve data
from inverters ...

See the issue "assume daytime when NTP not available".

Best,Karsten

ncqgm, 2023-05-16:
I missed the part that the web ui live page/serial debug show inverter data just fine
so this must be something other than night-comms vs assumed-time.

lumapu, 2023-07-23:
there is now a featrue `Start without time sync (useful in AP-Only-Mode)` in the dev-Version. You should try that again.

closed 2023-08-05 by lumapu