lumapu/ahoy #948: Leistungsreduzierung Eingangsseitig?

TSKStone, 2023-05-15:
Hallo, 

habe eine Ost/West Anlage in Betrieb genommen (je 2x 410W nach Osten bzw. Westen) an einem HM-1500.
Wenn ich jetzt die Leistungsreduzierung auf die aktuellen 600W bzw. 40% einstelle, werden die 
Eingänge auf 40% "gedrosselt"  - sprich die einzelne Modulleistung geht nicht über ca. 160W hinaus.
So komme ich morgends und abends eben "nur" auf 360W Spitzenleistung, lediglich mittags hab ich dann die vollen erlaubten 600W
für Balkonkraftwerke.
Ist es möglich die tatsächliche Ausgangsleistung des Wechselrichters zu steuern?
Die Leistungssteuerung scheint sich auf die Eingänge zu setzen.

## Hardware
  * [ ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

rejoe2, 2023-05-15:
Siehe auch https://github.com/lumapu/ahoy/issues/849

rmayergfx, 2023-05-18:
Schau mal hier: https://www.photovoltaikforum.com/core/article/49-ost-west-anlage-mit-einem-wechselrichter/

closed 2023-05-24 by TSKStone