lumapu/ahoy #949: [Bug] Total view: wrong values, sometimes multiple (different) are shown

rejoe2, 2023-05-16:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

DTU and MQTT work in general, 7 inverter (6*MI-1500/3rd gen, 1 MI-600)

### Error description

Not sure, if this has already been reported as official bug.

Sometimes one can see a strange behaviour of the "brown" total view, last user report was
https://discord.com/channels/984173303147155506/1029761098381017098/1107969021275541525 (single HM-600, but doubled values shown in the total view, V. 0.6.9 on ESP8266).
I myself can observe that kind of strange behaviour as well (same version, on ESP32, 7 inverter). Variations:
- Missing inverters (e.g. only last shown)
- multiple instances of the brown total filed (observed up to 4 fields in my setup), all with different total values, typically to high..

If desired, I might add some screenshots, but hope this is clear enough...

HorstyS, 2023-05-16:
ich habe ab und zu das selbe verhalten (ESP32). es tritt bei mir auf wenn die seite in einem browsertab offen steht und man ihn länger nicht neu öffnet (glaube ich). sobald der refresh stattfinden wenn der tab wieder aktiv ist wird die seite wieder normal angezeigt.

rejoe2, 2023-05-16:
Das mit dem offen gehaltenen Fenster könnte hinhauen. 
Allerdings war es bei meinem Schnelltest vorhin so, dass erst eine blanke Anzeige kam mit "every (5) seconds", und dann erst ein "doppeltes Lottchen" kam, dann nur total+der letzte angezeigt wurde, und dann wieder ein "doppeltes Lottchen". Von daher war ich davon ausgegangen gewesen, dass die komplette Seite neu geladen worden wäre.

Habe eben dann nochmal daen tab angesehen (weiter im Browser/FF unter Android) offen => wieder "every ..." dann das "doppelte Lottchen". Nach einem aktiven Refresh der Seite (nach unten ziehen) war dann in der Tat alles normal. Von daher könnte das schon mit Altdaten im Cache zusammenhängen.

Zur Klarstellung: ist eine kleine Irritation, die vorhin halt auf discord auch von jemandem anderen berichtet wurde. Falls man es einfach fixen kann, schön, falls nicht, ist es auch nicht dramatisch...

DSRocker, 2023-06-02:
Kann das bestätigen. Das taucht bei mir auch für wenige Sekunden auf.
[image]

lumapu, 2023-07-23:
ist das immer noch so?

stefan123t, 2024-10-27:
@rejoe2 ist das noch akut oder können wir das issue schließen ?

rejoe2, 2024-10-28:
Hab's schon länger nicht mehr gesehen, daher mal geschlossen...

closed 2024-10-28 by rejoe2