lumapu/ahoy #950: Hoymiles verweigert Gewährleistung

brunsheinz, 2023-05-16:
## Hardware
  * [ x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: _____
Retailer URL: ______

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x ] external antenna (SMA)

### Power Stabilization:
* [x ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [x ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

brunsheinz, 2023-05-16:
Hallo
Ich habe da ein Problem mit einem Hoymiles HM 1500 und Ahoy DTU und Hoymiles . Bei dem Wechselrichter wird auf Kanal 2 kein Strom mehr erzeugt . Liegt eindeutig am Eingang, denn beim Tausch der PV Anschlusskabel auf anderes Modul bleibt der Fehler im 2. Kanal bestehen . So eigentlich ist mit der Ahoy DTU alles O.K. Hoymiles lehnt jedoch einen Austausch ab , da ich die Diagnose mit Aho DTU gemacht habe . Also wenn ich die Ahoy DTU nutze geben Die keine Gewährleistung mehr .
[image]

lumapu, 2023-05-16:
was würde passieren, wenn du jetzt die Diagnose mit einer Hoymiles DTU machen würdest? Was soll Ahoy am Wechselrichter zerstören? 
Ahoy implementiert genau die Kommandos, die auch Hoymiles nutzt und hinterlässt keinerlei Spuren.
Evtl. kannst du das ganze auch über deinen Händler abwickeln. Danke fürs teilen deiner Erfahrung.

400g-Hammer, 2023-05-16:
Falls du das Teil nicht direkt bei Hoymiles gekauft hast, ist ein Händler für die Gewährleistung zuständig. Und sofern der WR nicht durch die AhoyDTU zerstört wurde (wovon ich nicht ausgehe) kann er die auch nicht ausschlagen, nur weil du ein Freeware-Toll verwendet hast. Dennoch könnte es geschickt sein, nix von der Ahoy zu sagen sondern nur die Diagnose mit den umgesteckten kabeln zu erwähnen - einfach nur um Diskussionen zu vermeiden.

roobbb, 2023-05-17:
hier ein guter Überblick: [https://www.allrecht.de/alles-was-recht-ist/beweislastumkehr/](url)
per 01.01.2022 gilt eine verlängerte Beweislastumkehr von 12 Monaten, statt bisher 6
ja, Händler/ Verkäufer ist 1. Ansprechpartner

brunsheinz, 2023-05-17:
Hallo
Der Händler ist bemüht mit Hoymiles eine Lösung zu finden , warten wir es ab .
Ohne Ahoy DTU ist es schwierig diesen Fehler zu lokalisieren . Ich habe am Anschluss noch einen Shelly hängen , aber der macht ja nur die Summe aller 4 Kanäle . Ich denke das Hoymiles nicht verlangen kann , das ich mir eine Hoymiles DTU beschaffe . Die haben sofort blockiert wenn man Ahoy DTU erwähnt .EBay habe ich auch schon informiert , vielleicht hilft das ja den Vorgang zu beschleunigen .

400g-Hammer, 2023-05-17:
Dein HM hat 4 Eingänge, aber nur 2 MPPT, d.h. 2 Module gehen auf je einen MPPT.
Arbeitet der WR, wenn nur ein einziges Solarmodul angeschlossen ist?
So wie du den Fehler beschreibst bzw. wie es im Ahoy-Screenshot dargestellt ist, anscheinend ja. 

Wenn du ein Energiemessgerät hast, mit dem sich Einspeisung erfassen lässt (nen Shelly tuts hier ja auch), dann könntest du folgendes probieren:
Schliess doch mal nur ein Modul an, und zwar nach einandern an jeden einzelnen Eingang.Dann müsstes du bei drei Eingängen Ertrag kriegen und bei dem einen defekten nicht. Den Ertrag misst du diesmal "politsch korrekt" mit dem externen Energiemessgerät.

Wenn das nichts bringt, schliesse halt zwei Module an den einen MPPT an und danach dieselben zwei Module an den anderen.
Auch da müsste ein Unterschied erkennbar werden. Auch hier könnte man den Eingang genauer bestimmen, z.b. durch Abdecken eines Moduls. Wird das am am defekten Eingang abgedeckt, ändert sich nichts, wird das a intakten Eingang abgedeckt, fällt der Ertrag weg.

Ist zwar umständlicher, als die Ahoy-Diagnose aber dafür kann sich auch niemand mehr querstellen.

rmayergfx, 2023-05-18:
Du hast doch hier schon die passende Antwort erhalten:
https://www.photovoltaikforum.com/thread/203383-hoymiles-wechselrichter-garantie/
Gerät zum Händler und fertig. Alles andere kann nicht dein Problem sein. 
Auf der Hoymiles Seite steht auch 12 Jahre Garantie, wichtig sind die Garantiebestimmungen.
Seltsamerweise steht auf der Hoymiles Homepage zum Thema Garantie widersprüchliche Angaben,
das kann aber nur ein Rechtsexperte zum Thema Garantie aufschlüsseln.
Im Text wird einmal von Garantie gesprochen:

> Unsere Mikro-Wechselrichter haben alle eine Standard-Garantiezeit von 12 Jahren. Sollten Probleme auftreten, reparieren wir das Gerät entweder vor Ort oder ersetzen es durch ein neues oder generalüberholtes Modell.

Dann wiederum von Gewährleistung:

> Von unserer 12-jährigen Standardgewährleistung bis hin zu flexiblen Erweiterungsmöglichkeiten – mit unserem Garantieprogramm nehmen Sie die Unsicherheiten aus Ihrer Installation.

https://www.hoymiles.com/de/support/warranty-information/

Wichtig ist das Kleingedruckte im PDF:

> Die standardmäßige Garantiezeit für Wechselrichter beträgt 144 Monate
> ...
> Diese Garantierichtlinie gilt nur für Wechselrichter, die von einem
> entsprechend qualifizierten Fachmann installiert wurden. Die Garantie wird nichtig, wenn Wechselrichter über nicht
> autorisierte Vertriebskanäle weiterverkauft werden. 
> ...
> Als privater Endverbraucher wenden Sie sich im Falle eines Garantieproblems bitte zunächst an Ihren Installateur.
> Hoymiles wird direkt mit dem Installateur zusammenarbeiten, um einen defekten Wechselrichter auszutauschen, wenn
> er gemäß den Bedingungen des Austauschprogramms dazu qualifiziert ist.
> ...
> 
> Bei Schäden, die auf die unten aufgeführten Ursachen zurückzuführen sind, werden keine Garantieleistungen
> übernommen. Defekte, die durch die folgenden Faktoren verursacht werden, fallen nicht unter die Gewährleistungspflicht
> von Hoymiles:
> a. höhere Gewalt (Sturmschäden, Blitzschlag, Überspannung, Feuer, Gewitter, Überschwemmung; soziale Ursachen
> wie Krieg, Unruhen, staatliche Eingriffe, Streiks, Embargos, Marktbedingungen usw.);
> b. unsachgemäße oder nicht bestimmungsgemäße Verwendung;
> c. unsachgemäße(r) Installation, Inbetriebnahme oder Betrieb (den Anweisungen in der Installationsanleitung, die mit
> jedem Produkt geliefert wird, nicht entsprechend);

Wichtig sind dabei die b) und c) da kann Hoymiles argumentieren das die Verwendung einer anderen DTU nicht bestimmungsgemäß ist.

Schick das Teil an den Händler zurück,  er soll sich darum kümmern.

400g-Hammer, 2023-05-18:
Ich fasse die Garantiebedingungen mal zusammen:
Grösstenteils wertlos (was micht nicht verwundert), denn spätestens mit diesem Satz wird klar, dass der BKW-Betreiber im Regen  stehen gelassen wird:

> Diese Garantierichtlinie gilt nur für Wechselrichter, die von einem
entsprechend qualifizierten Fachmann installiert wurden.

Der Reiz der Balkonkraftwerks ist ja gerade eben, dass man keinen Elektriker brauchen  soll, um es in Betrieb zu nehmen.
Ja, klar, zur Inbetriebnahm des HM1500, bedarf es  eigentlich eines Elektrikers, aber ich könnte mir vorstellen, dass das nicht immer so gehandhabt wird.

rmayergfx, 2023-05-18:
@brunsheinz 
Da das Thema Balkonkraftwerke und Solar allgemein aktuell recht präsent ist und sich verschiedene Zeitschriften u.a. c´t sich mit diversen Leitfäden zu diesem Thema beschäftigen, kannst du parallel dazu die Redaktion anschreiben:
vorsichtkunde@ct.de

rmayergfx, 2023-05-18:
> hier ein guter Überblick: [https://www.allrecht.de/alles-was-recht-ist/beweislastumkehr/](url) per 01.01.2022 gilt eine verlängerte Beweislastumkehr von 12 Monaten, statt bisher 6 ja, Händler/ Verkäufer ist 1. Ansprechpartner

Finde ich gut, aber leider in diesem Fall nicht hilfreich. Der WR ist lt. seiner Aussage erst 4 Wochen alt und zudem sagt Hoymiles selbst 12 Jahre Garantie, verknüpft mit den Garantiebedingungen.

rmayergfx, 2023-05-18:
Ist nicht hilfreich das dies schon seit 01.01.2022 gültig ist, also nichts neues und der WR ist 4 Wochen alt. Da ändert auch die Verlängerte Frist nichts daran. Zudem schreibt der TE das der Händler da bereits dran ist. Also abwarten.

kiu77, 2023-05-21:
In den ersten 6 Monaten muss der Kunde dem Händler nichts nachweisen, sondern der Händler muss bei Ablehnung von Gewährleistung belegen, dass der Fehler nicht im Produkt lag, sondern irgendwie beim Kunden (was er typischerweise nicht kann). Mit AhoyDTU kann sich der Händler jedenfalls nicht rausreden.

brunsheinz, 2023-05-22:
Ich habe den Händler  Sun_solar_24  nochmals aufgefordert mir einen Ersatz Wechselrichter zu senden . Bis jetzt habe ich noch keinen erhalten . Ich denke der Verkäufer hat sich auch auf die Hoymiles12 Jahres Garantie verlassen und versucht zu verzögern . Aber rechtlich denke ich hat der keine Chance .
Ich weiß ja nicht wie häufig es zu Fehlern an Hoymiles Wechselrichtern kommt und er nicht damit gerechnet hat

kiu77, 2023-05-22:
Normalerweise sind sie sehr zuverlässig (die Wechselrichter).
Du musst dem Händler Deinen Wechselrichter zuschicken mit einer Fehlerbeschreibung und einen Termin setzen.

Peertje62, 2023-05-23:
Hi, 
kontaktlose Strommessungen könnte eine Option sein.
Gr Herman (Hollander)

brunsheinz, 2023-05-24:
Ist jetzt bei Hoymiles zu spät ,egal was ich mache , da der WR mit SNr. registriert ist . Also versuche gerade das mit dem Händler abzuklären

lumapu, 2023-07-23:
gibt es einen neuen Status? Ist sicherlich auch für andere interssant.

brunsheinz, 2023-07-24:
Hallo
Der Verkäufer hat den Wechselrichter ausgetauscht , von Hoymiles habe ich nichts mehr gehört

closed 2023-07-24 by brunsheinz

lumapu, 2023-07-24:
danke für die Rückmeldung