lumapu/ahoy #952: Failed to execute 'open' on 'SerialPort': Failed to open serial port.[Bug]

ArminPelka, 2023-05-19:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

none

### Github Hash

none

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Just the ESP8266 module

### Error description

Failed to execute 'open' on 'SerialPort': Failed to open serial port.

This error message comes when using the Web-Installer. I tried different cables, browsers and PCs.

lumapu, 2023-05-19:
what does the device manager show? Are there any serial ports?

ArminPelka, 2023-05-19:
Yes, there is a serial port and the driver is actual. When I change the USB-Port the port number changes as well.

ArminPelka, 2023-05-20:
It has to do with the PC. I tried it with two PCs with Windows 11 and it did not work. Today I tried it with Windows 10 and it worked. Device manager shows the same driver CH340. Not sure if it has to do with the Windows version and some hidden security properties of the browsers (Edge and Chrome tried). But the ones, where USB devices can be disabled are on "ask for connection". I dont now the root cause finally but be convinced that it has nothing to do with Ahoy DTU. and this is why I close the issue.

closed 2023-05-20 by ArminPelka