lumapu/ahoy #954: No connection to HM-600

TomHW, 2023-05-20:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: D1 Mini ESP8266 WLAN
Retailer URL: makershop.de

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.6.9
Github Hash: 15ec6a0

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [x] ESP Tools

### Debugging:
* [x] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)
[serial-log.txt](https://github.com/lumapu/ahoy/files/11522136/serial-log.txt)
[settings.txt](https://github.com/lumapu/ahoy/files/11522137/settings.txt)

My Wemos isn't able to get a connection to the inverter. I tried several power levels without success and checked the wiring and the serial number several times. What else can I do?
[image]

TomHW, 2023-05-20:
After switching off the LEDs in config it works. The image had the LEDs on. You can close the issue.

knickohr, 2023-05-20:
You can close the issue by yourself 😉

TomHW, 2023-05-20:
Thanks and done.

closed 2023-05-20 by TomHW