lumapu/ahoy #955: [Bug] MQTT not working correctly in 0.6.9

XL-Reaper, 2023-05-20:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

- Inverter set to 10 seconds
- MQTT broker IP added, anonymous authentication

### Error description

When MQTT is enabled the ESP almost completely stops responding. It doesn't react to pings anymore, only every 10th or 20th ping gets a response, the webpage does not load anymore. 
Also the MQTT server (mosquitto) sees the client connecting and timing out again and again.

100% the same setup with version 0.5.66 works without any problems.

I've tried multiple times to flash versions 0.6.9 and also the development 0.6.12, but they both show the same behaviour.
Once I flash the ESP8266 back to 0.5.66 the setup just works right away after doing the same settings.

XL-Reaper, 2023-05-20:
Forogt to add: once I stop the mosquitto server, the ESP becomes reactive again. When I then once again start the mosquitto service, it almost immediately dies again.

MartinBachmannHD, 2023-05-24:
I had exactly the same behavior on 0.6.9, the UI was not responding at all, but still from time to time MQTT messages were sent. I changed yesterday to 0.6.14 and now it is working much, much better, UI is fully responsive now. I test everything with two different 8266, both had the '0.6.9-Slow-UI-Bug' and both work now fine.

lumapu, 2023-05-25:
thank you for the great feedback 😀

sweinreuter, 2023-05-26:
I can confirm that the UI responsiveness with MQTT connected is fixed in 0.6.14 but it's back in 0.6.15: Once MQTT is enabled and connected, the UI is essentially unusable (tested on ESP8266). 

Also, with 0.6.9 I had lots of MQTT reconnects which seems to be much better now with 0.6.14.

RedVision81, 2023-05-28:
where can i get the 0.6.14 ? latest development is already 0.6.15. thx

JeaneLG, 2023-06-05:
I also had the problem after activating MQTT with 0.6.9. Then, I disabled the display and everything is working fine. Can someone check if this also works with other set-ups? I also added an additional power supply for the 3.3V, but this did not help me.

lumapu, 2023-06-05:
> where can i get the 0.6.14 ? latest development is already 0.6.15. thx

here you go: https://github.com/lumapu/ahoy/actions/runs/5037786059

RedVision81, 2023-06-07:
thank you very much! installed it a few minutes ago and its much more responsive again.
today was one of the worst days where hardly any data where recorded:

[image]

bear30, 2023-06-22:
Hallo, ich hänge mich hier mal dran. Habe 2 Systeme die ich betreue. Beide haben identische Hardware mit ESP32 usw und beide haben die Ahoy Version 0.69. Bei einem klappt das MQQT problemlos, beim anderen System nicht. Einziger Unterschied für mich ist dass das funktionierende System 2 HM 1500 am Start hat, das was nicht geht hat 3 HM 1500. Selbe Fehlermeldung wie beim Kollegen: MQTT disconnected, reason: TCP disconnect

bear30, 2023-06-22:
Entwarnung: Ich habe bei mir den Fehler gefunden. Die Fritze war schuld ...

lumapu, 2023-07-03:
@bear30 kannst du das Problem noch genauer beschreiben, das die Fritzbox hatte? Evtl. hilft es auch anderen weiter.

bear30, 2023-08-03:
Ich hatte nur das 5MHz Wifi  am Start

closed 2023-08-05 by lumapu