lumapu/ahoy #957: YieldDay Nachts um 00:00h keine Reset auf 0

mymeyer, 2023-05-23:
## Hardware
  * [x] ESP8266
  
Modelname: Wemos 1D Mini

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
* [x] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [x ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.6.9

Hallo, der Wert von YieldDay wird Nachts nicht auf 0 zurückgesetzt , obwohl ich es in der Konfig so eingestellt habe.
Er zählt einfach weiter und meldet es natürlich über MQTT auch falsch.

M.M.

lumapu, 2023-05-25:
bitte mit dev `0.6.15` nochmal gegenchecken

mymeyer, 2023-06-03:
Funktioniert auch nicht mit der Version 0.6.15
Der Wert wird nicht auf 0 gesetzt !

Mymeyer

Rufus4x4, 2023-06-07:
Hallo zusammen,
auch bei mir kein Reset von "Yield Day" auf 0 zu Mitternacht feststellbar. Habe aktuellste Version installiert.

Rufus4x4

lumapu, 2023-06-16:
bitte testen, ob es jetzt gefixt ist - leider muss man hier bis morgen Abend warten ...

Rufus4x4, 2023-06-18:
Rückmeldung von mir: Nach Update aiuf 0.7.5 noch immer kein YieldDay-Reset um Mitternacht (in der Live-Anzeige des Wechselrichters), obwohl es in den "Inverter-Settings" eingeschaltet ist.
Vielleicht noch eine Anmerkung zu meinem Aufbau: Der Hoymiles-Wechselrichter hängt DC-seitig permanent am Akku (konstante Einspeisung einer Grundlast ins AC-Netz).

Gruß 
Rufus4x4

beebee, 2023-06-20:
Mit der 0.7.6 ging es heute via MQTT YieldDay kam um 0 Uhr mit 0 rein - aber: Im Live View GUI wird am Channel und Total vom Inverter weiter das YieldDay angezeigt.

Setup:
Batterie an HM-400.
Inverter war über Mitternacht DC verbunden aber AC abgeschaltet da SOC niedrig: "Inverter #0: Battery (v10010) is available and is not producing" .
"Reset values and YieldDay at midnight" ist aktiviert.

Versuche es nächste Nacht nochmal mit aktiver Einspeisung um Mitternacht was dort passiert.

beebee, 2023-06-20:
Seltsam auch: nachdem AC heute wieder verbunden wurde und eingespeist wurde schlägt das YieldDay wieder zurück auf den gestrigen Wert (wie in der GUI auch angezeigt):

[image]

Rufus4x4, 2023-06-20:
ohne DC-seitige Abschaltung wird "Yield Day" im Inverter vermutlich ständig weitersummiert. Ob das programmtechnisch überhaupt möglich ist, den Live-View um Mitternacht auf 0 zurückzusetzen? Der Hoymiles sendet ja immer den akkumulierten Yield Day.

Rufus4x4

Rufus4x4, 2023-06-20:
bei einem Reset ohne DC-seitige Abkoppelung wird der Yield Day ja auch nicht gelöscht sondern weiteraddiert.

closed 2023-08-05 by lumapu