lumapu/ahoy #959: [Bug]

agentsmith1612, 2023-05-24:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

ESP Tools (flash)

### Setup

- adding more than 3 inverters

### Error description

I have 6 inverters on my roof (5x HM-1500) and one HM-700. The system slows down the more inverters I add. MQTT is also activated. 
When I want to add the 4th inverter I have to wait 2 min so load the setting site correctly. 
When I enters the 4th inverter data and click saving, the browser want to download a file called "save". 
When I try to click the second time on "save" the site did not load and the ahoy crashs, without saving the data. 

Sometimes it wanted to save but ahoy said "could not save". 

That i very strange to me. I tried three different ESPs with different NRF boards. All behave the same.

fsck-block, 2023-05-25:
> When I enters the 4th inverter data and click saving, the browser want to download a file called "save".
> When I try to click the second time on "save" the site did not load and the ahoy crashs, without saving the data.

I can confirm this for my esp8266 / development version 0.6.14.
Adding the 5th inverter leads to a non-function 
```
...
I: TX 27B Ch61 | 15 83 16 55 93 88 42 51 75 80 0B 00 64 6E F5 75 00 00 00 00 00 00 00 00 2D 55 D1 
I: memory usage: 0
I: capacity: 0
I: max alloc: -408
E: buffer too small!
I: (#1) resetPayload
...
```

or a crash due to insufficient memory

```
...
User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM
```

lumapu, 2023-05-25:
someday the ESP8266 will be too small - I would recomend to use an ESP32 for more than 3 inverters.

agentsmith1612, 2023-05-26:
I will buy one and will report.

fsck-block, 2023-05-27:
> someday the ESP8266 will be too small - I would recomend to use an ESP32 for more than 3 inverters.

It may be usefull for beginners to describe this limitation in the documentation (Getting_Started) and to give a hint in config.h where MAX_NUM_INVERTERS is defined.

Or we implement MAX_NUM_INVERTERS based on the given hardware platform 
BTW: Reduced my development setting to 5 inverters which leads to about 5k of additional free heap memory.

@lumapu:  If you agree I will propose a PR.

agentsmith1612, 2023-05-27:
I remember that in previouse version 5,xx,xx there was a hard limit that you can only enter 4 inverters?

agentsmith1612, 2023-06-08:
I used an ESP32 now and all works as intendet. Thanks for the help. :-)

closed 2023-06-08 by agentsmith1612