lumapu/ahoy #960: [Bug]

GrisuXX, 2023-05-25:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Inverter set to 10 seconds

### Error description

Folgendes Verhalten habe ich festgestellt:
1.	Wenn nach dem Neustart der DTU nur ein Wechselrichter erreichbar ist werden keine „total“ Statistiken erstellt bzw. aktualisiert.
Andersherum, wenn zwei oder mehr WR erreichbar sind und alle bis auf einen wegfallen, werden die „total“ Statistiken weitergeführt.
2.	Fällt ein WR aus bleiben alle Werte auch die „total“ Werte wie YieldDay, P_AC, P_DC bestehen.
In Grafana wird der „total/P_DC“ Wert aller WR angezeigt inc. des Ausgefallenen WR.
Hier kann man im Setup/Inverter zwar das Häkchen setzen bei:
„Reset values when inverter status is 'not available'“
Leider wird dann auch der „total/YieldDay“ Wert zurückgesetzt. Die bis zum Ausfall produzierte Leistung ist dann aus der Statistik verschwunden.
3.	Reset values and YieldDay at midnight funktioniert scheinbar nicht.

LG

knickohr, 2023-05-25:
Siehe dazu auch #929

MartinBachmannHD, 2023-05-26:
Yesterday night I tried the latest 0.6.15 release. Unfortunately the 'Slow-UI-Bug' from 0.6.9 seems to be back for me that was fixed in 0.6.14. This time in 0.6.15 so slow, I only could USB-reflash to 0.6.14 (UI firmware update did not work...). For me it seems to be somehow MQTT-related, taking away to many resources from the small 8266...
[](https://github.com/lumapu/ahoy/issues/955)

GrisuXX, 2023-05-26:
Update zur Version 0.6.15

Die total Werte werden jetzt auch mit nur ein Inverter aktualisiert. (Sehr schön)
Wünschenswert das Feld YieldDay ausschließlich um 0Uhr zu resetten und nicht bei ´not available´
Mit dem ESP32 sind keine Performance Problem festzustellen.

LG

knickohr, 2023-05-26:
Dann nimm den Schalter „Reset values and YieldDay at midnight“ dazu.

GrisuXX, 2023-05-26:
Der Schalter "Reset values when inverter status is 'not available'" sollte den Wert YieldDay nicht anfassen.
Ob der Schalter  „Reset values and YieldDay at midnight“ funktioniert kann ich erst Morgen sagen.

GrisuXX, 2023-05-26:
Die Werte werden n der Ahoy DTU exakt um 0Uhr resettet. :-)
Nicht aber im WR. Beim nächsten auslesen stehen wieder die Werte vom Vortag drin.
Der WR ist für die Nachteinspeisung und ist um diese Uhrzeit erreichbar.
Da werde ich mich doch mehr um Grafana kümmern und die Zeiten hier irgendwie definieren müssen.
Ist alles noch recht neu für mich und arbeite mich noch ein.
Ist ja nur noch das i Tüpfelchen die Daten entsprechend optisch aufzubereiten. 

LG

knickohr, 2023-05-27:
Im WR selbst kann man nichts resetten oder verändern.

closed 2023-08-05 by lumapu