lumapu/ahoy #962: Forgot password to Login - Version 0.5.66

hanegg, 2023-05-26:
I protected the settings with a password.
Unfortunately, I forgot the password. 
How can I reset the password or get access to the settings again.
Any help and/or comment is much appreciated.

rmayergfx, 2023-05-26:
What areas are secured with the password?
Backup prior to Password settings available?
Can you view your necessary settings? If it is possible, take screenshots of your settings and start with a fresh setup and apply the settings again. 

think about a password manager.

Ollipop030, 2023-05-26:
Did you export your settings? In ahoy_setup.json you can see your pw, just open it with an editor.

hanegg, 2023-05-27:
@rmayergfx: 👍 Thank you for your fast answer.
Settings cannot be seen, as I protected the whole area.
The advice of Ollipop030 does not work, as there is no password in setup.json.
There is only the value, that password is activated, see extraction of setup.json:
..... "hostname":"AHOYDTU2023","pwd_set":true,"prot_mask":29,"sdk":"2.2.2-dev(38a443e)",..

Password manager is a good hint :-)

Does anybody have an idea whether there is a hardware reset?

rmayergfx, 2023-05-27:
Tell us the hardware you are running AhoyDTU and we can assist you with the right steps.
independet from the passwort it should always be possible to connect the ESP8266 or ESP32 with USB to the PC and install fresh with web-installer from the homepage.
Then you can try to apply a password and load your backup. After reboot you can check whether your new password is excepted or not. 
You can also make a copy from your setup.json and set pwd_set":false

hanegg, 2023-05-27:
@rmayergfx: Thank you so much for your support so far!

The hardware used is an ESP8266MOD with mini USB.
I tried the https://ahoydtu.de/web_install/ for installation.
But it seems nothing to happen.
Firefox is not supported, therefor I used Google Chrome.
Google Chrome shows Connect button.
After having selected the communication port (COM 1)
The Device Dashboard offers:
INSTALL AHOYDTU and
LOG&CONSOLE
A click on INSTALL AHOYDTU brings a progress symbol and it seems that nothing happens.
As the installation lasted 20 minutes and was not finished yet.

rmayergfx, 2023-05-28:
Wich Board exactly? Link to seller? Is there a RST button on the board?
if so, press RST before flashing the new firmware.
You can also try to install a clean ESPHome via https://esphome.github.io/esp-web-tools/ in chrome, this will always work. you can setup esphome matching your lan and then update firmware inside ESPHome with matching bin from ahoydtu.
Do not get confused, this will work. tested many times. i first setup all boards with ESPHome to test the Hardware or to get a clean flash before i apply another firmware to be save the hardware is not DOA.

DanielR92, 2023-08-25:
Makes me an impression that there is no longer a need. Please reopen if so.

closed 2023-08-25 by DanielR92