lumapu/ahoy #964: Example for pure Home Assistant MQTT Zeroexport / Nulleinspeisung with Battery & AHOY-DTU Automation

beebee, 2023-05-26:
Hello,

I tried to find a pure and simple Home Assistant only solution for a Zero Export / Nulleinspeisung automation without NodeRed and python scripts, additional RPIs, tools etc to simply work directly with HA & ahoy-du via MQTT and didn't find any matching my requirements.
So I created my own with some support from ChatGPT to enhance it further and would like to share it here so others can use it as a starting point.

You need:

- A HA sensor state which holds your current total household/grid power usage where negative values mean export to grid and positive import from grid. I have an extra sensor for each (positive) - you can replace/calculate household power to your needs in the automation _curGridUsage_. (I have Shelly 3EM and also Tibber Pulse)
- Power switch(_switch.switchonoff_inverter_AC_battery_to_house_) with Sensor (_sensor.power_battery_to_house_) for your Homilies Inverter AC output that you are using for unloading your battery. Automation can work without and would use/can use previous battery power setting if actual power not available alternatively. Might need to disable condition if no switch available.(I have Shelly Plug flashed and power calibrated with Tasmota)
- a new input_number and input_boolean for control of the zero export in your _configuration.yaml_. Not using any initial value means last value is saved between HA restarts. Can adjust min/max/step on slider which is holding/showing the calculated automation value or can be manually set when _battery_nulleinspeisung_ is off:
```
input_number:
  battery_limit:
    name: Battery Nulleinspeisung Limit
    min: 0
    max: 120
    step: 1
    mode: slider
    # initial: 0

input_boolean:
  battery_nulleinspeisung:
    name: Battery Nulleinspeisung
    icon: mdi:battery-heart-variant
    # initial: false
```
- Automation. You need to adjust the MQTT Topics to match your ahoy-dtu inverter target. Can set min, max battery power, hysteresis. If _input_boolean.battery_nulleinspeisung_  is flipped off, you can use the slider for _input_number.battery_limit_ to manually set the battery limit and skip the calculation. Inverter target will be turned off/on dynamically with the first MQTT publish when lower than 7W which you can adjust there of course in the template. My HM-400 does not follow any limit values below 8W so switching off is better for my battery for "idle" power saving (Bluetti EB70 only). Also HM400 stays in "hot-standby" and is turned on and power-set  directly when MQTT limit is calculated for higher again or manually set 
```
alias: AHOY-DTU Battery Nulleinspeisung mit Hysterese
description: >-
  Hält den Battery-Power-Limit für die Nulleinspeisung durch die Batterie
  aufrecht.
trigger:
  - platform: time_pattern
    seconds: /20
condition:
  - condition: state
    entity_id: switch.switchonoff_inverter_AC_battery_to_house
    state: "on"
    enabled: true
action:
  - service: input_number.set_value
    target:
      entity_id: input_number.battery_limit
    data_template:
      value: >-
        {% set minBatteryPower = 0 %}
        {% set maxBatteryPower = 75 %}
        {% set hysteresis = 5 %}

        {% set oldBatteryLimit = states('input_number.battery_limit') | float %}
        {% set newBatteryLimit = oldBatteryLimit %}

        {% set curGridUsage = states('sensor.power_import') | int - states('sensor.power_export') | int %}
        {% set curBatteryPower = states('sensor.power_battery_to_house') | int(oldBatteryLimit)  %}

        {% set powerFlow = curGridUsage + curBatteryPower %}

        {% if is_number(powerFlow) and is_state('input_boolean.battery_nulleinspeisung','on') %}
          {% if powerFlow >= maxBatteryPower and oldBatteryLimit != maxBatteryPower %}
            {% set newBatteryLimit = maxBatteryPower %}
          {% elif powerFlow <= maxBatteryPower and powerFlow + hysteresis < oldBatteryLimit %}
            {% set newBatteryLimit = powerFlow + hysteresis %}
          {% elif powerFlow <= maxBatteryPower and powerFlow - hysteresis > oldBatteryLimit %}
            {% set newBatteryLimit = powerFlow - hysteresis %}
          {% endif %}
        {% endif %}

        {% if newBatteryLimit <= minBatteryPower %}
          {% set newBatteryLimit = minBatteryPower %}
        {% endif %}

        {{ newBatteryLimit }}
  - service: mqtt.publish
    data:
      topic: inverter/ctrl/power/0
      payload_template: |-
        {% if states('input_number.battery_limit') | int < 7 %}
          0
        {% else %}
          1
        {% endif %}
      retain: true
  - service: mqtt.publish
    data:
      topic: inverter/ctrl/limit/0
      payload_template: "{{ states('input_number.battery_limit') |round }}W"
mode: single
```

Hints:
-  ahoy-dtu does not seem to read the retained power on/off state when subscribing to the MQTT broker/topic after a reboot from my experience. I suggest you set some lower persistent limit on your battery inverter directly via ahoy GUI.
- You can switch off the input_boolean and set battery_limit to 0 (so this automation will send an inverter off) before powering off the Inverter AC smart plug after a delay is a great other automation idea if your RCD/FI doesn't like full-load cutting ;)
- I turn of/on this automation here with others based on my virtual battery SOC

lumapu, 2023-06-16:
do you see any task for me so far? 

is it a big problem?
> ahoy-dtu does not seem to read the retained power on/off state when subscribing to the MQTT broker/topic after a reboot from my experience. I suggest you set some lower persistent limit on your battery inverter directly via ahoy GUI.

Should we think of publishing such tutorials / examples on https://ahoydtu.de?

beebee, 2023-06-16:
No big problem from my side about the retain. I was putting it here for documentation/people to fiddle with and thought it might be helpful for the GitHub Wiki for the HA users. No Idea really about how big the need is - Examples on ahoy website sound nice to me.

stefan123t, 2024-10-27:
@beebee @lumapu sollen wir das in tools / doc mit dem Script und einem Readme.md dokumentieren ?