lumapu/ahoy #965: (question) Will the WeMos S2 mini be supported?

bltoby, 2023-05-27:
A total newbie question 

WeMos S2 mini is also EPS32 based, although slashed to a single core, but comparing to D1 mini at about the same price, it provides a much 
better price performance ratio.

So my question is will WeMos S2 mini be supported by ahoy?

Thanks

tastendruecker123, 2023-05-31:
Someone tried to add support a while ago, and though it was possible to compile Ahoy for the S2, it crashed when it tried to establish a wifi connection. I have ordered an S2 and I might get to play around with it in July, so it should be possible to make Ahoy run on the S2. The question is just whether supporting another ESP could make it harder to maintain the project.

kscholty, 2023-08-18:
I tried it as well for the last few days.
It works fine in AP mode. As soon was it connects to to the WIFI in Station mode it seems to go into an endless loop. It's no crash, unfortunately but the chip gets extremely hot.
Tried the the latest  platform libs etc. to no avail.
Also deactivated all radio, mqtt, etc. but still it happens. The S2 mini does not expose the JTAG pins so that I can't debug it with my equipment.

It's probably something in the wifi setup. Which in theory should be quite straight forward.

lumapu, 2024-01-02:
should be supported since version `0.7.42` from 2023-08-27

closed 2024-01-02 by lumapu

DB1BMN-collection, 2024-03-04:
Hi, are there any progress supporting the S2 mini?
I've just received mine but can not install via Webinstaller (in Chrome of course).

I push the "0"-Butteon and then press Reset und releas it and release "0".
Some Devices (ESP32-S2, IcService_New and SMS/MMS) are listet in the Device Manager and a new Comport (COM6).
[image]

But in the Webinstaller I cannot flash it. It says "Failed to initialize". The COM-Port will be recognized anyway.
Tried to press "0"-Button again bevor clicking "Install" but no success.
[image]

Any ideas?

MetaChuh, 2024-03-05:
hi @DB1BMN-collection 

probier mal:

esp flash download tools hier herunterladen: [`Espressif Flash Download Tools V3.9.5`](https://www.espressif.com/sites/default/files/tools/flash_download_tool_3.9.5.zip)

v0.8.89 für esp32 s2 mini hier herunterladen: [`AhoyDTU-v0.8.89-ESP32-S2.zip`](https://github.com/lumapu/ahoy/files/14498785/AhoyDTU-v0.8.89-ESP32-S2.zip)

esp tools starten und uart auswählen.

screenshot:
[image]

alle 4 dateien genau wie auf folgendem screenshot hinzufügen,
anhaken zum aktivieren,
die richtige zieladresse angeben,
den com port der bei dir vorhanden ist und zutrifft auswählen.

screenshot:
[image]

dann zuerst <kbd>erase</kbd> klicken.
wenn das geklappt hat, dann auf <kbd>start</kbd> klicken und beten 😉

fehlersuche wenn es nicht funktioniert:
1) beim start von esp flash tools usb auswählen und erneut probieren.
2) diesen winchip ch34xser treiber für win10/11 arm64/x64 im gerätemanager installieren:
[`WinChip CH341SER (wch.cn) win11 x64-arm64 drivers.zip`](https://github.com/lumapu/ahoy/files/14499022/WinChip.CH341SER.wch.cn.win11.x64-arm64.drivers.zip)
danach im uart mode erneut probieren.
3) benzin holen, auf den s2 gießen, anzünden (im freien).
danach nicht mehr probieren, aber bitte fotos davon posten 😉

greetings,
metachuh

DB1BMN-collection, 2024-03-05:
Wow, danke, das Programmieren hat soweit geklappt, glaube ich.
[image]

Konsolen-Ausgabe:
`Uploading stub...
Running stub...
Stub running...
Changing baud rate to 115200
Changed.
SPI_BOOT_CRYPT_CNT 0
SECURE_BOOT_EN False
Compressed 13648 bytes to 9600...
Compressed 3072 bytes to 146...
Compressed 8192 bytes to 47...
Compressed 1194736 bytes to 784578...

 is stub and send flash finish`

Aber nach dem Neustart sehe ich keinen WLAN-Access-Point mit der SSID "AHOY-DTU".
Habe es dann auch mit "USB" für "LoadMode" probiert
Weitere Ideen?

Das mit dem Anzünden ist keine gute Option. FR4 brennt nicht besonders gut: https://www.mikrocontroller.net/topic/565226

MetaChuh, 2024-03-05:
erster schritt (flashen) ist mal ein teilerfolg 👍

> Aber nach dem Neustart sehe ich keinen WLAN-Access-Point mit der SSID "AHOY-DTU".

hänge den s2 mal am pc, linux, mac auf einen usb port,
starte chrome/chromium,
öffne die seite [`https://ahoydtu.de/web_install`](https://ahoydtu.de/web_install/)
klicke auf connect
wähle usb2.0-Ser aus
klicke auf `Logs & Console` wie am folgenden screenshot:

[image]

beobachte und poste das serial log in copy/paste textform, oder als screenshot hier.
beispiel screenshot:
[image]

hoffe du siehst keine fehlermeldung, und keinen black screen mit nur einem `>` cursor.

> Das mit dem Anzünden ist keine gute Option. FR4 brennt nicht besonders gut: [https://www.mikrocontroller.net/topic/56](https://www.mikrocontroller.net/topic/565226)

in dem thread hat ja auch noch keiner benzin darauf gegossen ... wär lustig wenn du den microcontroller thread noch mit dem wort "benzin" ergänzt 😂

MetaChuh, 2024-03-08:
... und dabei das lied "me gusta la gasolina" anhören ;-)

peterhax, 2024-04-16:
Moin,

ich habe auch noch etwas rumprobiert und ich konnte mein S2 mini mit dem esptool.py erfolgreich bespielen und in Betrieb nehmen ...

`python esptool.py -p COM5 -b 115200 --before default_reset --after hard_reset --chip auto  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader.bin 0x8000 partitions.bin 0xe000 ota.bin 0x10000 240302_ahoy_0.8.89_0d7c67d_esp32-s2-mini-de.bin --force`

[image]

peterhax, 2024-04-16:
Moin,

hat direkt funktioniert! Hab einen neuen S2 mini genommen und mit GPIO0 halten, Reset kurz drücken und GPIO0 loslassen in den Bootmodus versetzt, wurde direkt erkannt und der Flashvorgang wurde durchgeführt. 

[image]

[image]

Tolle Arbeit von Euch!