lumapu/ahoy #966: Brauche Hilfe beim Starten der Verbindung

Jo131313, 2023-05-28:
## Hardware
  * [ ] ESP8266
  * [ ] ESP32
  * [x ] Raspberry Pi

Modelname: __4B____
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Versuche schon seit Freitag die Verbindung herzustellen hab keine Ahnung wo ich jetzt noch ansetzen kann:
Bekomme folgende Fehlermeldungen:

Beim manuellen Starten:
bu@Raspi:/home/ahoy/tools/rpi $ sudo python3 -um hoymiles --log-transactions --verbose --config /home/ahoy/tools/rpi/aho
y.yml | tee -a log2.log
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 188, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.9/runpy.py", line 147, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.9/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/home/ahoy/tools/rpi/hoymiles/__init__.py", line 13, in <module>
    import crcmod
ModuleNotFoundError: No module named 'crcmod'

Das Modul ist aber installiert und auch im Pfad angegeben.

Beim systemd start:
bu@Raspi:~ $ sudo systemctl status ahoy
● ahoy.service - ahoy (lumapu) as Service
     Loaded: loaded (/home/ahoy/tools/rpi/ahoy.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Sun 2023-05-28 09:53:34 CEST; 2s ago
    Process: 714 ExecStart=/usr/bin/python3 -um hoymiles --log-transactions --verbose --config ahoy.yml (code=exited, status=1/FAILURE)
   Main PID: 714 (code=exited, status=1/FAILURE)
        CPU: 387ms

Wenn ich stoppe und manuell Systemd starte:
bu@Raspi:~ $ sudo systemctl status ahoy
● ahoy.service - ahoy (lumapu) as Service
     Loaded: loaded (/home/ahoy/tools/rpi/ahoy.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2023-05-28 09:56:30 CEST; 5s ago
   Main PID: 736 (python3)
      Tasks: 1 (limit: 3933)
        CPU: 334ms
     CGroup: /system.slice/ahoy.service
             └─736 /usr/bin/python3 -um hoymiles --log-transactions --verbose --config ahoy.yml

aber kurz danach:

bu@Raspi:~ $ sudo systemctl status ahoy
● ahoy.service - ahoy (lumapu) as Service
     Loaded: loaded (/home/ahoy/tools/rpi/ahoy.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Sun 2023-05-28 09:57:08 CEST; 4s ago
    Process: 741 ExecStart=/usr/bin/python3 -um hoymiles --log-transactions --verbose --config ahoy.yml (code=exited, s>
   Main PID: 741 (code=exited, status=1/FAILURE)
        CPU: 447ms

May 28 09:56:30 Raspi systemd[1]: Started ahoy (lumapu) as Service.

Verkabelt ist genau wie abgebildet

Für Hilfe dankbar
Michi

stefan123t, 2024-10-27:
@PaeserBastelstube kannst Du hier helfen ?

PaeserBastelstube, 2024-10-27:
Ich schaue es mir gleich einmal an, bin aber schon müde. Ausführliche
Antwort kommt morgen 

Viele Grüße aus der Päser-Bastelstube, Knut

(Issue #966)

 

<https://github.com/lumapu/ahoy/issues/966#issuecomment-2440163839> , or
unsubscribe
U5APAVCNFSM6AAAAABQWEN74KVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDINBQGE3DG
OBTHE> .
<https://github.com/notifications/beacon/A5DELEAO4OQHI4OHRDRV7E3Z5U5APA5CNFS
M6AAAAABQWEN74KWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTU

PaeserBastelstube, 2024-10-27:
Hallo Michi,
konntest Du die Fehlermelung vom 28.05.2023 bereits überwinden?
Falls Nein, melde Dich bitte noch einmal.
.
Zur weiteren Fehleranalyse stoppe bitte den AHOY System-Startprozess:
`sudo systemctl stop ahoy`
Schaue bitte mit `ps -ef | grep ahoy` ob wirklich alle alten Prozesse beendet sind. `kill -9` hilft an dieser Stelle.
Nun prüfe zuerst, ob wirklich alle notwendigen Python-Module installiert sind. Hier gibt es gerne den Fehler, dass die Python-Module als USER installiert wurden, später aber mit `sudo` als `root` gestartet werden sollen:
```
python3 -m pip list --verbose
sudo python3 -m pip list --verbose 
```
Nun müssen wir das AHOY-Startkommando manuell ausführen, ohne die Umleitung: `| tee -a log2.log` usw.
begib Dich zuerst in das AHOY-RPI-Verzeichnis:
```
cd /home/ahoy/tools/rpi
python3 -um hoymiles --log-transactions --verbose --config ahoy.yml

```
In diesem Verzeichnis wird die Log-Datei `RPI-AHOY-DTU.log` angelegt. Falls Du alles aus dieser Datei angezeigt haben willst, setze bitte vor dem Python-Start -einmalig- ein:  `tail -f RPI-AHOY-DTU.log &`  ab. Das läuft nun im Hintergrund und kann mit `jobs` überwacht und mit `fg` wieder in den Vordergrund geholt werden.
.
Nun hast Du etwas Rüstzeug, melde Dich bitte noch einmal mit der Log-Datei - viel Erfolg

stefan123t, 2024-10-29:
@PaeserBastelstube Danke für Deine Antwort. Da Du der letzte verbliebene Mohikaner mit dem Raspberry Pi bist, habe ich den Issue mal Dir zugeordnet. Auch die letzten Commits bzgl. des Python Codes stammen ja aus Deiner Bastelstube, dafür also herzlichen Dank!

Jo131313, 2024-11-01:
Hallo vielen Dank, für die Antwort.
Ich hatte schon gar nicht mehr damit gerechnet.
Hab mir in der Zwischenzeit einen Esp32 besorgt, der nun läuft.
Komme aber auf dein Angebot gerne zurück.
Mein Raspi ist derzeit aber anders eingesetzt.
Meinst du man bekäme es auch auf einem Zero oder einem Zero 2 zu laufen?
BG
Michi

PaeserBastelstube, 2024-11-01:
Hallo Michi,
ich denke schon, dass es funktionieren müsste - habe es aber nicht getestet.
Der Unterschied eines Raspberry Pi 4 und eines Zero:
   4 GB vs 0.5GB
   4 x 1.8 GHz vs 1 x 1 GHz
   40-poligen GPIO-Anschluss (nur Anschlusspunkte, keine Pins)
sollte für die AHOY-Pythen Programme kein Problem sein.
Aber was soll mit den eingelesenen Daten passieren?
Soll auf dem System auch ein MQTT-Broker laufen oder ein nginx - oder sogar ein Volkszähler?
Im erstem Moment sehe ich noch nicht den Vorteil beim Einsatz eines Zero's.
Könntest Du bitte Deine Motivation zum Einsatz eines Zero beschreiben.
.
Ich werd mir mal einen bestellen - dauert nun einige Tage

PaeserBastelstube, 2024-11-05:
Hallo Michi,
heute erhielt ich ein sehr großes Paket eines Elektronik Versandhandels von der Nordsee, darum war ein kleiner "RASP PI ZERO2 WH". Die Mini-SD eines anderen Raspi wurde eingesteckt, das RadioModul verdrahtet und es LÄUFT.
Sogar mit Volkszähler.
Du kannst nun testen ...

Jo131313, 2024-11-06:
Hallo Knut,

DANKE. Du bist aber fix.
Leider wird es bei mir etwas dauern bis ich zum Testen komme.
Hab nur die Wochenenden zur Verfügung und da hab ich derzeit einige andere Arbeiten zu erledigen.
Der Zero liegt zumindest startbereit in der Lade.

BG Michi.