lumapu/ahoy #967: keine Live Daten von Inverter, Keine Leistungsbegrenzung möglich

Roy12343, 2023-05-29:
## Hardware
  * [ X] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [X ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [X ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Ich bin absolut Beginner:
Folgendes Problem liegt vor: Eine Hoymiles 1500 mit 4 Modulen produziert Strom.
grünes LED blinkt im sekunden Takt
Meine AhoyDtu erkennt den Inverter und empfängt die seriellen Daten alle 30sec.
Leider werden aber keine LIVE Daten angezeigt, alles auf 0.
PowerLimit kann ich auch nicht setzen.
Error: inverter does not accept dev control request at this moment

ich arbeite mit der Ahoy-Dtu Oberfläche auf meinem Handy,
da meine Fritzbox Sie nicht findet, ein ping zur Ermittlung der ip-Adresse ist nicht erfolgreich. 
Alle WLAN Geräte sehen Ahoy DTU  

Viele Grüße

mtet2, 2023-05-30:
Du hast das Netzwerk in der Ahaydtu noch nicht konfiguriert.
Im folgenden Link das Kapitel "Ahoy Einrichten" genau befolgen. Es stehen dort alle erforderlichen Infos.
https://ahoydtu.de/getting_started/
Deine DTU arbeitet zur Zeit noch als "Access Point". Sie sollte im "Client Mode" mit deinem Router verbunden werden.

Roy12343, 2023-05-31:
Danke für den Hinweis. Aber bevor ich hier poste, habe ich getting_startet natürlich beachtet. 
Der ESP kann mein Netzwerk scannen aber nach Eingabe meines Wlanschlüssels und reboot, erkennt meine Fritzbox die AHOYDTU nicht in ihrem Netz.
Finde auch keine ip. Habs mit MACadresse und einer festen IP versucht. nichts klappt.
Das andere Problem ist ja das die AhoyDTU als Access Point meinen Inverter erkennt, aber nur die seriellen Daten alle 30 sec ausliest und nicht die Live-Daten, begrenzen kann ich ihn auch nicht. Fehlermeldung siehe oben

jimknopf63, 2023-06-01:
Du musst die Ahoy DTU ja auch in dein Heimnetzwerk einbinden. Sollte das nicht funktionieren hast Du entweder dein WLAN nur auf bekannte Geräte beschränkt oder Sonderzeichen/Leerzeichen in SSID bzw PW. Und nach dem Eintrag des Netzwerkes im Ahoy musst Du auch speichern und einen Reboot auslösen.

Roy12343, 2023-06-01:
Danke Jimknopf, ein Problem ist jetzt erledigt. Hatte nur bekannte Geräte zugelassen. Toller Hinweis!!
 
Hat noch jemand eine Idee, wie ich an meine Live-Daten komme?
Bekomme derzeit nur die seriellen Daten. Bei der Eingabe der Leistungsbegrenzung kommt immer noch die Fehlermeldung:
Error: inverter does not accept dev control request at this moment

Roy12343, 2023-06-04:
Ich habe mal das nRF24L01+ Modul getauscht. Leider kein positives Ergebnis.

Weiterhin nur serielle Daten. Es muss irgendwie an den Einstellungen auf der DTU liegen.

lumapu, 2023-06-04:
@Roy12343 hast du auch mal in #968 und #969 gelesen? Da führt der (bekannte) Hinweis IRQ und CE zu tauschen (Hardware und Software) zum Erfolg

Roy12343, 2023-06-05:
Bei mir hat's auch geklappt. IRQ von D3 nach D1 am wmos und settings IRQ auf D1,
dann Leistung setzen. Prima , Vielen Dank.

Roy12343, 2023-06-05:
So Long

closed 2023-06-05 by Roy12343