lumapu/ahoy #968: Neue WR scheinen nicht mehr zu gehen (HM600 Kein DTU empfang)

soehnholz, 2023-05-29:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Normal Setting, only Inverter was added

### Error description

The DTU can not Connect to Inverter.

heldmannm, 2023-05-29:
Hallo
habe das gleiche Problem, wo kann ich die alte Software finden

soehnholz, 2023-05-29:
Hi,

das wird dir evtl. auch nichts bringen. Hab schon 4 alte duchgetestet, immer das selbe. Wenn du es selber versuchen möchtest findest du sie oben unter "Actions"
Es will den Inverter nicht finden, ich habe sehr viel versucht. Ich hab schon die vermutung das der Inverter das DTU Modul nichts sendet (Kaputt, Standby.....) ich weiß es nicht. Das was unsere Programierer hier gemacht haben funktioniert aber (Danke an der Stelle für die sehr guten Anleitungen)
Ich vermute aber das evtl. der Hersteller hinterher gezogen hat und was "eingebaut/geflasht" haben was die Free DTU Geschichten unterbindet. Warten wir mal ob sich mehr Melden zu dem Thema hier. U R Welcome!!

heldmannm, 2023-05-29:
Warum hat es dann mit der alten Version über Monate funktioniert und jetzt mit der neuen nicht mehr das ganze ist nicht logisch. Kann man hier eine Anleitung finden zum downgrade da ich bis jetzt immer den Web Installer  genutzt habe

rejoe2, 2023-05-29:
> Warum hat es dann mit der alten Version über Monate funktioniert und jetzt mit der neuen nicht mehr das ganze ist nicht logisch. Kann man hier eine Anleitung finden zum downgrade da ich bis jetzt immer den Web Installer genutzt habe

_Falls_ du die PIN-Settings nicht gecheckt hast (alle LED auf off!), solltest du als erstes das ändern. Ansonsten kann man die "actiuons" downloaden und dann die passende Firmware-Datei per "update" über das Web-Interface von Ahoy hochladen, ist "selbsterklärend"...

rejoe2, 2023-05-29:
Was ist das für eine Seriennummer, die Probleme macht? (Bzw. was ist das darin vercodete Herstellungsdatum?)

Generell sollte zumindest die Abfrage der firmware- und Hardware-Revisionsnummer gehen - das sollte immer das "uralte" Protokoll sein. 
Wenn nicht, sollte man als erstes die Verkabelung checken, v.a. auch des Interrupt-Pins (den kann nämlich die firmware nicht als schlecht verbunden erkennen!).

Ansonsten mal die Abfragezyklen auf ca. 5 Sekunden oder sogar kleiner setzen.

soehnholz, 2023-05-29:
Ok , ich habe 2 DTU´s gebaut bei beiden auf die verkabelung geachtet, verschiedene Firmwarestände geflasht, und, und.... aber du schreibst was von LED OFF, äh ja bei mir leuchtet die dauerhaft (die neben der WLAN Antenne).... was meinst du mit interrupt Pin? Gibt es evtl. doch noch ein Problem was ich da nicht beachtet habe? Ich bin seeehr dankbar für einen Tip, bin schon davor die Hoymiles wieder zurück zu schicken, da ich die langsam in Vermutung habe. Liefert aber die volle Leistung, funkt nur nix (oder DTU empfängt nix)

lumapu, 2023-05-29:
beim ESP8266 muss man je nach Modul teilweise leider IRQ über CE vertauschen - sowohl Hardware als auch Software.

Wäre toll, den du uns die ersten Stellen deiner Seriennummer mitteilen könntest. Welche Serie verwendest du? HM oder HMS?

soehnholz, 2023-05-29:
Ja natürlich sehr gerne! Ich habe eine HM 600 S/N: 11418294xxxx.
Vielen dank für den Tipp, das muss ich sofort ausprobieren!
@lumapu 👍 Vielen dank an euch die das hier machen, ganz große Klasse!

Bastelfoehni, 2023-05-29:
Bei mir das gleiche

lumapu, 2023-05-29:
die blaue LED ist beim ESP8266 also Wemos mit dem Pin D4 verbunden, daher leuchtet die wenn man den Pin verwendet je nach Zustand des Pins

lumapu, 2023-05-29:
interessant wäre das USB Log gleich ganz am Anfang, wenn die ersten Anfragen an den Wechsellichter raus gehen. Bitte hierfür das Häkchen bei `Serial Debug` in den Einstellungen von Ahoy setzen, um auch die empfangen Daten sehen zu können, speichern und Reboot nicht vergessen 😉

Dymnag, 2023-05-29:
20:09:42 I: (#0) enqueued cmd failed/timeout
20:09:42 I: (#0) resetPayload
20:09:42 I: (#0) Requesting Inv SN 11418482****
20:09:42 I: (#0) prepareDevInformCmd 0x0b
20:09:42 15 pid: 80
20:09:42 I: TX 27B Ch23 | 15 84 82 21 86 84 50 82 63 80 0b 00 64 75 06 86 00 00 00 00 00 00 00 00 f1 c1 ab 
20:09:52 I: (#0) enqueued cmd failed/timeout

Dymnag, 2023-05-29:
Moin, ich hab das gleiche Problem.... bei mir ist D3-D4 noch vertauscht SW&HD.
Ich war bis an dem Inverter dran, um Empfangsproblemchen auszuschließen.

Dymnag, 2023-05-29:
nrf24l01+
is connected
Datarate
250 kbps
Power Level
LOW
TX count
48
RX success
0
RX fail
0
RX no answer
47
RX fragments
0
TX retransmits
0

und dem nach zu urteilen, geht einiges raus aber nichts kommt zurück.

Bastelfoehni, 2023-05-30:
Bei mir hat geholfen, als Namen auf Namen, welcher neben der Seriennummer steht, zu ändern. Dabei scheint es wichtig zu sein, das man die Leerschläge weglässt. Bei mir war das HRS0.EU.HM. Weiter habe ich die Sendeleistung auf High gestellt und die Akktualisierung auf alle 5Sekunden gestellt. Danach hat es funktioniert. Pinout habe ich nichts geändert. 

>
>
> Ich habe jetzt mal die Pinouts gedreht, CE und IRQ Hard wie auch Software und leider keine Veränderungen.............ich weiß auch net ......

-------------------------------------------------------------------------------------------------
FreeMail powered by mail.ch

Bastelfoehni, 2023-05-30:
Bei mir hat geholfen, als Namen auf Namen, welcher neben der Seriennummer steht, zu ändern. Dabei scheint es wichtig zu sein, das man die Leerschläge weglässt. Bei mir war das HRS0.EU.HM. Weiter habe ich die Sendeleistung auf High gestellt und die Akktualisierung auf alle 5Sekunden gestellt. Danach hat es funktioniert. Pinout habe ich nichts geändert.

>
>
> interessant wäre das USB Log gleich ganz am Anfang, wenn die ersten Anfragen an den Wechsellichter raus gehen. Bitte hierfür das Häkchen bei Serial Debug in den Einstellungen von Ahoy setzen, um auch die empfangen Daten sehen zu können, speichern und Reboot nicht vergessen 😉

-------------------------------------------------------------------------------------------------
FreeMail powered by mail.ch

soehnholz, 2023-05-30:
13:27:05 I: (#0) enqueued cmd failed/timeout
13:27:05 I: (#0) resetPayload
13:27:05 I: (#0) Requesting Inv SN 114182940496
13:27:05 I: (#0) enqueCommand: 0x01
13:27:05 I: (#0) enqueCommand: 0x0b
13:27:05 I: (#0) prepareDevInformCmd 0x01
13:27:05 15 pid: 80

leider nichts bis jetzt, auch mit ändern des Namens

nrf24l01+
is connected
Datarate
250 kbps
Power Level
HIGH
TX count
287
RX success
0
RX fail
0
RX no answer
286
RX fragments
0
TX retransmits
0

Dymnag, 2023-05-30:
Threat 2 mit dem gleichen Problem:

https://github.com/lumapu/ahoy/issues/969#issuecomment-1568423855

soehnholz, 2023-06-01:
Hi Leute,

ich habe Ahoy jetzt ans laufen bekommen jedoch nicht mit einem ESP 8266 sondern ESP32 NMCU. Ich weiß nicht ob ich von den 4 Funkmodulen 2 kaputt hatte oder , oder , oder......wie dem auch sei, danke für die Arbeit die Ihr euch hier macht und von der wir Profitieren, großes Kino!

Vermutung im Nachgang, da hat was mit der Pinbelegung des NRF net geklappt (hab aber penibelst darauf geachtet!)
Jetzt mit ESP32 und der Belegung die man auf der URL sehen kann funktioniert es (hab auch noch ein neues RF Modul benutz. werde defekte im nachhinein mal testen)

Danke!

PS: habt Ihr noch n Schema wo man die Display Belegung auf dem ESP32 macht?

johndoe0815, 2023-06-02:
Ich schließe mich an, gleiche Probleme wie hier beschrieben, neuer HM-600 mit Seriennummer 114184...aber: Es ging schon mal!
Ich habe einen ESP32 und ein nRF24L01+ mit Antenne.
Zuerst habe ich OpenDTU ausprobiert. Nach Pin-Korrektur (Pin 16 gibt es auf meinem ESP32 WROOM nicht), lief der Empfang aller Daten einwandfrei für ein paar Stunden, per USB noch im Computer eingesteckt.
Also wollte ich das Ding an seinen endgültigen Platz umziehen (noch deutlich näher am Wechselrichter), dort wieder eingesteckt, kein Empfang mehr. 
Alle möglichen Dinge rumprobiert, nichts. 
Dann AhoyDTU ausprobiert. Pin 16 auch hier korrigiert, Wechselrichter eingetragen (neben Seriennummer sogar mal den Namen wie auf dem Schild ausprobiert, weil das hier auf Github einer als Lösung nannte), kein Empfang.
Das Log zeigt connected und gibt nur timeouts aus, war in OpenDTU auch so:
[image]

System sieht für mich auch gut aus, außer dass keine erfolgreiche Verbindung:
[image]

Ich habe verschiedene Kabel und Netzteile probiert. Ich habe alle Kabel zwischen ESP und nRF noch mal erneuert. Nichts, bleibt weiter tot.
Am Wechselrichter kann es eigentlich nicht liegen, denn es ging ja mal. Hab den auch mal vom Strom genommen, hat nichts geändert. Außer einem defekten ESP32 oder nRF fällt mir gerade nichts mehr ein, wobei ja in AhoyDTU für mich alles gut aussieht, und er scheint ja auch Requests rauszuschicken, nur keine Antwort zu bekommen?!

leliep, 2023-06-02:
Hast Du alle LEDs auf „OFF/default“, LED polarity auf „low active“?

johndoe0815, 2023-06-02:
Ja, das ist ja auch die Standardeinstellung. Nur den IRQ-Pin geändert:
[image]

soehnholz, 2023-06-02:
Hi JohnDoe0815,

ich habe auch den "Leidensweg" durch. DIY halt! Was ich empfehlen kann ist mal ein anders NRF Modul zu benutzen "soweit vorhanden" hatte eine 4er Packung wo der 3 dann klappte, kann aber auch Einbildung sein. Mein Weg ! Ich habe alles auf default gelassen, soweit vorweg. ESP32 mit WEB Flasher geflasht und die Verdrahtung des NRF wie beschrieben gemacht (guckst du hier https://github.com/lumapu/ahoy/blob/main/doc/Wiring_ESP32_Schematic.png ) wie gesagt, die Einstellungen so lassen, aber Prüfen das MISO, IRQ, usw, da stecken wie auch am Modul. Ja und dann kennt jeder den Weg......über WLAN an der DTU Anmelden und das Netzwerk konfigurieren, wenn das gemacht ist, nix mehr weiter machen sondern die DTU abmachen vom PC und an einer externen (ausreichenden Spannungsversorgung anschließen min. 1A) und gucken das man per WLAN drauf kommt. Funktioniert das schon mal, gut! Dann die DTU in der nähe des WR in Betrieb nehmen (1-2 Meter weg von der DTU, mehr net) um auszuschließen das dort "irgendwas"stören könnte. Wenn du dann die DTU über WLAN erreichen kannst und sie nah am WR aufgebaut ist, gibts du erst die Seriennummer ein! Da hat es bei mir wie von Zauberhand angefangen zu funken. Ich weiß im nachgang auch noch nicht woran es gelegen hat aber ich habe stark die Verdrahtung in Verdacht, ist diese nicht richtig physisch wie logisch gemacht, geht der Spaß nicht. Die Software ist gut und funktioniert, bedenkt bitte "Fertigungstoleranzen" in den Bauteilen wie einem NFR Modul, statische Endladungen die man dem Bauteil beim anfassen unabsichtlich aussetzt usw. können alles Möglichkeiten sein wesegen sowas nicht geht. Am besten von jedem Bauteil 3 zur Auswahl haben und durchtesten, ist halt so. Ich bin an dieser Stelle vieleicht langsam lästig mit meinem Lob, aber was die Damen und Herren hier an Arbeit geleistet haben ist echt Spitze. Reveerse Engeneering muss man drauf haben, Respekt. Und es funktioniert und das ohne Cloud, geiiiil! THANK YOU Lumapu and others, i dont´n know. Und alle die es versuchen wollen oder noch am versuchen sind, macht weiter, es geht und lohnt sich um so mehr wenn es funktioniert

[image]

cheesemountain0815, 2023-06-02:
[image]

Hallo, ich bin neu in der Materie. Kann mir einer das Log erklären?
Grüße Sven

Jo131313, 2023-06-03:
Auch ich habe dasselbe Problem, allerdings mit einem HM800.
Nachdem ich vorige Woche mit meinem Raspberry gescheitert war, hab ich jetzt mit dem ESP8266 dasselbe Problem.
Laut Log wird gesendet aber nichts empfangen.
Mir ist aber nicht klar welche andere Verkabelung als die abgebildete ich versuchen könnte.
Möchte ja nicht den Rauch aus der Schaltung entweichen lassen.
Vielleicht würde ja doch etwas bei den neueren WR verändert

cheesemountain0815, 2023-06-03:
Ist denn mein Log soweit in Ordnung?

cheesemountain0815, 2023-06-03:
[image]

Jo131313, 2023-06-03:
Dein log ist genauso in Ordnung wie bei mir, ich kann auch nichts empfangen. Es fehlt die Antwort vom WR.
Du solltest die Seriennummer deines WR im Bild oben schwärzen (zumindest die 5 letzten Ziffern).

lumapu, 2023-06-04:
> Ist denn mein Log soweit in Ordnung?

deine `heap-fragmentation` ist etwas hoch, evtl. nochmal neu booten (den ESP). Ansonsten fehlen wie bei den anderen auch die Antworten vom Wechselrichter - echt komisch.

cheesemountain0815, 2023-06-04:
Ich habe gerade nochmal geschaut WR blinkt grün

cheesemountain0815, 2023-06-04:
[System.pdf](https://github.com/lumapu/ahoy/files/11646073/System.pdf)

lumapu, 2023-06-04:
testet doch bitte mal diese Firmware - ich denke das war noch vor der Sende- / Empfangsumstellung / -optimierung

[ahoy_0.5.107_esp8266_29315b9.bin.zip](https://github.com/lumapu/ahoy/files/11646249
/ahoy_0.5.107_esp8266_29315b9.bin.zip)

johndoe0815, 2023-06-04:
> testet doch bitte mal diese Firmware - ich denke das war noch vor der Sende- / Empfangsumstellung / -optimierung
> 
> [ahoy_0.5.107_esp8266_29315b9.bin.zip](https://github.com/lumapu/ahoy/files/11646249/ahoy_0.5.107_esp8266_29315b9.bin.zip)

Ich schaffe es leider nicht, die zu installieren. Egal welches Tool ich nehme, z.B. den NodeMCU-PyFlasher exakt wie hier beschrieben (https://github.com/lumapu/ahoy/blob/main/Getting_Started.md#using-a-ready-to-flash-binary-using-nodemcu-pyflasher), der Flashvorgang sieht zwar einwandfrei aus, aber er spannt danach kein WLAN auf. Das klappt nur wenn ich die aktuelle Version über den Web-Flasher installiere...

lumapu, 2023-06-04:
geht es auch nicht über die in Ahoy integrierte Update-Funktion?
Alternativ habe ich noch eine `esptool` Variante, allerdings ungetestet. Im `bat`-file muss noch der COM Port angepasst werden.

[ahoy_0.5.107_esp8266_29315b9-esptool.zip](https://github.com/lumapu/ahoy/files/11646464/ahoy_0.5.107_esp8266_29315b9-esptool.zip)

cheesemountain0815, 2023-06-05:
Ich habe 0.5.17 installiert,  jetzt komme ich nicht mehr in die Benutzeroberfläche (192.168.4.1)

johndoe0815, 2023-06-05:
Über die interne Update-Funktion lässt sich die 0.5.107  nicht installieren, da kommt "Update: failed" und er rebooted wieder in die vorherige Version.

Ich habe es jetzt mit deiner BAT probiert, wie bei den anderen Tools läuft auch hier die Installation offensichtlich einwandfrei durch:
```
ahoy_0.5.107_esp8266_29315b9-esptool>python esptool.py -p COM4 -b 921600 write_flash 0x0 ahoy_0.5.107_esp8266_29315b9.bin --erase-all
esptool.py v3.1
Serial port COM4
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:f0:9e:f8:66:bc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Erasing flash (this may take a while)...
Chip erase completed successfully in 11.5s
Compressed 545744 bytes to 388483...
Wrote 545744 bytes (388483 compressed) at 0x00000000 in 5.4 seconds (effective 803.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

wie auch bei meinen vorherigen Versuchen spannt er dann aber kein WLAN auf, mit dem ich mich verbinden könnte.

Mich wundert ja, dass in dem anderen Thread (https://github.com/lumapu/ahoy/issues/969) nun gleich mehrere User schreiben, Nutzung eines anderen Pins wäre die Lösung.
Ich nutze ja einen ESP32, habe auch da mal für CE einen anderen Pin probiert, aber das hat nichts geändert.
Normalerweise würde ich ja mit defekter Hardware beim ESP oder NRF rechnen und die neu bestellen, wäre nicht das erste Mal bei sowas. Aber die Vielzahl an Stimmen hier (wenngleich es bestimmt auch bei vielen klappt, die hier nicht schreiben), lässt einen ja schon stutzig werden, ob das tatsächlich alles Hardwarefehler bzw. nicht funktionierende Fake-Bauteile sein sollen...

Dymnag, 2023-06-05:
Moin,
ich habe nun auch zum Inverter. Ich musste eine neuen NRF+ und die PINS anpassen (IQR auf D5). Ach und ich hab die 0.6.15 Version drauf. Jetzt geht's, obgleich nur jede 3-5 Abfrage auch ankommt. ABER das liegt vermutlich daran, dass ich keinen Kondensator angeschlossen hab.
Vermutlich ist ein Container mit den China NRF fehlerhaft. Hab welche von AZ geholt. Die funktionieren erwartungsgemäß.
Danke an alle für die tolle Unterstützung!!

lumapu, 2023-06-05:
@johndoe0815 

> Ich nutze ja einen ESP32, habe auch da mal für CE einen anderen Pin probiert, aber das hat nichts geändert.

Die Firmware von oben ist nur für den ESP8266.
Bisher habe ich noch nicht gelesen, dass ein ESP32 andere Pins bräuchte. Ich hatte erst letzte Woche den ersten ESP32 in der Hand, der nicht einwandfrei funktioniert hat: Man konnte sich partout nicht mit dem AP WLAN verbinden, ich weiß nicht woran es lag

Pumapower-tech, 2023-06-05:
Hallo Zusammen, ich habe auch einen neuen HM-600 damit hatte ich mit der Firmware 0.6.9 auch am Anfang Probleme. TX gingen raus aber es kamen keine RX. Habe dann den NRF gewechselt und es lief sofort. Ich hatte mir die Bestückungsseite des NRF angeschaut und mit dem funktionierenden verglichen. Tatsächlich fehlte auf dem fehlerhaften Teil ein SMD Widerstand. Bei meinem 3. NRF war diese vorhanden aber schräg aufgelötet (nur 1 Pin angeschlossen). Ich hatte diesen dann gerichtet und siehe da, er spielte auch. Ich vermute das es hier eine Charge fehlerhafte Teile verkauft werden. Ich kann diese reparieren, aber für alle anderen sicher nervig.

dtuuser, 2023-06-06:
Mit der neuen Version 0.7.1 bekam ich auch zuerst keine Daten. NRF Modul nicht gefunden. Unter Setting war es aus.
Nach dem Einschalten und Reboot kamen nur manchmal Daten durch. 
Zurück zur 0.6.15 und alles klappt wieder super.
Auf meinem Produktionsstick habe ich seit 38 Tagen ohne Restart die 0.5.92 am Laufen.
Aber....immer schon CE an D2 und IRQ an D1

cheesemountain0815, 2023-06-06:
> Neues NRF ist erstmal ein guter Anfang. Aber vielleicht ist auch das Funkmodul des WR selbst defekt? Wenn du noch Zugriff auf einen anderen Hoymiles (vielleicht aus dem Bekanntenkreis) hättest, könntest du nochmal probieren deine DTU damit zu verbinden.

Ich habe leider keinen im Bekanntenkreis zum testen.
Wenn ich das Modul morgen habe werde ich testen und berichten.

johndoe0815, 2023-06-09:
So, update von meiner Seite:
Ich habe mir einen neuen NRF von Makershop bezogen, und es funktioniert.
Es lag also tatsächlich ausschließlich am NRF, das hätte ich nicht gedacht...

cheesemountain0815, 2023-06-09:
Ich hoffe morgen kommt das NRF Modul , dann kann ich auch testen.

Jo131313, 2023-06-12:
Gibt es eine Möglichkeit die Funktion der NRFs zu überprüfen. Ich hab hier vier und bekomme keine Verbindung mit dem WR zustande.
Da sie mein Händler Paar-weise vertreibt - kann man die Kommunikation von zweien herstellen und prüfen?

cheesemountain0815, 2023-06-12:
Auch von mir eine Erfolgsmeldung. Es lag an dem NRF. Neues mit Stabantenne funktioniert.

johndoe0815, 2023-06-13:
Offensichtlich also doch alles nur geballtes Hardware-Versagen, auch wenn die Fehleranalyse schwierig ist bzw. man auf Basis der angezeigten Daten schnell zu anderen Rückschlüssen kommt. Habe schon oft genug bei solchem Kleinkram günstig aus China bestellt, aber hier nun auch reklamiert und mein Geld zurück bekommen, der NRF von Makershop für ein paar Euro mehr war dann einwandfrei. Bei diesem Bauteil spart man sich also offensichtlich einiges an potenziellem Frust, wenn man nicht zum billigsten Anbieter greift.

cheesemountain0815, 2023-06-13:
Am Ende beziehen doch die hiesigen Händler doch auch nur aus China.
Nun ja wie lautet das Sprichwort: Wer billig kauft kauft zweimal😉

lumapu, 2023-06-15:
Lösung: 
> **Offensichtlich also doch alles nur geballtes Hardware-Versagen** 

Manche Module scheinen defekt zu sein, und da ist wohl gerade eine ganze Charge in Umlauf gebracht worden.

closed 2023-06-15 by lumapu