lumapu/ahoy #970: [Bug]

Dymnag, 2023-05-30:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.9

### Github Hash

15ec6a0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Diverses, da connection Probleme mit dem Inverter. Siehe #968

### Error description

GUI via PC funktioniert ohne Mängel.
GUI via Smartphone nicht.
Einstellungen (Setup) kann nicht geändert werden, Inverter ADD-button ohne funktion, viele Anzeigen sind nur unvollständig.
Destop-view hat hier etwas abhilfe geschaffen, aber nicht vollständig.  -_O_-

Ich bin neu hier und im Bereich ETI, aber geiler Scheiß den ihr hier aufgestellt habt.

khbraemer, 2023-06-01:
Hallo, ich würde ein aktuelles Smartphone nehmen so ab 2016.

Dymnag, 2023-06-01:
Nun, dann erfülle ich mit einem NOKIA X30 G5 die Voraussetzungen ;)

lumapu, 2023-06-04:
prüfe mal, ob der Browser evtl. `https` Verbdinungen erzwingt, das nervt zunehmend bei lokalen Zugriffen. Ahoy ist nur per `http` erreichbar

closed 2024-01-02 by lumapu