lumapu/ahoy #971: [Bug] Ahoy 0.6.15 freezes/dies due to poor wifi connection

tastendruecker123, 2023-05-31:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.6.15

### Github Hash

6dcf5ee

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

-One HM-1500 configured with a 30 sec interval
-MQTT broker configured (hosted locally)
-nothing else (web interface not used, no Prometheus etc.)

### Debug Serial Log output

```shell
After running for about 48 hours, Ahoy became unresponsive and either crashed/froze or lost its wifi connection and didn't reconnect.

I'm tracking ping responses for the ESP that Ahoy is running on and a separate one that's monitoring a Pylontech battery in Grafana. Both have somewhat marginal wifi reception and are located very close to each other.
```

### Error description

After running for about 48 hours, Ahoy became unresponsive and either crashed/froze or lost its wifi connection and didn't reconnect.

I'm tracking ping responses for the ESP that Ahoy is running on and a separate one that's monitoring a Pylontech battery in Grafana. Both have somewhat marginal wifi reception and are located very close to each other. Today's chart looks like this:

[image]

A value of 5 seconds means there was no ping response, which means Ahoy died right around 18:37 at a time when the reception was pretty bad. Since the only wifi activity should be MQTT (and perhaps NTP), it looks like one of those actions has the ability to crash Ahoy if the connection is marginal. Is there a timeout missing somewhere?

lumapu, 2023-06-04:
I had no crash for a long time an running always on the latest development build. Do you have logged the heap fragmentation too? I'm using a ESP8266 with one connected inverter.

reinhard-brandstaedter, 2023-07-22:
I'm experiencing something similar (ahoy-dtu dies sporadically). I haven't linked it to poor WiFi connection, but I will try to measure/ confirm it with ping transmits to two devices next to eachother as well.

lumapu, 2023-07-22:
have you tried latest development builds?

closed 2024-01-02 by lumapu