lumapu/ahoy #973: Bug oder Feature?: "your NRF24 module can't be reached"

nexulm, 2023-06-01:
Hallo,

wer kennt das Problem. Meine AhoyDTU lief 4 Wochen lang ohne Auffälligkeiten, damals mit der v0.6.0.
Seit dem Update auf 0.6.9 über 0.6.12, 0.6.14 und seit gestern Mittag 0.6.15 habe ich diese Auffälligkeit nach 1-x Tagen:
[image]
---
Gestern habe ich den ESP8266 einmal erased und neu mit 0.6.15 (zuvor 0.6.14) geflasht, da die 0.6.14 nach ca. 3/4 Tagen ohne Auffälligkeiten ein Kommunikationsproblem mit den NRF24 meldete.
Ist dies nun ein SW oder HW Thema?
Wer hat sonst noch solche Beobachtungen gemacht?
Ich werde nun wieder zurück auf 0.6.0 flashen und dann mal abwarten sowie berichten!

nexulm, 2023-06-01:
Zur Info:
Ich habe mich gegen "platt" machen entschieden und einfach nur eine PowerOnReset durchgeführt. Damit ging es zuletzt mit der v0.6.14 nie weshalb ich gestern vor dem HW-Tausch den SW-Tausch (= Update auf v0.6.15 mit Erfolg) vorgezogen habe.
[image]
Warum bitte funktioniert nun wieder alles.
Liegt das Problem am nächtlichen deaktivieren des Polling Modus oder am nächtlichen Ausschalten des WiFi-Routers?
Erklären kann ich mir diese Zusammenhänge (noch) nicht sehe hier nur unterschiedliches Verhalten.
Werde im ersten Schritt nun das nächtliche Polling durchlaufen lassen.

nexulm, 2023-06-01:
Zur Info:
Ich habe mich gegen "platt" machen entschieden und einfach nur eine PowerOnReset durchgeführt. Damit ging es zuletzt mit der v0.6.14 nie weshalb ich gestern vor dem HW-Tausch den SW-Tausch (= Update auf v0.6.15 mit Erfolg) vorgezogen habe.
[image]
Warum bitte funktioniert nun wieder alles?
Liegt das Problem am nächtlichen deaktivieren des Polling Modus oder am nächtlichen Ausschalten des WiFi-Routers?
Erklären kann ich mir diese Zusammenhänge (noch) nicht sehe hier nur unterschiedliches Verhalten.
Werde im ersten Schritt nun das nächtliche Polling durchlaufen lassen.

knickohr, 2023-06-01:
Das Problem lag wohl daran das die Einstellungen für die beiden LEDs nicht auf „off“ standen, sondern vermutlich auf 0.

Der Bug ist weithin bekannt, kann aber nicht gefixed werden. Tritt auch nur auf wenn von einer früheren Version unterhalb 0.6.9 upgedated wird.

nexulm, 2023-06-01:
Das kann ich so nicht bestätigen. Ich habe gestern Mittag mit einem Neuflash-Vorgang inkl. Erase per USB die v0.15 mit den Radio Default Einstellungen geflasht.
[image]

knickohr, 2023-06-01:
Du hast einen Erase gemacht 😉

nexulm, 2023-06-01:
Ja, alles neu konfiguriert. Zumindest hat der NodeMCU Py flasher mit „yes, wipes all data“ mir das Gefühl gegeben, start from scratch. 😀

nexulm, 2023-06-02:
**Update;** Heute Morgen lief alles unauffällig. Einziger Unterschied zum Vortag war, dass "Pause polling inverters during night" deaktiviert war.
Ich habe den soeben wieder aktiviert um eine Reproduzierbarkeit (morgen früh) zu prüfen.

lumapu, 2023-06-04:
der Fortschritt hier klingt vielversprechend. Ich nehme an, da man nichts mehr gehört hat funktioniert es jetzt 😉

nexulm, 2023-06-04:
Es lässt sich nicht so leicht reproduzieren. :-( 
Gestern Vormittag konnte der HM1500 problemlos -obwohl der Polling Mode über Nacht deaktiviert war- angesprochen und ausgelesen werden. Der Fehler mit dem NRF24 Modul konnte also nicht nachgestellt werden. 
Gestern Mittag musste dann aufgrund von Elektro-Installationsmaßnahmen das Haus einmal stromlos gemacht werden (= PowerOn Reset für alle Systeme ;-)). Auch hiernach hat gestern und heute alles erfolgreich funktioniert.
Somit heißt es dann Stabilität auf Beobachtung und hoffentlich läuft alles das "lange Wochenende" durch. Ist halt immer ärgerlich, wenn man nicht vor Ort ist, nicht an die Hardware rankommt und die kontinuierliche Datenaufzeichnung gestört wird. :-)

closed 2023-06-04 by nexulm