lumapu/ahoy #975: MQTT Datenübernahme InfluxDB

tester277, 2023-06-05:
Hallo Zusammen,
benötige Unterstützung:
Es geloingt mir nicht die MQTT Daten in meine Influx DB zu übernehmen.
Habe in die telegraf.conf den Abschnitt übernommen (oder gehört das in die addtions/inputs.mqtt_consumer.conf?):
```
[[inputs.mqtt_consumer]]`
  servers = ["tcp://mosquitto:1883"]
  topics = [
    "ahoy-dtu/+/+/+",
    "ahoy-dtu/+/+",
  ]
  data_format = "value"
  data_type = "float"

[[inputs.mqtt_consumer.topic_parsing]]
  topic = "ahoy-dtu/+/+/+"
  measurement = "measurement/_/_/_"
  tags = "_/name/ch/field"

[[inputs.mqtt_consumer.topic_parsing]]
  topic = "ahoy-dtu/+/+"
  measurement = "measurement/_/_"
  tags = "_/name/field"

[[processors.pivot]]
  tag_key = "field"
  value_key = "value"
```
Telegraf Log meldet:
2023-06-05T16:14:41Z I! Starting Telegraf 1.26.3
2023-06-05T16:14:41Z I! Available plugins: 235 inputs, 9 aggregators, 27 processors, 22 parsers, 57 outputs, 2 secret-stores
2023-06-05T16:14:41Z I! Loaded inputs: mqtt_consumer
2023-06-05T16:14:41Z I! Loaded aggregators:
2023-06-05T16:14:41Z I! Loaded processors: pivot
2023-06-05T16:14:41Z I! Loaded secretstores:
2023-06-05T16:14:41Z I! Loaded outputs: influxdb
2023-06-05T16:14:41Z I! Tags enabled: host=iotstack
2023-06-05T16:14:41Z I! [agent] Config: Interval:10s, Quiet:false, Hostname:"iotstack", Flush Interval:10s
2023-06-05T16:14:41Z W! [outputs.influxdb] When writing to [http://x.x.x.x:8086]: database "pv1" creation failed: Post "http://192.168.172.221:8086/query": read tcp 192.168.48.8:38078->x.x.x.x:8086: read: connection reset by peer
2023-06-05T16:14:41Z I! [inputs.mqtt_consumer] Connected [tcp://mosquitto:1883]
Ich sehe aber nicht das ein Wert in die Datenbank zu Ahoy in die Datenbank geschrieben wird.
Kann jemand eine funktionierende Konfig posten oder schauen was fehlt?

Vielen Dank

lumapu, 2023-06-05:
bitte in den issue nach `InfluxDB` suchen falls nicht schon geschehen. Hier wäre doch ein ganz guter Treffer:
https://github.com/lumapu/ahoy/issues/517

Ansosten, hier sind die Suchergebnisse:
https://github.com/lumapu/ahoy/issues?q=influxdb

HFarryranz, 2023-06-06:
Bei mir sieht das so aus: (Auszug aus telegraf.conf )

[[inputs.mqtt_consumer]]
  ## Broker URLs for the MQTT server or cluster.  To connect to multiple
  ## clusters or standalone servers, use a separate plugin instance.
  ##   example: servers = ["tcp://localhost:1883"]

  servers = ["tcp://IP:1883"]
  ## Topics that will be subscribed to.
  topics = [
    "inverter/#"
   ]

  data_format =  "value"
  data_type = "float"