lumapu/ahoy #976: > Ich habe nun fast alle Versionen durchprobiert,  keine Verbindung zum WR.

cheesemountain0815, 2023-06-05:
> Ich habe nun fast alle Versionen durchprobiert,  keine Verbindung zum WR.
> ... 
> Aktuell die 0.6.3
> Was sagt mir das?

Du müsstest unter System noch das Debug Häkchen setzen, dann kommt noch mehr Info.
Welcher ESP ist in Einsatz? Wurden die Verbindungen gelötet?

Edit: Bei der 0.5.107: Wird dort ein AP aufgespannt? evtl. geht er auch direkt in dein WLAN falls du mal Einstellungen gespeichert hattest, schau dich bitte mal in der Fritzbox oä

_Originally posted by @lumapu in https://github.com/lumapu/ahoy/issues/968#issuecomment-1577349914_

Es ist ein Esp8266 im Einsatz. Ich konnte mich mit allen Versionen im Ahoy wlan anmelden.
Einstellungen hatte ich keine gespeichert.

gmed-ed, 2023-06-11:
Sorry, aber du gibst so wenig Informationen zu deinem Aufbau an, da kann man unmöglich fundiert Hilfestellung leisten.

Das es ein 8266 ist, ist ja schön,
aber eie sieht deine Verbindung zum Funkmodul aus?
Wie sind deine Gpio Einstellungen? etc. pp.

Genau deswegen gibt es das Formular am Anfang als Hilfestellung, welche Info's gebraucht werden.

cheesemountain0815, 2023-06-11:
Wo kann ich denn meine Config eintragen?

gmed-ed, 2023-06-12:
Schau fir das mal an:

https://github.com/lumapu/ahoy/blob/main/Getting_Started.md

Unter wirering sind Beispiele für die Verbindung zwischen esp und Funkmodul.

Was hast du da genau?

Welchen Wevhselrichter hast du genau?

Wenn beides klar ist, kann man mehr sagen.

cheesemountain0815, 2023-06-12:
Das Thema kann als geschlossen betrachtet werden. Ein neues Funkmodul mit Stabantenne brachte Erfolg.

gmed-ed, 2023-06-12:
schön, das es nun läuft

cheesemountain0815, 2023-06-12:
Man hätte ja auch Glück haben können mit dem Chinaimport.
Jetzt mach ich mich an die Sache mit dem Homeassistant.

closed 2023-06-15 by lumapu