lumapu/ahoy #977: setting sporadisch weg...

smarteshome2020, 2023-06-06:
Hi, 
sporadisch ist die Webseite des Ahoy um einige Menüpunkte reduziert. 
Einige Minuten später oder nach einem Power down geht es wieder und alle Parameter sind wie gewohnt verfügbar.

Board: ESP8266
V0.6.9

[image]

smarteshome2020, 2023-06-06:
nochmal zum vergleich. Leider keine Ahnung was du für Daten zur Analyse benötigst.

[image]

HFarryranz, 2023-06-07:
Habe ich auch immer wieder einmal. Manchmal startet der ESP neu. 
Meist wenn ich auf LIVE schalte.

knickohr, 2023-06-07:
Das ist der 8266, der ist am Limit, muß zu viel machen.

Nimm einen ESP32 und die Probleme sind weg 😉

smarteshome2020, 2023-06-08:
Danke, aber was soll da jetzt "enabled" werden. Diesen Haken finde ich dort nicht. 
Habe mal auf "max" umgestellt, bin mir aber nicht sicher ob das gemeint war und was es bringt.

[image]

Harald-U, 2023-06-08:
Setzt den Powerlevel wieder runter auf LOW, sonst hast Du nur noch mehr Probleme ... außer Dein WR ist sehr weit weg.

Du must gar nichts enablen, ich hatte da ein Problem mit der neuen Dev-Version 0.7.1, da fehlt nach dem Flashen ein Haken fürs Funkmodul (Issue #979). Ich habe am Ende dieses Issue gelinkt, weil mit der Version 0.7.1 jetzt Dein Problem bei mir auch gehäuft auftaucht. Und Github fügt dann diesen Querverweis in Deinem (diesem) Issue ein ...

closed 2024-01-02 by lumapu