lumapu/ahoy #978: 0.7.1 meldet "your NRF24 module can't be reached, check the wiring and pinout"

HFarryranz, 2023-06-07:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.1

### Github Hash

f0848ec

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

ESP8266

CS D8
CE D2
IRQ D1

### Error description

Mit Version 0.7.1 bekomme ich nach dem Start sofort die Meldung "your NRF24 module can't be reached, check the wiring and pinout" und es findet keine Kommunikation mit dem WR statt.

Zurück zu 0.6.15 Meldung kommt nicht und alles funktioniert. Wiederholbar.

Meine Verbindung ist:

CS D8
CE D2
IRQ D1

soelli22, 2023-06-07:
Das Problem hatte ich auch. Mach noch mal das Update und schaue dann mal unter
- Settings
- System Config
- Radio
Dort war bei mir bei NRF24 kein Haken mehr drin. Den einmal rein und speichern, dann lief es.

HFarryranz, 2023-06-07:
Das war es!  Danke für den Hinweis.

Keine Ahnung wie oft ich gestern die Settings-Seite angeschaut habe. Das der Haken nicht sitzt habe ich nicht bemerkt.

closed 2023-06-07 by HFarryranz