lumapu/ahoy #979: [Bug] Version 0.7.1 "your NRF24 module can't be reached"

Harald-U, 2023-06-07:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.1

### Github Hash

f0848ec

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT Broker gesetzt, Intervall 0

### Error description

Über funktionierende Version 0.6.15 die Version 0.7.1 geflashed (über "Update" im Webinterface).

Nach Start zeigt die AhoyDTU-Seite "your NRF24 module can't be reached, check the wiring and pinout"

[image]

Log von Update auf 0.7.1 bis Downgrade 0.6.15: 
[Ahoy_0.7.1.log](https://github.com/lumapu/ahoy/files/11679804/Ahoy_0.7.1.log)

kiu77, 2023-06-08:
Der "bug" besteht lediglich darin, dass in 0.7.1 der Haken bei _Settings_ -> _System Config_ bei _NRF24 Enable_ nicht gesetzt ist. Setzen, saven und alles ist gut.

[image]

lumapu, 2023-06-08:
richtig, ich werde das für den ESP8266 entfernen, das es dort keinen Sinn ergibt.
Die Änderung kam durch den HMS/HMT Support rein.

Harald-U, 2023-06-08:
Das kann dann wohl zugemacht werden, danke für Eure Hilfe!!!

closed 2023-06-08 by Harald-U