lumapu/ahoy #980: Ordner Total/YieldDay und YieldTotal gehen nachdem der WR offline geht auf 0

teletapi, 2023-06-07:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.1

### Github Hash

GIT SHA: f0848ec :: 0.7.1

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

.

### Debug Serial Log output

```shell
.
```

### Error description

Seid version 0.6.10  werden im Ordner Mqtt/0/Inverter/Total/YieldDay und Yield Total die Werte nachdem der Wechselrichter offline geht genullt so das diese Werte für eine datenbank und Visualisierung nicht mehr zur Verfügung stehen.

knickohr, 2023-06-07:
Siehe #927

teletapi, 2023-06-07:
Hatte ich nicht auf dem Schirm

lumapu, 2023-06-09:
heute getestet und hat funktioniert - was sagt ihr?

knickohr, 2023-06-10:
Geht nicht, der Build ging in die Hose 😲 Außerdem kann ich erst testen wenn es auch wieder mit dem S3 Fusion klappt 😇

teletapi, 2023-06-11:
Läuft wieder super Danke

knickohr, 2023-06-11:
Ja, ich hab’s gestern auch bemerkt als das Total nicht auf Null ging. Dann probiere ich jetzt die Option aus 😉

teletapi, 2023-06-16:
Seid der Version 0.7.4 ist der mqtt Fehler mit yieldDay und yieldTotal im Ordner Total wieder da und geht nach offline auf 0

lumapu, 2023-06-17:
tatsächlich - ich sehe es auch

closed 2023-08-05 by lumapu